VoodooI2C/VoodooI2C #171: Please add support for ELAN0406

stonevil, 2019-04-08:
Please add support for ELAN0406 used in Razer Blade Advanced 2019 and other models.

I'm developer, so I'm ready to help with testing, etc.
Sorry, not a C/C++ dev.

williambj1, 2019-04-08:
Hi @stonevil ,

Could you provide more details about your problem or talk about it in Gitter, please.
I'm using ASUS's Zephyrus S with GX531GS,
I'm very curious about whether you are facing the same problem as mine. **(Unknown Slave address)**

Thx!

stonevil, 2019-04-15:
Hi @williambj1 ,

Sorry for long response. Traveling a lot lately.
Maybe. Maybe not. Somebody mentioned that ELAN0406 is not supported by  VoodooI2C project.

From log file:
2019-04-03 17:51:52.782141-0700 0x40f      Default     0x0                  0      0    kernel: I2C0: match category IODefaultMatchCategory exists
2019-04-03 17:51:52.788849-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 Probing controller
2019-04-03 17:51:52.788857-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 Found valid Synopsys component, continuing with initialisation
2019-04-03 17:51:52.789182-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a368 SSCN not implemented in ACPI tables
2019-04-03 17:51:52.789235-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerNub::pci8086,a368 FMCN not implemented in ACPI tables
2019-04-03 17:51:52.789238-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 Warning: Error getting bus config, using defaults where necessary
2019-04-03 17:51:52.789330-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 Publishing device nubs
2019-04-03 17:51:52.789342-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 Found I2C device: SYNA2393
2019-04-03 17:51:52.790318-0700 0x435      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOCannonLakeH
2019-04-03 17:51:54.883147-0700 0x5e3      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 I2C Transaction error details
2019-04-03 17:51:54.883151-0700 0x5e3      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 slave address not acknowledged (7bit mode)
2019-04-03 17:51:54.883155-0700 0x5e3      Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 I2C Transaction error: 0x08800001 - aborting
2019-04-03 17:51:54.883159-0700 0x5e3      Default     0x0                  0      0    kernel: (VoodooI2CHID) VoodooI2CHIDDevice::SYNA2393 Request for HID descriptor failed
2019-04-03 17:51:54.883161-0700 0x5e3      Default     0x0                  0      0    kernel: (VoodooI2CHID) VoodooI2CHIDDevice::SYNA2393 Could not get HID descriptor

williambj1, 2019-04-16:
@stonevil 
Seems that you are facing the exact issue as mine!
```
2019-04-03 17:51:54.883151-0700 0x5e3 Default 0x0 0 0 kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,a368 slave address not acknowledged (7bit mode)
```
I've asked @alexandred before about this, seems that the slave address in the DSDT isn't the real one. And I don't know how to solve it 😫

stonevil, 2019-04-16:
@williambj1 
From Windows

Driver Name: msmouse.inf
Class Guid: {4D36E96F-E325-11CE-BFC1-08002BE10318}
Driver Date: 06/21/2006
Driver Version: 10.0.17134.1
Driver Provider: Microsoft
Driver Section: HID_Mouse_Inst.NT
Driver Rank: 0xFF1004
Matching Device Id: HID_DEVICE_SYSTEM_MOUSE
Outranked Drivers: input.inf:HID_DEVICE:00FF1006
Device Updated: false
Parent Device: ACPI\ELAN0406\4&10968218&0s

williambj1, 2019-04-16:
[image]

This would be certain😀

stonevil, 2019-04-16:
@williambj1 

[image]

williambj1, 2019-04-16:
@stonevil So now we are actually facing the same issue. I think you may change the title of this `issue` to `slave address not acknowledged in new ELAN devices`
Hope this problem could be solved!

stonevil, 2019-04-20:
@williambj1 can you share url for this forums? Just to speedup

williambj1, 2019-04-21:
@stonevil 
I'm sorry that I can't find the forum and I can't remember where I saw it, but I found the same author's post on another forum.
https://askubuntu.com/questions/1036615/asus-gu501gm-zephyrus-elantech-touchpad-troubleshooting
Hope this can help you.

williambj1, 2019-04-22:
@stonevil Great news!!
This problem has been solved! 
I'll post the solution here later on.

Cheers!

williambj1, 2019-04-22:
1. Find the place where your SBFB is
2. Scroll up and you will find `FMCN`
3. Paste this above `FMCN`
```
            Method (SSCN, 0, NotSerialized)
            {
                Return (PKG3 (SSH1, SSL1, SSD1))
            }
```
4. Search `SSCN` and find another `SSCN` related to your I2C device. Delete it or rename it into something else.
5. Restart and have a try!

alexandred, 2019-04-27:
Solved as it is a DSDT issue which is fixed with a patch (in future we will add instructions on how to solve this in the guide).

closed 2019-04-27 by alexandred