VoodooI2C/VoodooI2C #179: Conflicts with ACPIBatteryManager kext

hrytskivr, 2019-04-15:
I2C/HID kext conflicts with ACPIBatteryManager kext which makes touchpad "crazy-jumpy" when on battery power, everything goes back to normal when plugged-in though.
I'm using VirtualSMC/SMCBattery kexts instead of FakeSMC/ACPIBatteryManager to avoid this issue.
Any thoughts what might be the cause?

LeeBinder, 2019-04-18:
@hrytskivr What's bad for you using VirtualSMC/SMCBattery kexts? Why would you prefer using FakeSMC/ACPIBatteryManager kexts?

hrytskivr, 2019-04-24:
I have non-stable boot with those for some reason.
3 times out of 10 won't boot.

LeeBinder, 2019-04-24:
Oh that sucks. I take for granted you have:
- FakeSMC removed
- VirtualSmc.efi present in drivers64UEFI
- SMCHelper-64.efi removed from drivers64UEFI
- both, VirtualSMC.kext and SMCBatteryManager.kext, installed correctly to /L/E (not just in Clover/kexts)
- updated all your kexts to latest versions, easiest via [Kext Updater](https://bitbucket.org/profdrluigi/kextupdater/downloads/)
- are running the latest version of Clover
?

hrytskivr, 2019-04-28:
@LeeBinder
Everything but VirtualSmc.efi present in drivers64UEFI, I though it's necessary for FireVault stuff only

LeeBinder, 2019-04-28:
@hrytskivr : oh you are right: "VirtualSmc-64.efi: Advanced Apple SMC Emulator. VirtualSMC.efi module is recommended for boot.efi compatibility when FileVault 2 is enabled"

https://github.com/acidanthera/VirtualSMC/blob/master/Docs/FAQ.md

Richylitschy, 2019-05-06:
apparently the Voodoo kext and the ACPI kext use the same BUS, which thus gets overloaded. Read it somewhere on tonymac, so not 100% sure.

FreeJHack, 2019-07-30:
I found the solution. Maybe could help, take a look.
https://www.tonymacx86.com/threads/solution-trackpad-voodooi2chid-conflicts-with-acpibatterymanager-kext.281416/

hrytskivr, 2019-08-01:
Thanks @FreeJHack !

closed 2019-08-01 by hrytskivr