VoodooI2C/VoodooI2C #183: Trackpad randomly stops working in polling mode

hieplpvip, 2019-04-21:
# Specs
* Asus Zenbook UX410
* ELAN1200 (buggy GPIO pinning so I have to use polling)

# Problem
Trackpad randomly stops working: the cursor wouldn't move. Sleeping and waking make it work again.

# Log files
will upload when I have them

overnightmare95, 2019-04-22:
Same happening to me. ELAN1200 with buggy DSDT.

black-dragon74, 2019-04-30:
I can confirm it happens on ASUS ROG GL-503.

ELAN 1200, polling mode. Can't use GPIO interrupts yet as it is not supported for CFL.

P.S: If you leave your trackpad for sometime it will start working automatically just it way it stopped. Putting to sleep and waking up always works though.

Regards

williambj1, 2019-05-05:
Met once, still need to confirm on my device

LeeBinder, 2019-05-14:
for Asus Laptops, maybe whatnameisit's polling SSDT + ACPI entry help:
https://github.com/tctien342/Asus-Vivobook-S510UA-High-Sierra-10.13-Hackintosh/issues/25
He created it for his X510UA-BQ490 (I think with ELAN1300).

whatnameisit, 2019-05-14:
I created the SSDT and Clover rename based on the one provided by @tctien342 and with helps from Ben at VoodooI2C troubleshooting chatroom...credits should go where they're supposed to :)

LeeBinder, 2019-05-14:
;) allright. Do you know the differences betw. 1200 and 1300 in regards to polling and SSDT modding?

whatnameisit, 2019-05-18:
@LeeBinder , I just began using VoodooI2C with my Asus laptop, so I don't know much. :/ 
Please read this guide: https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning
And I think the problem I had are somewhat different from the touchpad stopping its functionality until sleep and wake. It was that all gestures only continue to do their stuff for a few seconds, so I have to click or touch from where they stopped or redo them, i.e. I can only drag for half a second and it breaks and it leaves the file in the wrong place, and I have to start dragging from where the files are placed. Or two finger swipe scrolling stops and I have to take my fingers off and do the gesture again.

LeeBinder, 2019-05-18:
@whatnameisit : I can somewhat relate to "I can only drag for half a second and it breaks and it leaves the file in the wrong place, and I have to start dragging from where the files are placed", even though on my ELAN1300 it was only that bad with older revisions of VoodooI2C 2.1.x - much better now but for me still not as fluent as with v.2.0.3, which is why my main macOS is still High Sierra (v.2.0.3 is legacy and does NOT work in Mojave).

Thanks for the link - with my current time provision it would be too much for me to get into right now. I'm pretty happy with your polling SSDT fix + VoodooI2C legacy 2.0.3 in High Sierra :) With v.2.1.6 regardless of in Mojave or High Sierra, I am experiencing this small delay/ "stickiness" whenever I cut/ paste/ copy: the trackpad/ "mouse" pointer stalls for half a second or so, and I can't work like that, so I stick to 10.13.6 until this maybe gets fixed some time in the future (even though Alex wrote this is inherent to the magic trackpad driver engine, but I hope someone figures it out at some point).

williambj1, 2019-05-19:
> @whatnameisit : I can somewhat relate to "**_I can only drag for half a second and it breaks and it leaves the file in the wrong place, and I have to start dragging from where the files are placed_**" 

I'm facing the same issue mentioned in #190.

whatnameisit, 2019-05-19:
@williambj1 That's similar to the problem I had. First I switched from interrupts to polling and all gestures other than tapping repeated automatically so that if I click on a file/folder it automatically opens. Then I asked @ben9923 and he directed me to a hotpatch to set USTP equal to one which is kept in your repo!
As for physical buttons, I have to press harder than in Windows to perform a click or drag and keep them going. Maybe that's related to what @blankmac is saying?

HolzMichel, 2019-06-08:
i have the same problem on Asus UX310 (Kabylake), but never noticed this issue before replacing my built in Wifi card with a DW1560. In my eyes it seems like the stops are most times then, when there are state changes in Wifi, eg. searching for Network after wake or losing connection. For me, most times when the trackpad gets stuck, the Wifi icon in the menu bar is "greyed out" or shows the "searching animation". Maybe someting strange interferences here.

LeeBinder, 2019-06-09:
@HolzMichel : hm odd, because the WiFi is on the USB bus but the touchpad isn't.

Do you experience what you describe regardless of mode (interrupts AND polling) or only in one of the modes? I take for granted you are aware of whatnameisit's https://github.com/tctien342/Asus-Vivobook-S510UA-High-Sierra-10.13-Hackintosh/issues/25 ?

Also you should most likely better not be using USBInjectAll.kext (+ USB "UIAC" SSDT), if you do.

HolzMichel, 2019-06-09:
I am basically using the zenbook repository of @hieplpvip, so i would have to see how to find out in which mode my trackpad is running, but as @hieplpvip mentioned in the opening post we have "(buggy GPIO pinning so I have to use polling)" with these zenbooks, so I guess we are using polling mode. I don´t use USBInjectALL.kext, i use a USBPorts.kext created in hackintool. Never had any USB issues so far. I yesterday changed from ACPIBatterymanger.kext to SMCBatterymanager.kext and since then did not notice this hangs again.

ben9923, 2019-08-08:
@hieplpvip When it happens, can you try pressing `Print Screen` on your keyboard?
VoodooPS2 sends notifications for enabling/disabling the trackpad when pressing this key, I suspect a bug in VoodooPS2 might be causing that issue.

whatnameisit, 2019-09-03:
I had the same issue with the touchpad randomly stopping. I tried to collect I2C log, but nothing showed up. Then I disabled I2C0 device and I think it works all right, but recently I haven't used hackintosh much. So not sure if disabling I2C0 really fixed the issue. Can someone else test it?

black-dragon74, 2019-09-03:
@whatnameisit On my machine all other I2C devices are disabled from the beginning. Doesn't seem to make a difference at all.

whatnameisit, 2019-09-03:
@black-dragon74 :/ Then maybe the fix on my touchpad didn't work at all...

LeeBinder, 2019-09-03:
@whatnameisit let me ask you here re. I2C0 (I'll reply to your other remarks over at tonymac's): Can you please elaborate on "turn off I2C0 device by having M(_STA) on I2C0 return Zero"? Where and how exactly do I do that - what code to alter or insert, in SSDT-ELAN-POLLING.aml/ dsl, or DSDT via hotpatch in clover.plist?

whatnameisit, 2019-09-03:
@LeeBinder First I looked at IOReg and noticed I2C0 had VoodooI2C attached. Then I looked at my Clover-extracted DSDT and I2C0 had Method _STA inside some if condition return a nonzero value. So I decided to disable original M(_STA) by renaming it to XSTA with method header and some of its contents like in Rehabman's guide (https://www.tonymacx86.com/threads/guide-using-clover-to-hotpatch-acpi.200137/). I created a new _STA method in one of my SSDTs to return zero. After reboot, I2C0 was gone from the IOReg tree, and used the hack for about an hour and experienced no random stops. It was a pure guess and it might not really fix the issue because black-dragon74 said all I2Cx excluding the one with his touchpad are disabled from beginning. If I wait for some more time and the touchpad might stop I think. I'm on my phone so if you need the patch I can provide it when I get my hands on my machine

LeeBinder, 2019-09-03:
Gotcha. Yes, I'd appreciate if you sent me the patch at your convenience. I'm in Europe and it's far beyond bed time, so I won't get to this again before next nite anyway, so no hurry 👍 

It happens very rarely anyway, only when I'm in a hurry and "rape" my trackpad by performing too many motions in too short a time frame. I realized that in Windows 10 in the same situation, I get a lag glitch, too, but the Windows driver always manages to settle and stabilize the motion after 1 second or so.

black-dragon74, 2019-09-04:
@leebinder Yes this GPIO issue can be seen in Windows too. ASUS really needs to fix this. Too many devices with the GPIO bug.

whatnameisit, 2019-09-04:
@LeeBinder Here's the patch
config
Find: <140f5f53 544100a4 4c535441 534d4430>
Replace: <140f5853 544100a4 4c535441 534d4430>
If I remember correctly, the config patch changes the whole method
```
Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    Return (LSTA (SMD0))
                }
```
to
```
Method (XSTA, 0, NotSerialized)  // _STA: Status
                {
                    Return (LSTA (SMD0))
                }
```
so the original _STA isn't used anywhere. Please look at your own DSDT and see if you get the same Hex code as mine (please refer to https://www.tonymacx86.com/threads/guide-using-clover-to-hotpatch-acpi.200137/ if you need anything)
Then below SSDT has new _STA method to return zero and disable I2C0.
[SSDT-disable-I2C0.aml.zip](https://github.com/alexandred/VoodooI2C/files/3575338/SSDT-disable-I2C0.aml.zip)

LeeBinder, 2019-09-04:
superb, will do. SSDT-disable-I2C0.aml needs to be placed into Clover/ACPI/patched IN ADDITION to SSDT-ELAN-POLLING.aml, NOT replacing it, right?

whatnameisit, 2019-09-05:
That's right. SSDT-ELAN-POLLING.aml enables the touchpad at I2C1.ETPD, SSDT-disable-I2C0.aml disables the unused I2C0.

LeeBinder, 2019-09-11:
me being silent/ not hearing from me = not experiencing any hangs whatsoever (so far and hopefully for good) 🥇

LeeBinder, 2019-12-15:
> @LeeBinder Here's the patch
> [snip]
> [SSDT-disable-I2C0.aml.zip](https://github.com/alexandred/VoodooI2C/files/3575338/SSDT-disable-I2C0.aml.zip)

@whatnameisit , this proves not to be a good idea anymore: with the latest [VoodooI2C v.2.3](https://github.com/alexandred/VoodooI2C/releases/tag/2.3) this causes kernel panic on wake from sleep for me (regardless of in GPIO or polling mode). I've been testing back and forth for a while last hour: this really seems to be the culprit, because as soon as I deactivate the hotpatch, remove the SSDT and reboot, WUpFS works fine again (for good in Mojave, not tested Catalina yet).

Maybe because of this fix:

> Fix issue with some devices failing to meet the deadline in commandSleep:
>    • Correct IOCommandGate::commandSleep call

Thought you might want to know about this, too.

Will be observing now with v.2.3 if pointer hangs are gone for good now, hopefully..

whatnameisit, 2019-12-16:
@LeeBinder 
Thank you for reporting. I tested sleep and wake with the v2.3 kexts and I also have kp on wake.
Removing the patch to disable I2C0 fixed kp issues. I too hope the random stops would disappear with v2.3.

LeeBinder, 2019-12-16:
@whatnameisit 
glad I could save you debugging time :)

Now I hope that the really annoying regression from VoodooI2C "legacy" 2.0.3 to 2.1+ of highlighting/ dragging getting lost half way in the air constantly as perfectly described [here](https://github.com/alexandred/VoodooI2C/issues/208) by @williambj1 can get solved next. I am requesting that anybody affected adds their "me, too" over there so the issue gets some reinforcement. Right now there is too little subscribers over there which needs improvement, for starters.

ben9923, 2019-12-16:
@LeeBinder @whatnameisit Could either of you reproduce this kernel panic while booting with `keepsyms=1` and attach the panic log?
While you solved the panic by eliminating the patch, I'm still interested to know if we can avoid the KP altogether. Usually the expected result would be non-functional trackpad rather than a panic.

LeeBinder, 2019-12-16:
@ben9923 : my clover boot sections already has been containing keepsyms=1 which thus did not prevent KP on wake with the patch enabled.

So far, with v.2.3, no sudden stops that require a reboot :)

panic log: I'm very sleepy already so please give me a nudge and jog my memory how to get to the panic log the quickest/ easiest way (a key word suffices)

ben9923, 2019-12-16:
> @ben9923 : my clover boot sections already has been containing keepsyms=1 which thus did not prevent KP on wake with the patch enabled.
>
> So far, with v.2.3, no sudden stops that require a reboot :) [...]

`keepsyms=1` is only meant to keep symbols on panic for easier debugging, shouldn't really prevent any panic.

I believe you can find them in `/Library/Logs/DiagnosticReports` :)

whatnameisit, 2019-12-17:
@ben9923 , I wonder if this is what you wanted.

[Kernel_2019-12-17-200141_Jinmos-MacBook-Pro.panic.zip](https://github.com/alexandred/VoodooI2C/files/3972904/Kernel_2019-12-17-200141_Jinmos-MacBook-Pro.panic.zip)

For me, I had no sudden stops with v2.3 kexts. Will report when I get one.

LeeBinder, 2019-12-17:
@ben9923 : me, too :) custom provoked kp + log just for you, how's that for the fun of it (and neither ever any sudden stops while working in macOS Mojave with v2.3 kexts)

[Kernel_2019-12-17-115402_Lees-MacBook-Pro.panic.zip](https://github.com/alexandred/VoodooI2C/files/3973146/Kernel_2019-12-17-115402_Lees-MacBook-Pro.panic.zip)

ben9923, 2019-12-18:
@whatnameisit @LeeBinder Thank you for the logs :)

Did you also have a proper rename for the original `_STA` method?
Otherwise it's an invalid ACPI configuration (duplicate symbols), which might break the device initialization process.
When removing the disabled I2C controller from the kext's Info.plist (With same ACPI configuration), I assume there's no panic?

whatnameisit, 2019-12-19:
@ben9923 , LeeBinder used the patch I posted on Sept 5 by renaming M(I2C0._STA) to XSTA and returning false with a new M(_STA). I looked at his DSDT and the rename patch should apply to both of our systems correctly. I don't think he had experienced any kp until v2.3 VoodooI2C kexts. I switched from M(_STA) to XSTA to something else([else.txt](https://github.com/alexandred/VoodooI2C/files/3984275/else.txt)) to return false from M(I2C0._STA) and also had not experienced any kp until v2.3.
And yes, removing I2C0's pci8086,9d60 match from VoodooI2C.kext causes no kp upon wake from sleep, yet VoodooI2C does not attach to I2C0 as intended. Now I wonder if the Apple's stock I2C is only incompatible and does not affect any touchpad performance.
[Jinmo’s MacBook Pro.zip](https://github.com/alexandred/VoodooI2C/files/3984285/Jinmo.s.MacBook.Pro.zip)
Any thoughts, comments, and other things will be appreciated :)

LeeBinder, 2019-12-19:
correct, I did not experience any KP with stock alexandred v.2.2 and neither with [Erictoby's v2.2 + Beta M](https://github.com/Erictoby/VoodooI2C/releases/) (Oct 11, 2019). KPs with I2C0 disabled via above patch surfaced during the dev of v.2.3, most probable since al3xtjames's commits (either one or both): [5b1e29f](https://github.com/alexandred/VoodooI2C/commit/5b1e29f8367329d8794c439f65f52b808fde5b1a) | [ee26207](https://github.com/alexandred/VoodooI2CHID/commit/ee2620720850a652eb493dbd377da16c1fd57fef)

LeeBinder, 2019-12-20:
@whatnameisit I wrote to you on https://gitter.im/whatnameisit (not sure if you bet notifications).

ben9923, 2019-12-21:
@whatnameisit @LeeBinder Not sure why it panicked in 2.3 with _STA 0, then...
AFAIK, _STA 0 should prevent kext matching. Never investigated that too much, though.

As it seems like the issue we're commenting on is resolved, I'll close it.
If panics are showing up again for some reason, we should really have a new, separate issue.
@hieplpvip In case you still experience those stops, feel free to re-open.

closed 2019-12-21 by ben9923

brunogermain, 2020-04-04:
Hello everyone. I'm reopening this because I'm actually still experiencing random trackpad stops with VoodooI2C v2.3 and Mojave.
I read this thread over and over, but I couldn't understand whether or not I should patch VoodooI2C by removing pci8086,9d60 match from VoodooI2C.kext.
[debug_UX310UA_Mojave.zip](https://github.com/alexandred/VoodooI2C/files/4431123/debug_UX310UA_Mojave.zip)

LeeBinder, 2020-04-04:
@brunogermain try https://github.com/dhinakg/VoodooI2C/releases

brunogermain, 2020-04-05:
> @brunogermain try https://github.com/dhinakg/VoodooI2C/releases

@LeeBinder just replace my kext with this?

LeeBinder, 2020-04-05:
plural: kextS..

brunogermain, 2020-04-05:
VoodooI2C and VoodooI2CHID

LeeBinder, 2020-04-05:
Correct. I edited my previous comment

brunogermain, 2020-04-05:
if I try to click and drag, it just opens the document or folder i click. no dragging

Mateo1234454545, 2020-04-05:
@brunogermain , same problem see here : https://github.com/alexandred/VoodooI2C/issues/269
Can you try this :https://github.com/daliansky/XiaoMi-Pro-Hackintosh/files/4394815/VoodooI2C.Mar.27.zip

brunogermain, 2020-04-05:
@Mateo1234454545 , same behaviour

Mateo1234454545, 2020-04-05:
> @Mateo1234454545 , same behaviour

Can you check if you have voodooinput.kext loaded? e.g with hackintool.
If you do , disable it and reboot.

brunogermain, 2020-04-05:
No Voodooinput.kext loaded

brunogermain, 2020-04-05:
I've been experimenting with different kexts and it appears that the only one working properly, but still freezing every now and then is v2.3 only until the commit of February 18. 
The one I downloaded today from github (https://github.com/alexandred/VoodooI2C/releases/tag/2.3) broke dragging. Reverting to the one I downloaded one month ago brought everything back to normal

brunogermain, 2020-04-05:
@Mateo1234454545 , Voodooinput gets loaded with last v2.3, but not with the February 2020 version. How do I disable it?

Mateo1234454545, 2020-04-05:
You can just delete it from kexts/other , but if you do your trackpad will stop working. At least that happens for mine.

brunogermain, 2020-04-05:
> You can just delete it from kexts/other , but if you do your trackpad will stop working. At least that happens for mine.

Correct... I guess I'll stick with v2.3 before Voodooinput was added to the plugins. That one one works good, it just rarely freeze when you stress it with too many gestures. At least that's my guess

brunogermain, 2020-04-05:
The question is: does disabling I2C0 via SSDT and removing pci8086,9d60 (to prevent KP) solve the trackpad freeze issue?

LeeBinder, 2020-04-06:
Just FYI:  dhinakg's VoodooI2C.kext 2.3.2 from https://github.com/dhinakg/VoodooI2C/releases *contains* VoodooInput inside VoodooI2C.kext/Contents/PlugIns!!
Also FYI, dhinakg wrote in a note that he's aware that that's *not* how it's desired by the other devs who request VoodooInput to please be supplied **separately** as an additional kext...

> Can you try this :https://github.com/daliansky/XiaoMi-Pro-Hackintosh/files/4394815/VoodooI2C.Mar.27.zip

@Mateo1234454545 : I see this version does *not* contain VoodooInput inside VoodooI2C.kext/Contents/PlugIns. There is not ReadMe inside, so my questions are:

1. who compiled these kexts?
2. does this version require VoodooInput to be installed additionally?

LeeBinder, 2020-04-06:
> The question is: does disabling I2C0 via SSDT and removing pci8086,9d60 (to prevent KP) solve the trackpad freeze issue?

@brunogermain : try, test and report back?

Mateo1234454545, 2020-04-06:
@LeeBinder , daliansky compiled it and no , there is no requirement for voodooinput. That's the reason , I believe, it is working well on machine.

williambj1, 2020-04-06:
@Mateo1234454545

Understand what is VoodooInput before you speak, you are misleading others.

brunogermain, 2020-04-06:
@LeeBinder I'm currently testing, but I don't know how to reproduce the trackpad freeze, it just happens randomly.

One more question, is there a way remove pci8086,9d60 via Clover KextsToPatch? 

I wrote this to add to my config.plist:

			<dict>
				<key>Comment</key>
				<string>Remove pci8086,9d60 device from VoodooI2C</string>
				<key>Disabled</key>
				<true/>
				<key>Find</key>
				<data>
				OWQ2MA==
				</data>
				<key>InfoPlistPatch</key>
				<true/>
				<key>Name</key>
				<string>VoodooI2C</string>
				<key>Replace</key>
				<data>
				OWQ2OQ==
				</data>
			</dict>

It's disabled now, since it wasn't working. 
I just deleted the line from VoodooI2C's Info.plist to prevent KP on wake from sleep

Thanks for the support!

brunogermain, 2020-04-06:
> > The question is: does disabling I2C0 via SSDT and removing pci8086,9d60 (to prevent KP) solve the trackpad freeze issue?
> 
> @brunogermain : try, test and report back?

It keeps freezing. Debug attached

[debug_11465.zip](https://github.com/alexandred/VoodooI2C/files/4440331/debug_11465.zip)

[5 more replies]