VoodooI2C/VoodooI2C #185: swipes for the fifth time with a single finger, it won't work for that particular time

jinmu333, 2019-04-29:
swipes for the fifth time with a single finger, it won't work for that particular time
both GPIO and Polling

[bug.mp4.zip](https://github.com/alexandred/VoodooI2C/files/3128307/bug.mp4.zip)

kirainmoe, 2019-05-03:
Same problem here. Every five movement with a single finger, the fifth movement is ignored.  
I'm using VoodooI2C and VoodooI2CHID 2.1.6. My laptop is CoffeeLake.

jinmu333, 2019-05-03:
> Same problem here. Every five movement with a single finger, the fifth movement is ignored.
> I'm using VoodooI2C and VoodooI2CHID 2.1.6. My laptop is CoffeeLake.
> I'm wondering if this problem is related to GPIO interrupt as I'm using polling mode XD

My laptop is CoffeeLake too

jinmu333, 2019-05-03:
[bug.zip](https://github.com/alexandred/VoodooI2C/files/3140713/bug.zip)
And I find something new about the fifth movement

jinmu333, 2019-05-03:
> Same problem here. Every five movement with a single finger, the fifth movement is ignored.
> I'm using VoodooI2C and VoodooI2CHID 2.1.6. My laptop is CoffeeLake.
> I'm wondering if this problem is related to GPIO interrupt as I'm using polling mode XD
> 
> (Edit: I guess that this problem is suspected to happen on specific models? I have found that my laptop and jinmu333's laptop are all from an ODM named Tongfang.)

maybe, but latest windows and linux is ok

pexcn, 2019-05-07:
Same problem, CPU is CoffeeLake, i7-8750H.

jinmu333, 2019-05-08:
> Same problem, CPU is CoffeeLake, i7-8750H.

My CPU is CoffeeLake, i7-8300H.

Teufeuleu, 2019-08-09:
Same problem here, on a TongFang laptop too (GK5CN6Z, i7 8750H, macOS 10.14.6).

@ben9923 thanks for your answer. The problem is still here even without Karabiner, and I never experienced this issue on Linux (I had Manjaro then PopOS installed before macOS).
However I remind another issue with this laptop's trackpad working on Linux, [tracked and solved here](https://bugzilla.kernel.org/show_bug.cgi?id=200787). Maybe it can help.

kirainmoe, 2019-08-09:
We have tongfang GK5CN5X/GK5CN6X(are aliases of GK5CN5Z/GK5CN6Z) with the same problem. My touchpad works on Arch Linux.

ben9923, 2019-08-09:
> Same problem here, on a TongFang laptop too (GK5CN6Z, i7 8750H, macOS 10.14.6).
>
> @ben9923 thanks for your answer. The problem is still here even without Karabiner, and I never experienced this issue [...]

This Linux issue seems to be GPIO related. I believe the necessary fixes are already implemented in VoodooGPIO. Anyway, @jinmu333 verified it's also happening with polling mode, so I guess it's not the same issue as Linux. Description is also different.

jinmu333, 2019-08-10:
> This Linux issue seems to be GPIO related. I believe the necessary fixes are already implemented in VoodooGPIO. Anyway, @jinmu333 verified it's also happening with polling mode, so I guess it's not [...]

I  think it  is a bug of muti-touch.Swipes for the fifth time with a single finger, it won't work,

Goshin, 2019-12-18:
Same problem. Is there any update or workaround for it now?

Goshin, 2019-12-21:
After a night of debugging...

A quick fix here,
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)

kirainmoe, 2019-12-21:
> After a night of debugging...
> 
> A quick fix here,
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)

It works!

Vnzen, 2019-12-21:
> After a night of debugging...
> 
> A quick fix here,
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)

thanks,It works!

ben9923, 2019-12-21:
Thank you for the effort @Goshin !
Would be great if you open a PR once you think the fix is better.

Goshin, 2019-12-21:
Check out my patch commit ( https://github.com/Goshin/VoodooI2CHID/commit/d729e26a124432d7fdc1758eb446c118a06c34c9 ) here if you are interested.

---

It is something to do with the driver I am using, `VoodooI2CPrecisionTouchpadHIDEventDriver`.

In `VoodooI2CHID/VoodooI2CMultitouchHIDEventDriver.cpp`, 
method `VoodooI2CMultitouchHIDEventDriver::handleDigitizerReport(AbsoluteTime timestamp, UInt32 report_id)`, since https://github.com/alexandred/VoodooI2CHID/commit/d4ffd7609d92aec5003390086f5714272f932bc9 ,

```cpp
    UInt8 finger_count = digitiser.fingers->getCount();
    
    if (finger_count) {
        // Check if we are sending the report to the right wrapper, 99% of the time, `digitiser.current_report - 1` will
        // be the correct index but in rare circumstances, it won't be so we should ensure we have the right index
    
        if (!wrapper->first_identifier)
            return;
    
        UInt8 first_identifier = wrapper->first_identifier->getValue() ? wrapper->first_identifier->getValue() : 0;
    
        UInt8 actual_index = static_cast<int>(roundUp(first_identifier + 1, finger_count)/finger_count) - 1;
    
        if (actual_index != digitiser.current_report - 1) {
            wrapper = OSDynamicCast(VoodooI2CHIDTransducerWrapper, digitiser.wrappers->getObject(actual_index));
            if (!wrapper)
                return;
        }
    }
```

I am not 100% sure, but the code here seems to attempt to correct the wrapper index when the `first_identifier` changed.

I didn't figure out what exactly are the `first_identifier` and `finger_count` for. 
In my case, `finger_count` is always 4, and `first_identifier` is zero for the 4/5 times of single-finger touches, then changes to 4 for the fifth time touching, leading `actual_index` to be 1 instead of 0. 

I think it might be a model-specific problem. Though a simple revert just works well for me, this is only a simple fix and we still need to dig into it.

ben9923, 2019-12-21:
@alexandred Any idea about that code?

1129921824, 2019-12-21:
> After a night of debugging...
> 
> A quick fix here,
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)
It will not boot properly，Is there any configuration required？My laptap is same to @jinmu333

Goshin, 2019-12-21:
> > After a night of debugging...
> > A quick fix here,
> > [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)
> 
> It will not boot properly，Is there any configuration required？My laptap is same to @jinmu333

No additional config is required. And I think the patch is nothing to do with the core of VoodooI2C.

You may want to check whether the latest VoodooI2C 2.3 works on your laptop.

Teufeuleu, 2019-12-21:
> 
> 
> After a night of debugging...
> 
> A quick fix here,
> ~Archive.zip~
> [VoodooI2C.zip](https://github.com/Goshin/VoodooI2CHID/releases/download/fix%23185/VoodooI2C.zip)

Does not work for me either. It boots but the TP is not working at all. I have the GK5CN6Z (i7 8750H). Also tried the last VoodooI2C 2.3 update, it works but still with the 5fth touch issue.

Goshin, 2019-12-21:
So it seems not compatible with your touchpad.

Though I've only got very limited knowledge for the above snippet's function, you can try to use this debugging version [VoodooI2C-debug.zip](https://github.com/alexandred/VoodooI2C/files/3991744/VoodooI2C-debug.zip), which will do some log when you touch. Trim the log and attach here, for both good and bad touch, and let's see what can we do with it.

You can check out the log in 5s in your terminal with
```bash
log show -last 5m | grep "Voodoo"
```

1129921824, 2019-12-22:
> No additional config is required. And I think the patch is nothing to do with the core of VoodooI2C.
>
> You may want to check whether the latest VoodooI2C 2.3 works on your laptop.

latest VoodooI2C 2.3 works properly but still with the 5fth touch issue.

1129921824, 2019-12-22:
> > > > > After a night of debugging...
> > > > > A quick fix here,
> > > > > [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)
> > > > 
> > > > 
> > > > It will not boot properly，Is there any configuration required？My laptap is same to @jinmu333
> > > 
> > > 
> > > No additional config is required. And I think the patch is nothing to do with the core of VoodooI2C.
> > > You may want to check whether the latest VoodooI2C 2.3 works on your laptop.
> > 
> > 
> > latest VoodooI2C 2.3 works properly but still with the 5fth touch issue.It also will not boot properly.
> 
> Nothing I can do about it, sry.
> 
> You may try to build it by yourself, or attach some log here and see if anyone can help you.

thank u ，Maybe it's my computer

1129921824, 2020-01-04:
> Nothing I can do about it, sry.
>
> You may try to build it by yourself, or attach some log here and see if anyone can help you.

可以给我一下你的联系方式吗？

kirainmoe, 2020-01-04:
> > > > > > After a night of debugging...
> > > > > > A quick fix here,
> > > > > > [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990872/Archive.zip)
> > > > > 
> > > > > 
> > > > > It will not boot properly，Is there any configuration required？My laptap is same to @jinmu333
> > > > 
> > > > 
> > > > No additional config is required. And I think the patch is nothing to do with the core of VoodooI2C.
> > > > You may want to check whether the latest VoodooI2C 2.3 works on your laptop.
> > > 
> > > 
> > > latest VoodooI2C 2.3 works properly but still with the 5fth touch issue.It also will not boot properly.
> > 
> > 
> > Nothing I can do about it, sry.
> > You may try to build it by yourself, or attach some log here and see if anyone can help you.
> 
> thank u ，Maybe it's my computer's issue

Is your macOS system version 10.14.x? If so, his build is not compatible. Try the build in [here](https://github.com/kirainmoe/hasee-tongfang-macos/tree/master/kexts/Other)

kirainmoe, 2020-01-04:
[VoodooI2C-tongfang-fix-10.14.zip](https://github.com/alexandred/VoodooI2C/files/4021833/VoodooI2C-tongfang-fix-10.14.zip)

1129921824, 2020-01-04:
> [VoodooI2C-tongfang-fix-10.14.zip](https://github.com/alexandred/VoodooI2C/files/4021833/VoodooI2C-tongfang-fix-10.14.zip)

测试可用，谢谢大佬

kirainmoe, 2020-01-04:
Touchpad settings should be normal. Did you patch your battery status correctly? Or try `sudo kextcache -i /`.

jinmu333, 2020-01-04:
> Touchpad settings should be normal. Did you patch your battery status correctly? Or try `sudo kextcache -i /`.

Good Job!

Teufeuleu, 2020-01-04:
> Does not work for me either. It boots but the TP is not working at all. I have the GK5CN6Z (i7 8750H). Also tried the last VoodooI2C 2.3 update, it works but still with the 5fth touch issue.

> 
> 
> So it seems not compatible with your touchpad.
> 
> Though I've only got very limited knowledge for the above snippet's function, you can try to use this debugging version [VoodooI2C-debug.zip](https://github.com/alexandred/VoodooI2C/files/3991744/VoodooI2C-debug.zip), which will do some log when you touch. Trim the log and attach here, for both good and bad touch, and let's see what can we do with it.
> 
> You can check out the log in 5s in your terminal with
> 
> ```shell
> log show -last 5s | grep "Voodoo"
> ```

> 
> 
> [VoodooI2C-tongfang-fix-10.14.zip](https://github.com/alexandred/VoodooI2C/files/4021833/VoodooI2C-tongfang-fix-10.14.zip)

Indeed, the first fiw did not work because I'm still on 10.14.
The build from @kirainmoe works perfectly for me. Thanks!

Goshin, 2020-04-08:
The latest 2.4 release should fix this. Can anyone test it?

kirainmoe, 2020-04-09:
confirmed that 2.4 fixed this problem.

kprinssu, 2020-04-09:
Great, please reopen the issue if it happens.

Thanks @Goshin for the fix and @kirainmoe for testing.

closed 2020-04-09 by kprinssu