VoodooI2C/VoodooI2C #188: the multitouch gesture on the touchscreen conflicts with the stylus pen input.

Jamesxxx1997, 2019-05-09:
- machine model : Miix 520-121KB (Type 81CG）
- CPU model ： i7-8550u
- OS version : 10.14.4
- VoodooI2C version : 2.1.6
- bug description : the multitouch gesture on the touchscreen conflicts with the stylus pen input. **The stylus pen input doesn't conflict with single touch on touchscreen ; However , multitouch gesture on touchscreen will make conflict with stylus pen input** , making the touchscreen cannot sense stylus pen until single touch on touchscreen.

files
video.zip : https://1drv.ms/u/s!Asfvb_2uj9y7mXCw3-db8IYwq3CS
debug file : https://1drv.ms/u/s!Asfvb_2uj9y7mW_KpJcEHompjQRi

blankmac, 2019-05-19:
This isn't a bug really, but a byproduct of design choice / compromise.  Because the MT2 is a touchpad device, we only use the simulator for multitouch gesture input (ie - anything with 2+ fingers of input) and write single touch input ( stylus or single finger input) directly.  By attempting to use multitouch and stylus input simultaneously, you are locking the pointer in an unusable state.  By then touching with a single finger, the fingerLift event is correcting the problem.  Clicking with your touchpad or mouse will probably accomplish the same thing.  I can infrequently reproduce the issue you describe but irregardless, the solution is simple --  Don't try to use multitouch gestures and the stylus input simultaneously, as the touchscreen input is currently written, it won't work.  Hope that clarifies things for you.

Jamesxxx1997, 2019-05-19:
thanks! Your explanation is detailed enough XD

I've tried solve the conflict by left click on my Magic Trackpad 2 and Magic Mouse 2 , but it can't :(

Sometimes **after** the multitouch on touchscreen , there is a chance that stylus won't work too.(video:https://1drv.ms/u/s!Asfvb_2uj9y7mXSeZ2VVWRKap3EP)

Will the multitouch conflict also happen on usb-hid touchscreen  , with or without touch-base UPDD driver installed?

Thanks a lot!

blankmac, 2019-05-19:
Hmmm.. ok, not sure if there is a solution for that, but I will look into it.

I've only used UPDD with VoodooI2C (which will suffer from the issue possibly) so I can't comment with any certainty about a USB touchscreen.  Sorry.

Jamesxxx1997, 2019-05-20:
Hi , @blankmac , if I want to ask about usb-hid touchscreen tablet , do you know what forum should I post my question?
I've ask touch-base , and the guy Dave seems don't know the answer to my question ... or can I ask about this question in your gitter chat room ?
Thank you very much ~~~

blankmac, 2019-05-20:
Touch-base allows you to test their driver for free, why don’t you just try it?

Jamesxxx1997, 2019-05-20:
Okay , thanks...
For I don't have that laptop(yoga x1 3rd) currently...
Will buy one and test it with updd...

Jamesxxx1997, 2019-07-18:
@blankmac I've bought thinkpad x1 yoga 3rd , and it use **usb-hid** touchscreen.
With voodooi2c&voodooi2chid , the multitouch gesture works!
Moreover , I've found that **the multitouch conflict with stylus pen doesn't happen on my laptop** ; 
In fact , **if I put finger and stylus pen on touchscreen , the touchscreen will sense **both** finger and stylus pen , so the mouse pointer will run back and forth between them.(Without using voodooi2c , the mouse pointer have same "run back and forth" behavior)**

Here is my video recording  : https://youtu.be/afTku96HaDA
And also upload my ioreg (usb HS10 is the pen and multitouch sensor) , from touch-base people said  the pen and multitouch sensor are on the same interface (1), different end points, 81 (touch) and 82 (pen).
Hope this information is beneficial to solve the conflict problem in i2c-hid touchscreen.
Thank you very much~~~

[Thinkpad x1 Yoga 3rd.ioreg.zip](https://github.com/alexandred/VoodooI2C/files/3405670/Thinkpad.x1.Yoga.3rd.ioreg.zip)

Jamesxxx1997, 2019-08-23:
Hi @blankmac , by default , the touchscreen seems to have simple palm rejection function via voodooi2c. If I touch the screen with pen and single finger simultaneously , the touchscreen will only sense the pen input. 
Is it possible to make the touchscreen to sense both pen and single finger simultaneously ? I wonder if this change might solve the multitouch confilct issue , thank you very much~~~

kprinssu, 2020-05-15:
Please use the latest version of VoodooI2C as it should fix this issue. Please create a new issue if you experience any issues.

closed 2020-05-15 by kprinssu