VoodooI2C/VoodooI2C #189: VoodooI2C 2.1.6 - Invalid Signatures

JayMonkey, 2019-05-09:
Hi Guys,

Not sure if its already been posted but it seems that there is an issue with the signatures.

With VoodooI2C installed in /L/E and running 

`sudo kextcache -i /`

The following errors are seen ..

`Invalid signature -67050 for kext <OSKext 0x7fc4aa198020 [0x7fffa22b98f0]> { URL = "VoodooI2CHID.kext/ -- file:///Library/Extensions/", ID = "com.alexandred.VoodooI2CHID" }
Invalid signature -67050 for kext <OSKext 0x7fc4aa1c6ff0 [0x7fffa22b98f0]> { URL = "VoodooI2CServices.kext/ -- file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "com.alexandred.VoodooI2CServices" }
Invalid signature -67050 for kext <OSKext 0x7fc4aa1a07c0 [0x7fffa22b98f0]> { URL = "VoodooGPIO.kext/ -- file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "org.coolstar.VoodooGPIO" }
Invalid signature -67050 for kext <OSKext 0x7fc4aa18f8c0 [0x7fffa22b98f0]> { URL = "VoodooI2C.kext/ -- file:///Library/Extensions/", ID = "com.alexandred.VoodooI2C" }
`

The kexts still load ok and work fine though ...

Normally running 

`sudo kextcache -i /`

One would see something like 👎 

`Kext with invalid signatured (-XXXXX) allowed: `

Not sure if this needs looking at or not but thought i would report it as it does seem to me to be unusual.
 
Cheers
Jay

tprimus83, 2019-05-09:
https://www.imore.com/how-turn-system-integrity-protection-macos

JayMonkey, 2019-05-09:
@ tprimus83,

I currently have 

`<key>RtVariables</key>
    <dict>
        <key>BooterConfig</key>
        <string>0x28</string>
        <key>CsrActiveConfig</key>
        <string>0x67</string>
        .
        .
        .
    </dict>`

In my config.plist .. which works fine for all other 3rd party kexts ... I know i can fully disable SIP which would resolve the issue with VoodooI2C but seems inappropriate just for one kext.

Like i said, VoodooI2C is working fine as things are, just thought i'd mention this as it should be a relatively easy fix in the projects Xcode dev id's definitions.

Cheers
Jay

hrytskivr, 2019-05-09:
I had this errors too.
Had to rebuild from source by myself, then it worked.

williambj1, 2019-05-10:
I don't think Apple will provide signs for Hackintosh Kexts😀
https://developer.apple.com/library/archive/documentation/Security/Conceptual/System_Integrity_Protection_Guide/KernelExtensions/KernelExtensions.html

gongzhenjeef, 2019-05-15:
Do you mean just keep them as "Not Signed"?

gongzhenjeef, 2019-05-15:
I tried to get rid of the extra warning "Invalid signature" by following modification to *.xcodeproj/project.pbxproj
1. changing  ProvisioningStyle = Automatic to Manual
2. remove every "CODE_SIGN_IDENTITY" line
Since there's no selection for no signing certificate in Xcode user interface after disabling "Automatically manage signing"
Then I compiled again and the output extensions are not signed. The size is slightly reduced for no embedded signature.

Snikii, 2019-05-17:
@williambj1 the point is not to get a signed kext, it's just that only voodooI2C kexts are giving the following error ok kextcache output.
All other kexts are unsigned but they don't show that message:
`Kext with invalid signatured (-67062) allowed: <OSKext 0x7f96ebf1b570 [0x7fff96ae58f0]> { URL = "AppleALC.kext/ -- file:///Library/Extensions/", ID = "as.vit9696.AppleALC" }
Kext with invalid signatured (-67062) allowed: <OSKext 0x7f96ebf19c60 [0x7fff96ae58f0]> { URL = "AirportBrcmFixup.kext/ -- file:///Library/Extensions/", ID = "as.lvs1974.AirportBrcmFixup" }
Kext with invalid signatured (-67062) allowed: <OSKext 0x7f96ebf14540 [0x7fff96ae58f0]> { URL = "ACPIBatteryManager.kext/ -- file:///Library/Extensions/", ID = "org.rehabman.driver.AppleSmartBatteryManager" }
Invalid signature -67050 for kext <OSKext 0x7f96ec0ea950 [0x7fff96ae58f0]> { URL = "VoodooI2CServices.kext/ -- file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "com.alexandred.VoodooI2CServices" }
Invalid signature -67050 for kext <OSKext 0x7f96ec0d2150 [0x7fff96ae58f0]> { URL = "VoodooGPIO.kext/ -- file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/", ID = "org.coolstar.VoodooGPIO" }
Invalid signature -67050 for kext <OSKext 0x7f96ec0d42f0 [0x7fff96ae58f0]> { URL = "VoodooI2C.kext/ -- file:///Library/Extensions/", ID = "com.alexandred.VoodooI2C" }
Invalid signature -67050 for kext <OSKext 0x7f96ec095180 [0x7fff96ae58f0]> { URL = "VoodooI2CHID.kext/ -- file:///Library/Extensions/", ID = "com.alexandred.VoodooI2CHID" }
Invalid signature -67030 for kext <OSKext 0x7f96ebf8d8c0 [0x7fff96ae58f0]> { URL = "VoodooI2CSynaptics.kext/ -- file:///Library/Extensions/", ID = "com.alexandred.VoodooI2CSynaptics" }
KernelCache ID: B73E15B3338421E5C32AF728445FCCA8
Kext with invalid signatured (-67062) allowed: <OSKext 0x7faa8362b630 [0x7fff96ae58f0]> { URL = "ACPIBatteryManager.kext/ -- file:///Library/Extensions/", ID = "org.rehabman.driver.AppleSmartBatteryManager" }
Kext with invalid signatured (-67062) allowed: <OSKext 0x7faa8362c800 [0x7fff96ae58f0]> { URL = "Lilu.kext/ -- file:///Library/Extensions/", ID = "as.vit9696.Lilu" }
Kext with invalid signatured (-67062) allowed: <OSKext 0x7faa8362da60 [0x7fff96ae58f0]> { URL = "AirportBrcmFixup.kext/ -- file:///Library/Extensions/", ID = "as.lvs1974.AirportBrcmFixup" }
Kext with invalid signatured (-67062) allowed: <OSKext 0x7faa836321d0 [0x7fff96ae58f0]> { URL = "AppleALC.kext/ -- file:///Library/Extensions/", ID = "as.vit9696.AppleALC" }`

blankmac, 2019-05-19:
If this bothers you guys, then go to the code signing section of the project, select the "Code Signing Identity" key entry and hit backspace to delete whatever was selected.  The kext will then show as 'unsigned' and all your errors will be the same.  :-D

vectorsigma72, 2019-12-26:
Guys, all unsigned kext (with an executable) will give you this warning. Of course if you see those warns is because is the only one installed in /S/L/E (or /L/E). You will never see that for kexts injected by the bootloader.

fewtarius, 2019-12-26:
The proper solution here is to stop putting your kexts in /L/E or /S/L/E, and to also not disable SIP.

ben9923, 2020-04-05:
Keeping this one opened for satellite kexts, VoodooPIO.

kprinssu, 2020-05-15:
This should be fixed with 2.4.0 as the kexts are no longer signed. Please retest and reply back here.

kprinssu, 2020-05-21:
Please reopen if you encounter the issue again.

commit 388e63c2 (2020-04-05): Use Xcode 11.3 recommended settings

This should change code signing settings, and fix #189 for core kext.
closed 2020-05-21 by kprinssu