VoodooI2C/VoodooI2C #190: ELAN1203 problems

animator013, 2019-05-11:
Hi there,

installed latest kext version 2.1.6 + sattelite kext VoodooI2CCHID. My ELAN1203 works but there are these quirks:

- left button and right button works only when i am touching the trackpad and right button does the same thing as left button :)

I've made dsdt patching according to guide.

Can you help me with this ? What do you need from me to provide ?

animator013, 2019-05-11:
Ah, it seems i did it wrong.

Tried to do it again on ETPD device. After inserting GPIO pin list to 0x47 as my APIC is 0x5f i get Busy timeout ETPD in logs...

What to do ?

ben9923, 2019-05-12:
Notice GPIO interrupts are not yet supported in CFL+. Pin calculation is also different.
Please ask for help in the Gitter chat, attach your files as noted [here](https://voodooi2c.github.io/#Troubleshooting) and [here](https://voodooi2c.github.io/#Common%20Errors).

animator013, 2019-05-12:
Ok, posted there. Thank you, hope it can be solved.

williambj1, 2019-05-12:
Same Issue:
Physical buttons only work when there is at least one finger on the trackpad.
When using physical buttons to highlight words, it behaves like continuing for a few seconds then it suddenly breaks which makes it impossible to use.

animator013, 2019-05-12:
Exactly the same thing.

williambj1, 2019-05-19:
### Video

[image]

blankmac, 2019-05-19:
The button behavior isn't entirely unexpected.  Think of it this way, the MT2 that we are simulating is an integrated one-button touchpad device.  You can't actuate the button without being in contact with it and since it is only one button, there are no left/right click methods *other* than the options contained in the trackpad system preference pane.  Unfortunately, this doesn't translate well to devices that have separate physical buttons.  I'm not sure what the solution would be other than to address specific devices with the problem.

As for the selection behavior, try this so we can see if the two issues are related -- with one finger, move the cursor into position to select some text, leave that finger in place as an anchor, click/hold your physical button, and then use an additional finger to drag/select while leaving the first finger in place and see if it's still the same.  (unless that's what you're already doing, of course)

LeeBinder, 2019-05-19:
These issues (and one more) are the reason I'm still using v.2.0.3 and am running High Sierra (because 2.0.3 is not Mojave compatible). To me the switch to MT2 was a HUGE regression.

kprinssu, 2020-05-15:
Please note that this will probably will not be supported due to use interfacing directly with macOS (original Apple hardware do not have physical buttons and we will not support it in VoodooI2C). Please contact the VoodooInput devs and get support in the upstream repo.

If you have any issues related to VoodooI2C itself, please create a new ticket.

closed 2020-05-15 by kprinssu

williambj1, 2020-05-16:
This issue was solved by Goshin in v2.4.1, thumbs up!