VoodooI2C/VoodooI2C #192: Voodooi2c problem with latest releases.

Mateo1234454545, 2019-05-31:
I have a problem with releases later than 2.1.4.

With 2.1.4 all working fine , but after that (it happens also with latest 2.2) pointer freezes randomly for about 1sec and then starts again. This happens all the time making it impossible to use. Both on HS and Mojave.

Seems like  another device interfering with trackpad?

Can you take a look at my dump files? One with all working (2.1.4) and one (2.2) with the probem described.

[2.1.4.zip](https://github.com/alexandred/VoodooI2C/files/3239829/2.1.4.zip)
[2.2.zip](https://github.com/alexandred/VoodooI2C/files/3239830/2.2.zip)

mefflecakes, 2019-08-12:
I'm also having this issue. Did you get any resolution?

Mateo1234454545, 2019-08-12:
Unfortunately no.

kecinzer, 2019-08-12:
I get VoodooI2C to work today and I have sams issue. Random freezes for 1 second. I have also version 2.2.

mefflecakes, 2019-08-12:
I tried going back to 2.1.4 and I even have problems with that version... Perhaps I'm not clearing it up properly...

kecinzer, 2019-08-15:
Version 2.1.3 solved this issue, but the I have problems with boot.

ben9923, 2019-08-24:
Seems like it's caused by https://github.com/coolstar/VoodooGPIO/commit/9e0b3e2e6a64c5d057cad16d94df32f962c45292.

Found by northernlights at tonymacx86:
https://www.tonymacx86.com/threads/voodooi2c-help-and-support.243378/post-1997654

Mateo1234454545, 2019-08-31:
Hope a fixed version be uploaded soon.

LeeBinder, 2019-09-13:
Here's the symptoms on my Asus: after any combined keyboard operation incl. a modifier key (cmd+ c copy, cmd + s save, cmd + x cut etc.), the mouse pointer is "paralyzed"/ like in stasis for maybe half a second. @ben9923 's [08-23 kexts](https://www.tonymacx86.com/threads/voodooi2c-help-and-support.243378/page-126#post-1997676) (neither 'no-lag' nor 'lag) change anything.

regae, 2019-10-27:
try to edit VoodooI2CHID.kext info.plist, change QuietTimeAfterTyping to 0

LeeBinder, 2019-10-27:
@regae WHHHAAAAAT ?? THAT'S IT 👍 💯 🥇  !!

How on the planet do you know this? I've even been communicating with @alexandred about this.

Why the heck is this default to being with? So this means that *_everybody_* using current VoodooI2CHID has been having these lags? How can one work like that.

I totally completely utterly don't get it, this is a head-banger.

@alexandred ??

Mateo1234454545, 2019-10-28:
That's the default timeout at Macos.
Maybe if we change it to 0 we will have problems , like battery drain or anything else?

@LeeBinder , have you tried that ? What are your impressions?

LeeBinder, 2019-10-28:
@Mateo1234454545 
1. that value in VoodooI2CHID is naturally NOT the default timeout in/of macOS but the timeout VoodooI2CHID regulates .. grab a current Mac(book) with a current macOS and you will sure NOT encounter any such lag. Otherwise this topic would not exist to begin with,  I dare say

2. with what logic could changing this timeout value alter battery life? What this function does is store the keystroke performed while the function is on wait for the time specified in ms, then perform it anyway. No delay can ever save energy, and esp. not one of 500ms (default value). In contrast, storing event data costs energy (esp. our energy waiting 500ms all the time for  
things to happen).

3. Sure thing I've applied this right away with a positive result, therefore my reaction. I figured that'd be obvious and self explanatory.. but I'll add that to my above posting for hopefully anybody to understand.

I still don't dig why this odd "feature" or rather (not just to me but apparently also to others) BUG was introduced from 2.0.3 to 2.1.4+, and everybody seems to be putting up with it. Or maybe they don't, because in some peep's h'tosh configs, this setting might be ignored and result in 0 to begin with? Odd no matter how.

andrewwyatt, 2019-10-28:
It looks like it might be some sort of palm rejection functionality.

https://github.com/alexandred/VoodooI2CHID/blob/master/VoodooI2CHID/VoodooI2CMultitouchHIDEventDriver.cpp#L762
and
https://github.com/alexandred/VoodooI2CHID/blob/master/VoodooI2CHID/VoodooI2CMultitouchHIDEventDriver.cpp#L100

LeeBinder, 2019-10-28:
if so then it does not make any practical sense because 500ms is too little during a full typing "spree", and obviously too annoyingly much for quick combined keyboard commands involving CMD / ALT/ etc.

regae, 2019-10-28:
to disable palm rejection, disable RM,deliverNotification in "VoodooI2CHIDDevice Precision Touchpad HID Event Driver"
its working without setting QuietTypeAfterTyping to 0

LeeBinder, 2019-10-28:
regae, do you know if there is any tangible advantage (like higher reliability or alike) of toggling "RM,deliverNotification" to 'No' over setting "QuietTypeAfterTyping" to '0', or are they simply two paths of equal value to achieve basically the same?

LeeBinder, 2019-10-29:
I think that neither, setting "QuietTimeAfterTyping" to '0' or "RM,deliverNotification" to 'No' as suggested are optimal because both basically deactivate palm rejection, which is a good feature, needless to explain. I am playing around with values betw. 50 and 100 right now. It's all about finding the value according to one's workflow-speed. It should be as high as possible without causing the observed unergonomic "stickiness" (lags/ delays/ freezes).

As a side note: I was experiencing that esp. High Sierra is very picky about that kext's permissions and how it is integrated into the kext cache. Our classic commands (and also KCPM Utility Pro) have been failing me, resulting in a KP on reboot after each change, then having to boot with -x. What DOES always work after each change is [Kext Updater](https://bitbucket.org/profdrluigi/kextupdater/downloads/): Tools/ Rebuild kext cache & repair perms[..], reboot

ben9923, 2019-11-01:
Hmm, was the quiet time actually related? It should only have effect while typing, of course.
Nothing related to GPIO, are you sure you're talking about the same problem?
@Mateo1234454545 Can you verify it's really related, as you opened the issue?

Anyway, default quiet time was changed to 0.1s, as 0.5s is really too much.

Mateo1234454545, 2019-11-01:
> Hmm, was the quiet time actually related? It should only have effect while typing, of course.
> Nothing related to GPIO, are you sure you're talking about the same problem?
> @Mateo1234454545 Can you [...]

No , it is only releted to typing as you said , at least for my laptop.
Only solution is described at #213 by Erictoby.

ben9923, 2019-11-01:
> No , it is only releted to typing as you said , at least for my laptop.
> Only solution is described at #213 by Erictoby.

I see, thank you for verifying :)
Can you try latest master of all kexts? (VoodooI2C, VoodooI2CHID, VoodooGPIO)?
There are pull requests by @al3xtjames that were just merged, and I suspect they can have effect on your issue too.

Mateo1234454545, 2019-11-01:
@ben9923 , I could but I don't have XCode and I can't compile them.
Maybe you could upload a release?

andrewwyatt, 2019-11-01:
> Hmm, was the quiet time actually related? It should only have effect while typing, of course.
> Nothing related to GPIO, are you sure you're talking about the same problem?
> @Mateo1234454545 Can you [...]

Maybe something to investigate, but I noticed a definitive change to tap & drag once it was set to 0.

LeeBinder, 2019-11-02:
I just installed [Erictoby's VoodooI2C v.2.2 + Beta M](https://github.com/Erictoby/VoodooI2C/releases/) latest from 2019-Oct-11 edited with QuietTimeAfterTyping "100". So far so good. I really appreciate the collaboration and sharing of experiences here!

Mateo1234454545, 2019-11-03:
> I see, thank you for verifying :)
> Can you try latest master of all kexts? (VoodooI2C, VoodooI2CHID, VoodooGPIO)?
> There are pull requests by @al3xtjames that were just merged, and I suspect they can [...]

No , unfortunately they don't.

LeeBinder, 2019-11-03:
@Mateo1234454545 can you please post a link here to compiled binaries of al3xtjames latest kexts? I looked thoroughly but could not find any. I want to see if there is any difference with my trackpad. Thanks.

Mateo1234454545, 2019-11-03:
@LeeBinder , https://files.gitter.im/onemanosx/uQJG/VoodooI2C-02112019.zip

onemanosx, 2019-11-03:
I have to say, my experience with the most recent compile is simply the best! In previous compilation, I had to deal with cursor pointer randomly jumps off and this makes typing a chore. But, now all of that is gone! *thumbs up* @al3xtjames

LeeBinder, 2019-11-03:
Thanks to all of you, but for me on my Asus Vivobbok S15 w. ELAN1300, dead pointer with the kexts (VoodooI2C & VoodooI2CHID) from inside VoodooI2C-02112019.zip.

Issue remains the same thing with:
- onemanosx's VoodooI2C.kext + original VoodooI2CHID.kext
- vice versa
- polling or GPIO-interrupts mode via clover config.plist + respective SSDTs

Now running again on [Erictoby's compile](https://github.com/Erictoby/VoodooI2C/releases/tag/v2.2-beta.M).

onemanosx, 2019-11-03:
> Thanks to all of you, but for me on my Asus Vivobbok S15 w. ELAN1300, dead pointer with the kexts (VoodooI2C & VoodooI2CHID) from inside VoodooI2C-02112019.zip.
>
> Issue remains the same thing with: [...]

How about using VoodooI2C-02112019.zip's VoodooI2C + VoodooI2CHID instead of mix and match between builds?

LeeBinder, 2019-11-03:
read again what I wrote - that's what I did to begin with, needless to say .. ;)

LeeBinder, 2019-11-06:
@onemanosx which sdk did you compile with in Xcode: only for Catalina? Which macOS are you running?

onemanosx, 2019-11-07:
> @onemanosx which sdk did you compile with in Xcode: only for Catalina? Which macOS are you running?

compiled in Catalina

LeeBinder, 2019-11-07:
al3xtjames's compiled by onemanOSX/VoodooI2C.kext/Contents/Info.plist:
LSMinimumSystemVersion: 10.11

I very much doubt you compiled with your Xode running in Catalina with ALL SDKs active from 10.11-10.15... this MinimumSystemVersion value rather seems to me like a bug in Xcode..

I don't have Catalina installed yet. Did you or @Mateo1234454545 ever try your compile's kexts in Mojave or even High Sierra?

onemanosx, 2019-11-09:
[I2C - 10112019.zip](https://github.com/alexandred/VoodooI2C/files/3827834/I2C.-.10112019.zip)

Anyone up to try on Mojave or High Sierra is welcome.

Mateo1234454545, 2019-11-10:
@onemanosx  , version I2C.-.10112019 doesn't load on my laptop at Catalina.
Maybe a GPIO pin problem?
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/3827930/Archive.zip)
[Geni2c.pdf](https://github.com/alexandred/VoodooI2C/files/3827931/Geni2c.pdf)

LeeBinder, 2019-11-10:
@Mateo1234454545 onemanosx *clearly* wrote that [I2C - 10112019.zip](https://github.com/alexandred/VoodooI2C/files/3827834/I2C.-.10112019.zip) is explicitly meant for Mojave and High Sierra, in contrast to the compile before which he made for Catalina. Why on the planet would you install 10.13/10.14 kexts in 10.15 to begin with, when you already had the 10.15 kexts working for you ??????? Totally beats me, mate ....

@onemanosx : thanks for your effort compiling, very much appreciated! Will try very soon in both, HS & M. Good news: your Catalina compile does work mighty fine for me in 10.15.1 which I just installed today.

Will report back once I tested.

LeeBinder, 2019-11-10:
@onemanosx : I2C - 10112019 kexts neither working for me in HS nor in M: pointer doesn't move at all.

BTW, I just now see that the link behind  I2C - 10112019.zip points to alexandred/VoodooI2C/files/ and not to you as before. So what's the story with this compile: made by you based on al3xtjames's fork, or made by (who) based on (alexandred master trunk or someone's branch or fork)?

Mateo1234454545, 2019-11-10:
> @Mateo1234454545 onemanosx _clearly_ wrote that [I2C - 10112019.zip](https://github.com/alexandred/VoodooI2C/files/3827834/I2C.-.10112019.zip) is explicitly meant for Mojave and High Sierra, in [...]
Which one is for Catalina? I don't see it anywhere. Could you send me the link?

LeeBinder, 2019-11-10:
below link you yourself provided to onemanosx's compile

@onemanosx I had to drop [VoodooI2C-02112019](https://files.gitter.im/onemanosx/uQJG/VoodooI2C-02112019.zip) even for Catalina 15.1 because the kexts (or one of them) cause KP on waking from sleep.

I also tried the two relevant kexts from [I2C - 10112019.zip](https://github.com/alexandred/VoodooI2C/files/3827834/I2C.-.10112019.zip) in Catalina for the heck of it, but as expected they have no effect, being for Mojave and/ or High Sierra, as you wrote.

Again thanks for sharing, but  for now I'm back to Erictoby's compile for Mojave & Catalina. Would sure still enjoy to get as you fully rid of the random pointer jump offs..

kecinzer, 2020-04-08:
This issue still persist in version 2.4.

ben9923, 2020-04-29:
Can we close this one after 2.4.2 was released?

Mateo1234454545, 2020-04-29:
@ben9923 ,
yes you can.
Latest release is a masterpiece.

ben9923, 2020-04-29:
> @ben9923 ,
> yes you can.
> Latest release is a masterpiece.

Neat. Finally!

closed 2020-04-29 by ben9923

LeeBinder, 2020-04-29:
agree, masterpiece! You devs please pop the corks/ bust the caps 🥇 🥇 🥇 !!!

kecinzer, 2020-04-29:
Yes, agree. 2.4.2 is the best version yet. Touchpad is like on real MacBook :).