VoodooI2C/VoodooI2C #193: prefpanel not appear on macOS 10.15

johnnync13, 2019-06-04:
Voodooi2c and voodooi2csynaptics works but not appear prefpanel

gongzhenjeef, 2019-06-04:
You have provided so little information and Catalina is a beta version. Check if you have correct battery indicator. If the battery is not recognized, the prefpane won't show. SMCBatteryManager seems broken and try ACPIBatteryManager instead.

johnnync13, 2019-06-04:
@gongzhenjeef battery don´t show. 
I just saw one of the last commitments of virtualsmc fixing the battery

closed 2019-06-04 by johnnync13

tankren, 2019-09-21:
> You have provided so little information and Catalina is a beta version. Check if you have correct battery indicator. If the battery is not recognized, the prefpane won't show. SMCBatteryManager seems broken and try ACPIBatteryManager instead.

hello，电池能显示百分比，信息也能看到型号序号之类打，电池应该驱动了吧，kextstat能看到Voodoo的kext都加载了但是设置面板就是看不到触控板，望指点迷津，谢谢