VoodooI2C/VoodooI2C #198: Add palm rejection logic to touchpad satellites?

sunrez, 2019-07-11:
Hi,

First thanks for this amazing work, I realize this feature request is a "nice to have" when a lot of core functionality is still being worked on...

Request:  Add some kind of palm rejection logic to the touchpad multi-touch logic.

Right now there is a 'mask time' which basically prevents touchpad inputs from being registered within some time-window of typing on the keyboard, but this is a bit too coarse.  I notice with just my hands/palms resting on the keyboard false inputs are registered.

Thinking about how to implement this, usually a palm inputs would appear like 'blobs' on the left+right side of the touchpad, not discrete fingers/circles on touchpad.  One could do a search on the radius of the touch (> than a threashold) and see if it intersects a 'bounding box' that's defined by upper left/right areas.

sunrez, 2019-12-30:
Hi,

I have been thinking more about implementing this functionality and would like to take a stab at it.

Is there any diagnostics or way to visualize or display what the touchpad sees?  Something simple like a GUI window that shows dots of "touch" data?

kprinssu, 2020-05-15:
This is not supported and probably will never will be as we directly interface VoodooInput which in turn sends the events to macOS. Please contact the dev team over at VoodooInput if you wish to implement this.

closed 2020-05-15 by kprinssu

JerryI, 2023-08-04:
It looks like that in 2.6 it works, kinda strange a bit. But in general, it rejects.
Lenovo Yoga 520
[image]

kprinssu, 2023-08-04:
@JerryI Please do not post in a closed issue. As previously mentioned, this will never be supported.