VoodooI2C/VoodooI2C #201: Panic before start with any version

Edoardo001, 2019-07-27:
I use HID version, and every two attempt I've get a panic, reboot, and then boot fine. Attached the log with various panic(last 5 are related to this problem)
[debug_10714.zip](https://github.com/alexandred/VoodooI2C/files/3437687/debug_10714.zip)

Edoardo001, 2019-07-27:
Scuse me, here the correct report with only the panic concerned to that kext
[debug_13156 2.zip](https://github.com/alexandred/VoodooI2C/files/3437990/debug_13156.2.zip)

Erictoby, 2019-09-17:
This bug may be fixed with https://github.com/coolstar/VoodooGPIO/pull/7/commits/eb42516854b9cb5fbe7eb2ad071cbdf1da316425   It can cause unexpected results, such as causing kernel panic, crashing other kexts, or making the touchpad unresponsive sometimes.

idaima, 2019-10-09:
> This bug may be fixed with [coolstar/VoodooGPIO@eb42516](https://github.com/coolstar/VoodooGPIO/commit/eb42516854b9cb5fbe7eb2ad071cbdf1da316425) It can cause unexpected results, such as causing kernel panic, crashing other kexts, or making the touchpad unresponsive sometimes.

it's still not solve kernel panic problem , i'm so sad~

Anonymous UUID:       FFE5AC5F-EEE3-8F4F-7FD6-23BB10100C6E

Wed Oct  9 18:30:10 2019

*** Panic Report ***
panic(cpu 6 caller 0xffffff800b2dbadd): Kernel trap at 0xffffff800b8596b6, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000021da0000, CR4: 0x00000000003626e0
RAX: 0xffffff80402a7b80, RBX: 0xffffff80402a7d00, RCX: 0x0000000000000017, RDX: 0xffffff803fd93d00
RSP: 0xffffff81fe323da0, RBP: 0xffffff81fe323dc0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000006, R11: 0x0000000000000000
R12: 0xffffff7f8f1236c8, R13: 0x0000000000000000, R14: 0x0000000000000000, R15: 0x0000000000000017
RFL: 0x0000000000010202, RIP: 0xffffff800b8596b6, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x6, PL: 0, VF: 1

Backtrace (CPU 6), Frame : Return Address
0xffffff81fe323870 : 0xffffff800b1ae6ed mach_kernel : _handle_debugger_trap + 0x47d
0xffffff81fe3238c0 : 0xffffff800b2ea185 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81fe323900 : 0xffffff800b2db8ba mach_kernel : _kernel_trap + 0x50a
0xffffff81fe323970 : 0xffffff800b15bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81fe323990 : 0xffffff800b1ae107 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff81fe323ab0 : 0xffffff800b1adf53 mach_kernel : _panic + 0x63
0xffffff81fe323b20 : 0xffffff800b2dbadd mach_kernel : _kernel_trap + 0x72d
0xffffff81fe323c90 : 0xffffff800b15bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81fe323cb0 : 0xffffff800b8596b6 mach_kernel : __ZN22IOInterruptEventSource23normalInterruptOccurredEPvP9IOServicei + 0x56
0xffffff81fe323dc0 : 0xffffff7f8f11d403 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO32intel_gpio_community_irq_handlerEP15intel_communityPb + 0xb3
0xffffff81fe323e10 : 0xffffff7f8f11d705 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO22interruptOccurredGatedEv + 0x4d
0xffffff81fe323e50 : 0xffffff800b85ab85 mach_kernel : __ZN13IOCommandGate13attemptActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb5
0xffffff81fe323eb0 : 0xffffff7f8f11cf54 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO17InterruptOccurredEP8OSObjectP22IOInterruptEventSourcei + 0x40
0xffffff81fe323ed0 : 0xffffff800b858d8c mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x13c
0xffffff81fe323f30 : 0xffffff800b856e72 mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x1e2
0xffffff81fe323f70 : 0xffffff800b8563cc mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x2c
0xffffff81fe323fa0 : 0xffffff800b15b0ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[AE359004-5BA4-384A-95E4-7134700BB0BA]@0xffffff7f8f11a000->0xffffff7f8f130fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3E8E9870-2E41-33F6-ACBF-35DDC08F22B5]@0xffffff7f8c0c5000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 kext-dev-mode=1 -cdfon -igfxhdmidivs alcid=27 keepsyms=1 -hbfx-dump-nvram

Erictoby, 2019-10-09:
@idaima The fix is not yet included in the release. You can compile it or if you want to test it personally, you may try this -> [beta](https://github.com/Erictoby/VoodooI2C/releases).

idaima, 2019-10-10:
> @idaima The fix is not yet included in the release. You can compile it or if you want to test it personally, you may try this -> [beta](https://github.com/Erictoby/VoodooI2C/releases).

I have tested the beta version of voodooi2c (v2.2 + Beta L)  and the result is the same...kernel panic

Erictoby, 2019-10-10:
@idaima  Could you test it with the file again ([v2.2 + beta M](https://github.com/Erictoby/VoodooI2C/releases))? There might be a possibility accessing array out of bounds in the interrupt call.

idaima, 2019-10-11:
> @idaima Could you test it with the file again ([v2.2 + beta M](https://github.com/Erictoby/VoodooI2C/releases))? There might be a possibility accessing array out of bounds in the interrupt call.

i have tested the beta version of voodooi2c (v2.2 + Beta M) and i get this kernel panic :

Fri Oct 11 12:06:10 2019

*** Panic Report ***
panic(cpu 3 caller 0xffffff80058dbadd): Kernel trap at 0xffffff8005e596b6, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001c395000, CR4: 0x00000000003626e0
RAX: 0xffffff803a981980, RBX: 0xffffff803a981b00, RCX: 0x0000000000000017, RDX: 0xffffff803a43e200
RSP: 0xffffff81f89f3da0, RBP: 0xffffff81f89f3dc0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000003, R11: 0x00000000ad972002
R12: 0xffffff7f897196c8, R13: 0x0000000000000000, R14: 0x0000000000000000, R15: 0x0000000000000018
RFL: 0x0000000000010202, RIP: 0xffffff8005e596b6, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x3, PL: 0, VF: 1

Backtrace (CPU 3), Frame : Return Address
0xffffff81f89f3870 : 0xffffff80057ae6ed mach_kernel : _handle_debugger_trap + 0x47d
0xffffff81f89f38c0 : 0xffffff80058ea185 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81f89f3900 : 0xffffff80058db8ba mach_kernel : _kernel_trap + 0x50a
0xffffff81f89f3970 : 0xffffff800575bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f89f3990 : 0xffffff80057ae107 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff81f89f3ab0 : 0xffffff80057adf53 mach_kernel : _panic + 0x63
0xffffff81f89f3b20 : 0xffffff80058dbadd mach_kernel : _kernel_trap + 0x72d
0xffffff81f89f3c90 : 0xffffff800575bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f89f3cb0 : 0xffffff8005e596b6 mach_kernel : __ZN22IOInterruptEventSource23normalInterruptOccurredEPvP9IOServicei + 0x56
0xffffff81f89f3dc0 : 0xffffff7f89713436 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO32intel_gpio_community_irq_handlerEP15intel_communityPb + 0xbe
0xffffff81f89f3e10 : 0xffffff7f8971373f org.coolstar.VoodooGPIO : __ZN10VoodooGPIO22interruptOccurredGatedEv + 0x4d
0xffffff81f89f3e50 : 0xffffff8005e5ab85 mach_kernel : __ZN13IOCommandGate13attemptActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb5
0xffffff81f89f3eb0 : 0xffffff7f89712f8d org.coolstar.VoodooGPIO : __ZN10VoodooGPIO17InterruptOccurredEP8OSObjectP22IOInterruptEventSourcei + 0x55
0xffffff81f89f3ed0 : 0xffffff8005e58d8c mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x13c
0xffffff81f89f3f30 : 0xffffff8005e56e72 mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x1e2
0xffffff81f89f3f70 : 0xffffff8005e563cc mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x2c
0xffffff81f89f3fa0 : 0xffffff800575b0ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[D8F8D631-1C28-33A3-92F1-14646789FBB4]@0xffffff7f89710000->0xffffff7f89726fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3E8E9870-2E41-33F6-ACBF-35DDC08F22B5]@0xffffff7f866c5000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 kext-dev-mode=1 -cdfon -igfxhdmidivs alcid=27 keepsyms=1 -hbfx-dump-nvram -v debug=0x100 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Kernel UUID: C41337A1-0EC3-3896-A954-A1F85E849D53
Kernel slide:     0x0000000005400000
Kernel text base: 0xffffff8005600000
__HIB  text base: 0xffffff8005500000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)

ben9923, 2019-10-11:
I believe the attached build should solve the panic mentioned in the [issue](https://github.com/alexandred/VoodooI2C/issues/201#issuecomment-515667785) (As other panic logs are produced by custom builds, I cannot verify them).
Please check it.
@idaima If it does not help, please attach a panic log made with 'official' VoodooI2C release (and the attached build).
That said, it's not an expected situation. A firmware update could help too, if available.

idaima, 2019-10-12:
@ben9923 it's not working and i still get kernel panic when first cold boot macOS
**kernel panic log:**

*** Panic Report ***
panic(cpu 4 caller 0xffffff80012dbadd): Kernel trap at 0xffffff80018596b6, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000017d94000, CR4: 0x00000000003626e0
RAX: 0xffffff80362d7300, RBX: 0xffffff80362d7a80, RCX: 0x0000000000000017, RDX: 0xffffff8035dc1d00
RSP: 0xffffff81f3e03dc0, RBP: 0xffffff81f3e03de0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000006, R9:  0x0000000000000000, R10: 0x0000000000000004, R11: 0x0000000000000000
R12: 0xffffff8035618c08, R13: 0x0000000000000017, R14: 0x0000000000000000, R15: 0xffffff7f851186c0
RFL: 0x0000000000010202, RIP: 0xffffff80018596b6, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffff81f3e03890 : 0xffffff80011ae6ed mach_kernel : _handle_debugger_trap + 0x47d
0xffffff81f3e038e0 : 0xffffff80012ea185 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81f3e03920 : 0xffffff80012db8ba mach_kernel : _kernel_trap + 0x50a
0xffffff81f3e03990 : 0xffffff800115bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f3e039b0 : 0xffffff80011ae107 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff81f3e03ad0 : 0xffffff80011adf53 mach_kernel : _panic + 0x63
0xffffff81f3e03b40 : 0xffffff80012dbadd mach_kernel : _kernel_trap + 0x72d
0xffffff81f3e03cb0 : 0xffffff800115bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f3e03cd0 : 0xffffff80018596b6 mach_kernel : __ZN22IOInterruptEventSource23normalInterruptOccurredEPvP9IOServicei + 0x56
0xffffff81f3e03de0 : 0xffffff7f8511250f org.coolstar.VoodooGPIO : __ZN10VoodooGPIO32intel_gpio_community_irq_handlerEP15intel_community + 0xa1
0xffffff81f3e03e30 : 0xffffff7f851127b6 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO22interruptOccurredGatedEv + 0x2e
0xffffff81f3e03e60 : 0xffffff800185a908 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0x138
0xffffff81f3e03ed0 : 0xffffff8001858d8c mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x13c
0xffffff81f3e03f30 : 0xffffff8001856e72 mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x1e2
0xffffff81f3e03f70 : 0xffffff80018563cc mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x2c
0xffffff81f3e03fa0 : 0xffffff800115b0ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[E1D06308-22BF-3DA6-A226-37E027A066F1]@0xffffff7f85110000->0xffffff7f85125fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3E8E9870-2E41-33F6-ACBF-35DDC08F22B5]@0xffffff7f820c5000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 kext-dev-mode=1 -cdfon -igfxhdmidivs alcid=27 keepsyms=1 -hbfx-dump-nvram

ben9923, 2019-10-14:
@idaima Please give this build a try too :)

[Fix GPIO boot panic 2.zip](https://github.com/alexandred/VoodooI2C/files/3726714/Fix.GPIO.boot.panic.2.zip)

idaima, 2019-10-15:
> @idaima Please give this build a try too :)
> 
> [Fix GPIO boot panic 2.zip](https://github.com/alexandred/VoodooI2C/files/3726714/Fix.GPIO.boot.panic.2.zip)

i have try the Fix GPIO boot panic 2 , but it still not working . the kernel panic log as below :

*** Panic Report ***
panic(cpu 4 caller 0xffffff80056dbadd): Kernel trap at 0xffffff8005c596b6, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001c194000, CR4: 0x00000000003626e0
RAX: 0xffffff803a784480, RBX: 0xffffff803a784900, RCX: 0x0000000000000017, RDX: 0xffffff803a15b200
RSP: 0xffffff81f86f3dc0, RBP: 0xffffff81f86f3de0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0x0000000000000000, R13: 0x0000000000800000, R14: 0x0000000000000000, R15: 0xffffff7f895186c0
RFL: 0x0000000000010202, RIP: 0xffffff8005c596b6, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffff81f86f3890 : 0xffffff80055ae6ed mach_kernel : _handle_debugger_trap + 0x47d
0xffffff81f86f38e0 : 0xffffff80056ea185 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81f86f3920 : 0xffffff80056db8ba mach_kernel : _kernel_trap + 0x50a
0xffffff81f86f3990 : 0xffffff800555bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f86f39b0 : 0xffffff80055ae107 mach_kernel : _panic_trap_to_debugger + 0x197
0xffffff81f86f3ad0 : 0xffffff80055adf53 mach_kernel : _panic + 0x63
0xffffff81f86f3b40 : 0xffffff80056dbadd mach_kernel : _kernel_trap + 0x72d
0xffffff81f86f3cb0 : 0xffffff800555bb40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f86f3cd0 : 0xffffff8005c596b6 mach_kernel : __ZN22IOInterruptEventSource23normalInterruptOccurredEPvP9IOServicei + 0x56
0xffffff81f86f3de0 : 0xffffff7f89512515 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO32intel_gpio_community_irq_handlerEP15intel_community + 0x97
0xffffff81f86f3e30 : 0xffffff7f895127b6 org.coolstar.VoodooGPIO : __ZN10VoodooGPIO22interruptOccurredGatedEv + 0x2e
0xffffff81f86f3e60 : 0xffffff8005c5a908 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0x138
0xffffff81f86f3ed0 : 0xffffff8005c58d8c mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x13c
0xffffff81f86f3f30 : 0xffffff8005c56e72 mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x1e2
0xffffff81f86f3f70 : 0xffffff8005c563cc mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x2c
0xffffff81f86f3fa0 : 0xffffff800555b0ce mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.coolstar.VoodooGPIO(1.1)[6A0F03A7-C46A-3304-8074-87AC35DECB9B]@0xffffff7f89510000->0xffffff7f89525fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3E8E9870-2E41-33F6-ACBF-35DDC08F22B5]@0xffffff7f864c5000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 kext-dev-mode=1 -cdfon -igfxhdmidivs alcid=27 keepsyms=1 -hbfx-dump-nvram 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Kernel UUID: C41337A1-0EC3-3896-A954-A1F85E849D53
Kernel slide:     0x0000000005200000
Kernel text base: 0xffffff8005400000
__HIB  text base: 0xffffff8005300000
System model name: MacBookPro13,3 (Mac-A5C67F76ED83108C)

zhangyin-github, 2019-11-06:
I think I have the same problem here.
Dell Latitude 7490
Using the clover config file attached here at https://osxlatitude.com/forums/topic/11410-dell-latitude-7490-with-i7-8650u-intel-uhd-620-and-1920x1080-lcd-mojavecatalina/
[image]

djlucas123456, 2019-11-16:
I have the same problem.  I found out that as soon as the loading process shows and I press any key so panic will not occur!

hugotx, 2019-12-27:
> I have the same problem. I found out that as soon as the loading process shows and I press any key so panic will not occur! Dell G3 i2c+HID

ME too!!!!!!!!!!! Why is this you know?

djlucas123456, 2019-12-27:
> > I have the same problem. I found out that as soon as the loading process shows and I press any key so panic will not occur! Dell G3 i2c+HID
> 
> ME too!!!!!!!!!!! Why is this you know?

update the latest version of voodoops2!

https://github.com/acidanthera/VoodooPS2/releases

ben9923, 2020-04-25:
Does it reproduce with latest VoodooI2C release?

djlucas123456, 2020-04-25:
The latest released version is fine.

idaima, 2020-04-26:
it's working well with the latest version

closed 2020-04-29 by ben9923