VoodooI2C/VoodooI2C #205: ELAN1200 I2C Interface (Coffelake Chipset)

dinimuhtar, 2019-08-01:
Please support ELAN1200 Touchpad (I2C for Coffelake is not supported by VoodooI2C + Satelite Kexts yet). Thanks.

williambj1, 2019-08-01:
CoffeeLake devices are fully supported in GPIO and Polling Mode currently.
In some situations, you may need to patch `USTP` or `FMCN` & `SSCN` in your DSDT (They store bus speed values)

I personally have a CFL-H device with an ELAN 1404 trackpad and it works like a charm.
I'd suggest you to try first instead of just asking for support here  🙃

dinimuhtar, 2019-08-01:
can you check what wrong with my dsdt?
[DSDT.aml 2.zip](https://github.com/alexandred/VoodooI2C/files/3456128/DSDT.aml.2.zip)

ben9923, 2019-08-01:
Full Coffee Lake support was introduced in v2.2.
If you need some help, please attach a full troubleshooting archive in the [VoodooI2C Gitter chat](https://gitter.im/alexandred/VoodooI2C).
Make sure you attach at least origin + patched DSDT, IOReg and your kernel log.

closed 2019-08-01 by ben9923