VoodooI2C/VoodooI2C #206: HID touchpad fails every fifth single touch

Teufeuleu, 2019-08-09:
Hi,
I've this weird issue since the beginning of my hackintosh experience with a chinese laptop, a TongFang GK5CN6Z, including a HID precision touchpad.

Trackap works with almost all multitouch gestures, prefPan is available, but the main issue is that every 5th single touch (for moving the cursor or a single tap to click) is simply ignored.
A multitouch feature (double tap for right click, scroll, switch space...) will always work.
A single touche (whereas it is for moving the cursor or single tapto click) will almost be ignored if happening after a double tap (right click).

I updated macOS, Clover, drivers and kexts to their latest version, and the issue is still here.

My main doubts: I missed something during installation process. For instance I have a IOInterruptSpecifiers liine not in the TPAD section of my IOReg, but in one of its parents. Anyway it's under 2f. Also it seems that from my understanding, GPI0 is properly pinned in the original DSDT. However I included in the report zip all of the ssdt patches in case of any conflict.

I'm quite new in reporting issues here and in hackintosh in general so i'm sorry in advance if there is some missing/useless files into the report archive.
[GK5CN6Z trackpad troubleshooting 2.zip](https://github.com/alexandred/VoodooI2C/files/3486934/GK5CN6Z.trackpad.troubleshooting.2.zip)

ben9923, 2019-08-09:
Seems like GPIO is working well (Though kernel log isn't full).
I see Karabiner is installed, can you try removing/disabling it?
Please verify it's working in Linux, you can use Ubuntu Live CD.

Anyway, closing as it appears to be a duplicate of #185 
Please continue discussion there.

ben9923, 2019-08-09:
Duplicate of #185

closed 2019-08-09 by ben9923