VoodooI2C/VoodooI2C #208: Support individual trackpad buttons

williambj1, 2019-08-16:
**This may be the same issue as #182 #186 #190 #207**

## Issue description

VoodooI2C simulates Magic Trackpad 2, which is an integrated one-button trackpad, so every button press will have a touching point as well.

However, trackpads with individual trackpad buttons and won't be a touching point when the button is clicked.

[image]
`Image from https://www.techadvisor.co.uk/review/ultraportable-laptops-ultrabooks/asus-rog-zephyrus-s-3682482/`

> Lenovo Thinkpads have similar buttons

This causes an issue when I'm trying to select text with my thumb on the button and one finger on the trackpad, macOS has no idea what's going on since it's expected to have two touching points present. When I'm trying to select a long line of text, I can't leave my moving finger (Leaving breaks everything).

[Leaving-breaks-everything.zip](https://github.com/alexandred/VoodooI2C/files/3508196/Leaving-breaks-everything.zip)
`This video shows what will happen when you leave your finger.`

What's more, single click and right-click doesn't work as well due to ***the lack of one finger*** on the left/right corner of the trackpad.

## Temporary Solution

[As blankmac suggested:](https://github.com/alexandred/VoodooI2C/issues/190#issuecomment-493789921)

> As for the selection behavior, try this so we can see if the two issues are related -- with one finger, move the cursor into position to select some text, leave that finger in place as an anchor, click/hold your physical button, and then use an additional finger to drag/select while leaving the first finger in place and see if it's still the same. (unless that's what you're already doing, of course)

This solution works well, but it's inconvenient, there's a requirement that the **additional holding finger** must be below your moving finger which means you have to put your fingers in an uncomfortable position otherwise everything breaks again.
Another solution is to stretch out your thumb a bit and make a touching point at the bottom of the trackpad. It feels a little weird and needs time to get used to it and it doesn't work for Thinkpads since their buttons are on the top.

## Code based solution

I think VoodooI2CHID or VoodooInput could check whether there is a touching point at the corners when buttons are pressed. If not, add a virtual one.

I have no experience in IOKit and C++ at all. So could anyone with such a trackpad help with this issue? 

-----
Gitter discussion:

https://gitter.im/alexandred/VoodooI2C?at=5d55457f5178a724769062b9

-----

### Thanks in advance!!!

LeeBinder, 2019-12-16:
Hello @williambj1, thank you for your detailed issue description. I've been dealing with this very annoying regression ever since the MT2 switch from v.2.0.3 to v.2.1+, too. Very sadly this issue remains even in the latest 2.3, so I feel it needs reinforcement.

Here my suggestions/ requests, so this gains momentum again towards a final fix:

1) can please anybody affected (with ELAN 1200/ 1300 and other similar touchpads) add their "**_me, too_**" in here, upvote William's post above, + add some hardware details if you feel inclined to?
@animator013 @whatnameisit @black-dragon74 @hieplpvip @HolzMichel @overnightmare95 @fewtarius @Sniki 👍 

2) William, I think your issue has attracted little attention because the title might not be optimal. It is technical but does not describe the issue in layman's terms. I think it would help a lot if you edited it into something more catching/ emotional/ "sexy" ;), like "Dragging/ highlighting/ moving *horrible* on touchpads with individual trackpad buttons"

3) This worked before in 2.0.3. If this can't be fixed with the MT2 engine, I suggest a 10.14+ port of the [2.0.3 code](https://github.com/alexandred/VoodooI2C/releases/tag/v2.0.3). @Sniki managed to create such a partially working compile before (see [here](https://github.com/alexandred/VoodooI2C/issues/108#issuecomment-448022766)), demonstrating it's not impossible but would need to be perfected.

**I'M POSITIVE WE CAN GET THIS RESOLVED :)**

For completion:

I previously described this bug in another topic, so let me just quote my wording form over there so it's all in one place here:
> any two-finger actions involving pressing down the left trackpad button (index finger) + at the same time sliding over the trackpad with the middle finger in stages barely work anymore, like:
> - dragging a window over the screen
> - highlighting text in a document
> - moving a slider left <-> right

My hardware: ELAN 1300 I2C Controller (8086 : 9d60 / 9d61, subsys 1043 / 1D30) in a Asus Vivobook S15.

@ben9923 (Ben Raz)'s encouraging remarks from the communication over at [gitter/VoodooI2C](https://gitter.im/alexandred/VoodooI2C?at=5d55457f5178a724769062b9):

<details>
  <summary>click to expand</summary> 
Ben:

> It is possible, yeah. Should be implemented in VoodooI2CHID, probably won't be too hard.
> I don't have such hardware so I can't really do it on my own, I'm afraid...

William:
> I could test it :). I think a solution is to check whether there is a touching point at the corners when buttons are pressed. If not, then add a virtual one."

Ben:
> I think such a check can actually be eliminated - and we should just report one extra finger in the correct corner.
> Anyway, debugging that can be really inconvenient, I'd really prefer someone with both knowledge and hardware will get into it...
> Hopefully it gets implemented soon. IMO, It won't be on VoodooInput. The multitouch engine is really responsible to just get the input and reformat it to the correct MT2 protocol. HID kext should be responsible for emulating the extra fingers for buttons.

> I do, however, see that currently the MT2 simulator does get a button status in its report, I think it shouldn't be like that...
</details>

HolzMichel, 2019-12-16:
@LeeBinder  Which kind of hardware information do you need? I have not yet updated to 2.3, i use 2.2 on Mojave and also had it running with Catalina with no more (at least noticeable) interrupts or lags at using the touchpad, and I did not plan to touch that running system, but for test purposes i will for sure help to test if i find time. The short version of the problem is, that with 2.3 there will be a Sleep/Wake Issue, that wasn´t there in previous versions? Feel free to correct me if i got it wrong.

LeeBinder, 2019-12-16:
Thanks @HolzMichel 
1. this issue has been dragging along and effecting two physical buttons touchpads ever since VoodooI2C v.2.1 and simply remains the same with v.2.3, so no need to update to be effected by it - either you are or you're not.
2. I figured out how to solve the kernel panic on wake issue with VI2C v.2.3, see [here](https://github.com/alexandred/VoodooI2C/issues/183#issuecomment-565858834) :)

LeeBinder, 2020-01-04:
@williambj1 I have a special VoodooI2C compile with 2 buttons support here which works perfectly, all issues solved! Let me know if you're interested and I'll post the link.

ladeiko, 2020-01-04:
> @williambj1 I have a special VoodooI2C compile with 2 buttons support here which works perfectly, all issues solved! Let me know if you're interested and I'll post the link.

Hi. If possible, give the link to compiled version and source code if possible, thanx.

williambj1, 2020-01-04:
@LeeBinder 

Thanks for tracking this issue and being passionate to solve it!

If you have the source and the compiled kext, could you please post it here?

I’m extremely busy for my collage work right now and I apologize that I appeared to be missing and forgot about this issue, in fact, I’m still in the community and there’s still a lot worth to explore.

Thanks again & happy new year!

LeeBinder, 2020-01-04:
hey @williambj1 , good to hear you're not vanished but still around and just focused. And good going on changing the title to something more tangible & appealing.

Also @ladeiko : the driver has been around for 1,5 years, buried in a closed issue.. Download stevezhengshiqi's v.2.0.3 compile from [here](https://github.com/alexandred/VoodooI2C/issues/70#issuecomment-405458057), apply, reboot, and hopefully enjoy. It works for me in Mojave and Catalina. His 2.1 and 2.1.1 compiles don't, but they are meaningless for our purpose anyway because v.2.0.3 was the latest with independent 2-button support anyway.

It misses some gestures but is simply more solid & reliable. I still consider VoodooI2C 2.1+ beta until integration of VoodooInput incl. engine.

stevezhengshiqi did not supply his source, so you would need to address him, best in that thread over there, I'd say.

Looking forward to hearing how this kext works for you two.

ladeiko, 2020-01-05:
@LeeBinder just tested on mojave  - it works, thanks

LeeBinder, 2020-01-05:
@ladeiko great, glad to hear. Let Steve know [here](https://github.com/alexandred/VoodooI2C/issues/70#issuecomment-405458057) where he shared with us, too, with a thumbs up - it's his compilation work and credit :)

williambj1, 2020-03-22:
@LeeBinder 

Goshin, an absolute genius, solved this issue by tweaking some `if` expressions in the code in the MT2 Engine, dragging is never this stable than before.

I will close this issue since the primary part (Dragging) is solved, buttons for direct click responding should be another story.

closed 2020-03-22 by williambj1

LeeBinder, 2020-03-22:
that's great news, @williambj1 - thank you for keeping me in the loop!

There doesn't seem to be any public pre-release compile of VoodooI2C with the latest commits yet, so I patiently wait for the upcoming release.