VoodooI2C/VoodooI2C #210: Trackpad not showing up in preferences pane / tracking speed / right click

tortuga37, 2019-08-29:
The version that introduced right click also nerfed the trackpad mouse speed (too slow) and two finger drag touchscreen scroll speed (too fast) and on my 2017 Razer Blade stealth Hackintosh. I do not want to have to go back to using an older version and sacrifice two finger right click to maintain the correct aforementioned speeds, as the trackpad does not show up in preferences pane and speed can not be changed.

ben9923, 2019-08-30:
Do you have battery status working? It's necessary to make it appear.

tortuga37, 2019-08-31:
Thank you that solved it.  It is interesting how patching the DSDT with the battery hot patch would effect the trackpad.  I am glad it worked.

ben9923, 2019-08-31:
It's something in Apple's design. A method to determine it's a laptop, maybe? (Though they can just use the SMBIOS...)
Not completely sure why, but lack of battery status is known to make the PrefPane empty.

Closing as it's resolved 😀

closed 2019-08-31 by ben9923