VoodooI2C/VoodooI2C #211: Touchpad sometimes can't detect my finger

esefberfu, 2019-08-29:
when i want to control for only to move to cursor, sometimes touchpad does not detect my finger. so that i release the my finger and i again touch to touchpad for to control to cursor with only one finger. An other option, Touchpad that if i hold on one finger on work fine. So one finger doesn't move and other finger control to cursor,
if i control with three fingers too but two finger is detected by touchpad and scrolling works

on Windows OS, touchpad don't has any problem, everything stable.
there is not any problem to scrolling with two fingers

joeyave, 2020-04-05:
> if i control with three fingers too sometimes only two fingers is detected by touchpad and scrolling works

the same here.

ben9923, 2020-04-11:
@esefberfu @joeyave Have you tried v2.4?

joeyave, 2020-04-11:
> @esefberfu @joeyave Have you tried v2.4?
gonna try it now!

joeyave, 2020-04-12:
> @esefberfu @joeyave Have you tried v2.4?

It works perfectly with 2.4. Thank you!

closed 2020-04-13 by ben9923