VoodooI2C/VoodooI2C #214: Catalina Beta 7

Mateo1234454545, 2019-09-11:
On latest developer beta Catalina 7 , voodooi2c doesn't load.
Only way to do so is to use \System\Library\Extensions\IOGraphicsFamily.kext at config/plist/forcekextstoload

ben9923, 2019-09-11:
Did you rebuild your kextcache after updating?
`sudo kextcache -i /`

Mateo1234454545, 2019-09-11:
Yes , off course I did.
To be precise , hackintool shows that voodooi2c is loaded but there is no trackpad option in preferences. That happens with latest 2.2 version and 2.1.4 as well.
If the patch is applied at config everything is back to normal.

ben9923, 2019-09-14:
@Mateo1234454545 Developer beta 8 (19A558d) is available, have you tried it?
You'll need to attach a troubleshooting archive for diagnosis.

Mateo1234454545, 2019-09-14:
@ben9923 , beta 7 is developer beta 8.
I can't create a debug file (catalina shows error debug) , so I sent you as much as I can. If you need something more please let me know.
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/3612725/Archive.zip)

Andrw0830, 2019-09-16:
I can also confirm the issue here too. I couldn't find a way to force IOGraphicsFamily with OpenCore so booted my Clover install in the meantime. Hopefully this helps find a solution in general to load VoodooI2C since us having to force IOGraphics to load or rebuild kext cache has been happening for awhile.

ben9923, 2019-09-16:
> @ben9923 , beta 7 is developer beta 8.
> I can't create a debug file (catalina shows error debug) , so I sent you as much as I can. If you need something more please let me know.
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3612725/Archive.zip)

Kernel log and kextcache log would be helpful.
Which errors do you get? Have you tried running the debug script from your home or Downloads folder, for example?

What's the output of this command (Navigate to the directory containing VoodooI2CHID.kext)?
`kextlibs -undef-symbols VoodooI2CHID.kext`

Please also try building with the kexts with the 10.15 SDK if you can. Some `NULL` -> `0` changes are probably necessary around the code, see https://github.com/ben9923/VoodooGPIO/commit/00b138251b251a1353c6839f16f451e8bc8f7562 for example (Integer expected, while `NULL` is `nullptr` instead of `0` in the 10.15 SDK)

rossna, 2019-09-17:
10.15 Developer Beta 8 with the same issue, here's the output of said command:

```
For all architectures:
    com.apple.iokit.IOACPIFamily = 1.4
    com.apple.iokit.IOGraphicsFamily = 558.3
    com.apple.iokit.IOHIDFamily = 2.0
    com.apple.kpi.iokit = 19.0
    com.apple.kpi.libkern = 19.0
    com.apple.kpi.mach = 19.0

For x86_64:
    31 symbols not found in any library kext:
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService10Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService15Ev
	__ZN24VoodooI2CDigitiserStylus9metaClassE
	__ZN18VoodooI2CDeviceNub9metaClassE
	__ZN15OSMetaClassBase25_RESERVEDOSMetaClassBase3Ev
	__ZN28VoodooI2CMultitouchInterfaceC1Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService20Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService13Ev
	__ZN18VoodooI2CDeviceNub12writeReadI2CEPhtS0_t
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice13Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService18Ev
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService8Ev
	__ZN28VoodooI2CMultitouchInterface21handleInterruptReportE24VoodooI2CMultitouchEventy
	__ZN18VoodooI2CDeviceNub7readI2CEPht
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService11Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService16Ev
	__ZN28VoodooI2CDigitiserTransducer9metaClassE
	__ZN24VoodooI2CDigitiserStylus6stylusE23DigitiserTransducerTypeP12IOHIDElement
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice12Ev
	__Z14getMatchedNameP9IOService
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice15Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService21Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService14Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService19Ev
	__ZN28VoodooI2CDigitiserTransducer10transducerE23DigitiserTransducerTypeP12IOHIDElement
	__ZN18VoodooI2CDeviceNub8writeI2CEPht
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService7Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService12Ev
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService9Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService17Ev
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice14Ev
```

HunterAhlquist, 2019-09-17:
Having the same problem with ELAN1300, the output looks the same as the above guy's but I mostly just wanted to keep this issue active.
```
For all architectures:
    com.apple.iokit.IOACPIFamily = 1.4
    com.apple.iokit.IOGraphicsFamily = 558.3
    com.apple.iokit.IOHIDFamily = 2.0
    com.apple.kpi.iokit = 19.0
    com.apple.kpi.libkern = 19.0
    com.apple.kpi.mach = 19.0

For x86_64:
    31 symbols not found in any library kext:
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService10Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService15Ev
	__ZN24VoodooI2CDigitiserStylus9metaClassE
	__ZN18VoodooI2CDeviceNub9metaClassE
	__ZN15OSMetaClassBase25_RESERVEDOSMetaClassBase3Ev
	__ZN28VoodooI2CMultitouchInterfaceC1Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService20Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService13Ev
	__ZN18VoodooI2CDeviceNub12writeReadI2CEPhtS0_t
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice13Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService18Ev
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService8Ev
	__ZN28VoodooI2CMultitouchInterface21handleInterruptReportE24VoodooI2CMultitouchEventy
	__ZN18VoodooI2CDeviceNub7readI2CEPht
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService11Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService16Ev
	__ZN28VoodooI2CDigitiserTransducer9metaClassE
	__ZN24VoodooI2CDigitiserStylus6stylusE23DigitiserTransducerTypeP12IOHIDElement
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice12Ev
	__Z14getMatchedNameP9IOService
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice15Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService21Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService14Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService19Ev
	__ZN28VoodooI2CDigitiserTransducer10transducerE23DigitiserTransducerTypeP12IOHIDElement
	__ZN18VoodooI2CDeviceNub8writeI2CEPht
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService7Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService12Ev
	__ZN17IOHIDEventService27_RESERVEDIOHIDEventService9Ev
	__ZN17IOHIDEventService28_RESERVEDIOHIDEventService17Ev
	__ZN11IOHIDDevice22_RESERVEDIOHIDDevice14Ev
```

Erictoby, 2019-09-17:
IOGraphicsFamily dependency is required only to support the screen rotation in tablet(mode) having an accelerometer. Most desktops or laptops don't need it now but it will become a good feature in future.

tankren, 2019-09-21:
> Same problem here, touchpad works after forcing IOGraphicsFamily.kext to load.

hello there, the trick works for me and the touchpad works now, but I cannot see my touchpad in the Preference. How about yours?

Mateo1234454545, 2019-09-23:
Problem solved on Catalina beta 9.
No need for the patch anymore.

kirainmoe, 2019-09-24:
> > Same problem here, touchpad works after forcing IOGraphicsFamily.kext to load.
> 
> hello there, the trick works for me and the touchpad works now, but I cannot see my touchpad in the Preference. How about yours?

I can see it in the System Preference. I'm using VoodooI2CHID.kext. Have you tried to rebuild the kextcache?

ben9923, 2019-09-27:
Closing as issue opener confirmed it's a macOS beta issue that was fixed in developer beta 9.

closed 2019-09-27 by ben9923