VoodooI2C/VoodooI2C #215: Touchpad acting strangely

djale, 2019-09-18:
Hello, I have Lenovo Ideapad 520-15IKB with macOS 10.14.4

The touchpad is giving me headaches when satellite CHID is used it "works" but very buggy, and it's not shown in system prefs (I have battery info working fine).
Video : https://www.youtube.com/watch?v=aFscOezzPdU

If I change satellite to ELAN or Synaptics it doesn't work at all.

My device is Synaptics, vendor ID: 0x6cb.

[image]

[image]

I have patched GPIO interrupts!

Someone told me to change the name match in Info.plist to SYNA2B3C, but I'm not clear with that, what should I exactly edit!?

Thanks
[debug_12960.zip](https://github.com/alexandred/VoodooI2C/files/3623883/debug_12960.zip)

onemanosx, 2019-09-18:
VoodooI2CSynaptics>Show Package Contents>Contents>info.plist
Use xcode/plist editor/textmate and edit IOPropertyMatch>name to "SYNA2B3C"

Remove VoodooI2CHID and use only VoodooI2C + VoodooI2CSynaptics kext and reboot.

[image]

djale, 2019-09-18:
@onemanosx Thank you for your time, and for helping out! 
Apparently, when I change satelite to synaptics, touchpad doesn't work at all and I also changed the name as you guided me.. 

It's strange because touchpad only "works" with CHID, but it's like on the video as I posted, with other satellites it's dead. I checked gpio and changed this in info.plist, and now I don't have idea where is the issue.

onemanosx, 2019-09-18:
check a branch release  here https://github.com/EmotionalLove/VoodooI2CSynaptics/releases/tag/2.3

johnnync13, 2019-09-24:
https://github.com/williambj1/GenI2C

djale, 2019-09-24:
@onemanosx 
No luck with new kext..

@johnnync13 
Thanks for the reply, I'm not sure should I patch it again with that generator, because I believe that is well patched already... Not sure what to do.

djale, 2019-09-24:
Just to be sure that my patching is right 

`Scope (_SB.PCI0.I2C0)
    {
        Device (TPD0)
        {
            Name (HID2, Zero)
            Name (SPED, Zero)
            Name (BADR, Zero)
            Method (_INI, 0, NotSerialized)  // _INI: Initialize
            {
                If (LEqual (TPTY, One))
                {
                    Store ("ELAN0611", _HID)
                    Store (0x15, BADR)
                    Store (One, HID2)
                }

                If (LEqual (TPTY, 0x02))
                {
                    Store ("SYNA2B3C", _HID)
                    Store (0x2C, BADR)
                    Store (0x20, HID2)
                }

                Store (0x00061A80, SPED)
                Return (Zero)
            }

            Name (_HID, "XXXX0000")  // _HID: Hardware ID
            Name (_CID, "PNP0C50")  // _CID: Compatible ID
            Name (_S0W, 0x03)  // _S0W: S0 Device Wake State
            Method (_DSM, 4, Serialized)  // _DSM: Device-Specific Method
            {
                Return (CDSM (Arg0, Arg1, Arg2, Arg3, HID2))
            }

            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                If (LEqual (TPTY, Zero))
                {
                    Return (Zero)
                }
                Else
                {
                    Return (0x0F)
                }
            }

            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                If (LEqual (TPTY, One))
                {
                    Name (SBFB, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x0015, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                    })
                }

                If (LEqual (TPTY, 0x02))
                {
                    Name (SBFS, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x002C, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                    })
                }

                Name (SBFI, ResourceTemplate ()
                {
                    Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, _Y1D)
                    {
                        0x00000000,
                    }
                })
                CreateDWordField (SBFI, \_SB.PCI0.I2C0.TPD0._CRS._Y1D._INT, INT2)  // _INT: Interrupts
                Store (INUM (0x0202000F), INT2)
                If (LEqual (TPTY, One))
                {
                    Return (ConcatenateResTemplate (SBFB, SBFI))
                }

                If (LEqual (TPTY, 0x02))
                {
                    Return (ConcatenateResTemplate (SBFS, SBFI))
                }
            }

            Method (TPRD, 0, Serialized)
            {
                Return (^^^LPCB.EC0.ECTP)
            }

            Method (TPWR, 1, Serialized)
            {
                Store (Arg0, ^^^LPCB.EC0.ECTP)
            }
        }
    }`

djale, 2019-09-24:
Just to be sure that my patch is okay.

Scope (_SB.PCI0.I2C0)
    {
        Device (TPD0)
        {
            Name (HID2, Zero)
            Name (SPED, Zero)
            Name (BADR, Zero)
            Method (_INI, 0, NotSerialized)  // _INI: Initialize
            {
                If (LEqual (TPTY, One))
                {
                    Store ("ELAN0611", _HID)
                    Store (0x15, BADR)
                    Store (One, HID2)
                }

                If (LEqual (TPTY, 0x02))
                {
                    Store ("SYNA2B3C", _HID)
                    Store (0x2C, BADR)
                    Store (0x20, HID2)
                }

                Store (0x00061A80, SPED)
                Return (Zero)
            }

            Name (_HID, "XXXX0000")  // _HID: Hardware ID
            Name (_CID, "PNP0C50")  // _CID: Compatible ID
            Name (_S0W, 0x03)  // _S0W: S0 Device Wake State
            Method (_DSM, 4, Serialized)  // _DSM: Device-Specific Method
            {
                Return (CDSM (Arg0, Arg1, Arg2, Arg3, HID2))
            }

            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                If (LEqual (TPTY, Zero))
                {
                    Return (Zero)
                }
                Else
                {
                    Return (0x0F)
                }
            }

            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                If (LEqual (TPTY, One))
                {
                    Name (SBFB, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x0015, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                    })
                }

                If (LEqual (TPTY, 0x02))
                {
                    Name (SBFS, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x002C, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                    })
                }

                Name (SBFI, ResourceTemplate ()
                {
                    Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, _Y1D)
                    {
                        0x00000000,
                    }
                })
                CreateDWordField (SBFI, \_SB.PCI0.I2C0.TPD0._CRS._Y1D._INT, INT2)  // _INT: Interrupts
                Store (INUM (0x0202000F), INT2)
                If (LEqual (TPTY, One))
                {
                    Return (ConcatenateResTemplate (SBFB, SBFI))
                }

                If (LEqual (TPTY, 0x02))
                {
                    Return (ConcatenateResTemplate (SBFS, SBFI))
                }
            }

            Method (TPRD, 0, Serialized)
            {
                Return (^^^LPCB.EC0.ECTP)
            }

            Method (TPWR, 1, Serialized)
            {
                Store (Arg0, ^^^LPCB.EC0.ECTP)
            }
        }
    }

onemanosx, 2019-09-25:
from your debug files, your track pad is in polling mode, despite the attempt of SSDT for interrupt mode. In my limited knowledge, your SSDT is doing nothing.

djale, 2019-09-27:
@onemanosx I realized it's in polling mode, and don't have an idea what to change now..

djale, 2019-09-27:
@onemanosx I patched again dsdt and now it works perfectly! Thank you very much for helping out! Cheers

closed 2019-09-27 by djale