VoodooI2C/VoodooI2C #216: ELAN 1300 Not Working at All in macOS 1.15 Developer Beta 8

OmerFlame, 2019-09-22:
Hi, I have a ELAN 1300 trackpad and it wouldn't work under macOS Catalina 10.15 Beta 8.

I have an ASUS VivoBook X510UAR.

Any fixes? If I had any debugging data I would have given you it but i don't have any... I also used the kext patch for the Apple Intel I2C drivers and still nothing.

Any help would be appreciated.

juanaleixo, 2019-09-24:
I have the same problem, but with ELAN 469D! Beta 8 (19A573a).

williambj1, 2019-09-24:
Tried the solution here? #214

juanaleixo, 2019-09-25:
Yep, keep don't working!

ben9923, 2019-09-27:
#214 suggests it was fixed in developer beta 9 (19A573a).
In any case, you must attach troubleshooting files (IOReg, kernel log, DSDT...) for us to diagnose, how can we debug it otherwise? :)

juanaleixo, 2019-09-27:
Thank You, I solved that removing kexts from S/L/E:AppleIntelLpssI2C.kext, AppleIntelLpssI2CController.kext and AppleHPM.kext. As described here: https://www.tonymacx86.com/threads/wip-voodooi2c-i2c-trackpad-limited-support.204227/

ben9923, 2019-09-28:
> Thank You, I solved that removing that kexts from S/L/E:
> 
> * AppleIntelLpssI2C.kext
> * AppleIntelLpssI2CController.kext
> * AppleHPM.kext.
> 
> As described here: https://www.tonymacx86.com/threads/wip-voodooi2c-i2c-trackpad-limited-support.204227/

Removing those kexts is not recommended nor necessary, there are kext patches you can add to your config.plist. You're looking at an old thread.
Notice next VoodooI2C release will probably not require even those patches, see https://github.com/alexandred/VoodooI2C/commit/c6e3c278cda84a26f400a77f5ea57d819df9e405.

As the Catalina beta problem is already reported to be resolved, I'll close this issue too.

closed 2019-09-28 by ben9923