VoodooI2C/VoodooI2C #217: Problem with SYNA2B33

Dicky-MF, 2019-09-26:
I have some problem on I2C device it's SYNA2B33
When i palace VoodooI2CHID.kext (Of course i place VoodooI2C.kext too)
The device (trackpad) is not working, and when i use VoodooI2CHID.kext it's work but not fine
It's still move the cursor until i give a (click) on trackpad, or sometime the touchpad is too sensitive (VERY SENSITIVE)

Please someone give me a help

[Send me Fizers-MacBook-Pro.zip](https://github.com/alexandred/VoodooI2C/files/3657249/Send.me.Fizers-MacBook-Pro.zip)

onemanosx, 2019-09-26:
You should be using VoodooI2CSynaptics instead of VoodooI2CHID for SYNA2B33

Dicky-MF, 2019-09-27:
Alright, and then i just have to make TPD0._CRS hotpatch and _CRS work fine, right ?
* I inject GPI0 on DSDT

onemanosx, 2019-09-27:
You already have native GPI0 pin. If you had really read the guide, all you have to do is make sure _CRS  Return (ConcatenateResTemplate (SBFF, SBFG). There is no need for you to use SSDT or hotpatches. The SSDT has the wrong configuration as I had pointed out in my earlier post.

Anyway, this is installation issue and all queries regarding installation should be directed to the gitter chat. https://gitter.im/alexandred/VoodooI2C

Please respect the Dev's rules.

Dicky-MF, 2019-09-27:
Thanks for your help, i will read the guide carefully again

Issue closed, thanks

closed 2019-09-27 by Dicky-MF