VoodooI2C/VoodooI2C #218: Touchpad I2C MSFT0001 (TPAD) not working [2019 Lenovo C340]

Henri75, 2019-10-03:
Hi,
Touchscreen is working great on my Whiskey Lake laptop from scratch but no luck with trackpad MSFT0001 (stock DSTD with XOSI patch, polling, GPIO pinning, missing SSCN/FMCN patching  ...). 
MSFT0001 works perfectly with Ubuntu as a generic I2C touchpad device.
whatever I tried it ends with : 

VoodooI2CControllerDriver::pci8086,9de8 I2C Transaction error details
VoodooI2CControllerDriver::pci8086,9de8 slave address not acknowledged (7bit mode)
VoodooI2CControllerDriver::pci8086,9de8 I2C Transaction error: 0x08800001 - aborting
VoodooI2CHIDDevice::MSFT0001 Request for HID descriptor failed
VoodooI2CHIDDevice::MSFT0001 Could not get HID descriptor

Attached is a troubleshot archive

Thanks !
[TPAD troubleshot(runme).zip](https://github.com/alexandred/VoodooI2C/files/3686748/TPAD.troubleshot.runme.zip)

ben9923, 2019-10-04:
Closing as it's a (resolved) setup issue, and not a problem with VoodooI2C :)

closed 2019-10-04 by ben9923

alexnov-tech, 2020-02-05:
Hey what's the setup issue was there? I have exactly the same problem in I2C log. The i2C device is TPAD MSFT0001. Catalina
[image]