VoodooI2C/VoodooI2C #219: Kexts are not working on Mac OS X Catalina

dogaoz, 2019-10-05:
Kexts are not working on Mac OS X Catalina

ben9923, 2019-10-05:
AFAIK they work just fine in macOS Catalina.
Did the same setup work for you in Mojave?

Without a troubleshooting archive we cannot tell what's wrong. Please attach one.

dogaoz, 2019-10-05:
@ben9923 My DSDT/SSDT with I2C are working fine on Mac OS Mojave and prior.

I updated to latest Catalina. Everything else is working well after updating Clover etc.
but Trackpad and touchscreen stopped working with the exactly same DSDT/SSDT.
I can switch config files on clover boot device selection screen and boot into Mac OS Mojave with the same clover configuration and the trackpad and touchscreen are working perfectly fine on my Dell XPS 13 9350

What do you mean by troubleshooting archive? Which ones should I attach here?

ben9923, 2019-10-05:
> @ben9923 My DSDT/SSDT with I2C are working fine on Mac OS Mojave and prior.
>
> I updated to latest Catalina. Everything else is working well after updating Clover etc.
> but Trackpad and touchscreen [...]

Attach at least your DSDT/SSDTs (origin + patched), IOReg and kernel log.
Read the documentation linked from the README.

williambj1, 2019-10-06:
@dogaoz Try these two solutions and see if it works.

I'm running Catalina Developer Beta 11 (19A582a) and everything works perfectly.

### 1:

```bash
sudo mount -uw / && killall Finder
sudo kextcache -i/
```

---

### 2.

https://github.com/alexandred/VoodooI2C/issues/125#issuecomment-461927427

dogaoz, 2019-10-06:
> @dogaoz Try these two solutions and see if it works.
> 
> I'm running Catalina Developer Beta 11 (19A582a) and everything works perfectly.
> 

> ### 2.
> [#125 (comment)](https://github.com/alexandred/VoodooI2C/issues/125#issuecomment-461927427)

2nd one worked. Woo-hoo!!
ForceKexts method is the key for the IOGraphics Part!!

closed 2019-10-06 by dogaoz

dogaoz, 2019-10-08:
@andrewwyatt hmm. You might be right. I also did kextcache -i / and probably that triggered something too. However, it didn't work on first restart. It started working on the second for some reason right after I added the IOGraphics lines in the config.

andrewwyatt, 2019-10-08:
> @andrewwyatt hmm. You might be right. I also did kextcache -i / and probably that triggered something too. However, it didn't work on first restart. It started working on the second for some reason right after I added the IOGraphics lines in the config.

Just the kextcache invalidate along with the kext force didn't work for me at all.  I spent a lot of time trying to debug it and finally resorted to clover not being able to properly inject the dependency and shoved it in C/k/O.

jief666, 2019-12-07:
I've just tried Catalina 10.15.1.
I have no trackpad during install. Once installed, no trackpad at first. I did "kextcache -i /" and reboot and it works.
The force kext to load trick also worked.

Unfortunately, I couldn't have the trackpad working when running the installer, though. Is there a way to have that working in the installer ? Is that a Clover injection problem ?