VoodooI2C/VoodooI2C #222: ELAN1200 Stop working

Jie2GG, 2019-10-22:
I use the combination of VoodooI2C + VoodooI2CHID driver, which may stop working after normal use for a period of time.

My laptop CPU is Intel Core i5 8265U which is within the range of support. My trackpad is running in interrupt mode.

ben9923, 2019-10-22:
I suspect it’s a VoodooPS2 issue.
Can you try pressing `PrintScreen` when it happens? If you’re using VoodooPS2 for keyboard, anyway.

Jie2GG, 2019-10-22:
> I suspect it’s a VoodooPS2 issue.
> Can you try pressing `PrintScreen` when it happens? If you’re using VoodooPS2 for keyboard, anyway.

I am sorry that there is no PrintScreen key on the keyboard of my notebook, so I don't know how to deal with this problem now, because I don't understand how it happened.

ben9923, 2019-10-25:
> I am sorry that there is no PrintScreen key on the keyboard of my notebook, so I don't know how to deal with this problem now, because I don't understand how it happened.

Try searching how to disable's VoodooPS2's trackpad disabling notifications.
You can also try disabling `RM, deliverNotifications` in your satellite's info.plist.

Jie2GG, 2019-10-25:
> Try searching how to disable's VoodooPS2's trackpad disabling notifications.
> You can also try disabling `RM, deliverNotifications` in your satellite's info.plist.

I think this may be the bug of VoodooI2C, I have disabled RM, deliverNotifications in VoodooPS2Controller as you instructed, but it seems to have no obvious effect after restarting. My touch pad will still respond to the 5-finger gesture with a long press and click.My trackpad still recognizes a long left press as a quick click after using the five-finger gesture.

ben9923, 2019-10-25:
> I think this may be the bug of VoodooI2C, I have disabled RM, deliverNotifications in VoodooPS2Controller as you instructed, but it seems to have no obvious effect after restarting. My touch pad [...]

You described a different issue - you previously said the trackpad will simply stop working after a while.
Anyway, please attach a full troubleshooting archive.
Notice I suggested disabling it in the satellite's info.plist (VoodooI2CHID), _not_ VoodooPS2.

Jie2GG, 2019-10-25:
> You described a different issue - you previously said the trackpad will simply stop working after a while.
> Anyway, please attach a full troubleshooting archive.
> Notice I suggested disabling it in [...]

Sorry sorry, I am trying to test the problem described before.

ben9923, 2019-11-01:
> Hello, after my long test, there is a problem that will stop working under ELAN1200, and it does not conflict with voodoops2controller. kext

Did you verify disabling the deliverNotifications key doesn't resolve the issue?
Please attach a troubleshooting archive made when it's disable, after trackpad has stopped working.

closed 2020-03-08 by Jie2GG