VoodooI2C/VoodooI2C #223: ELAN 2202 drag or copy/paste not working

tmbt78, 2019-10-24:
I've done some research and i can say that since the introduction of the Native gesture engine  in 2.1 my trackpad never worked without this bug (the last working version is 2.0.3) . I will attach a video i created using FingerMGMT to better show the problem.

[Error.mp4.zip](https://github.com/alexandred/VoodooI2C/files/3768215/Error.mp4.zip)

i'm right handed so at first you see the movement of my right finger on the trackpad
after few seconds suppose i want to select something so i left click
then i start moving the cursor over the text i want to select
then i release ONLY my right finger so i'm still pressing the left button BUT as you can see for the trackpad both fingers disappear.
This happens only when i left click and my right finger is already on the trackpad. If i first left click and then i start moving the cursor around with my right finger this doesn't happen.

tmbt78, 2019-10-25:
@ben9923 seems to have understood the issue : 
I have a guess about your issue - The multitouch engine assigns the click 'property' to the first finger in the array.
https://github.com/alexandred/VoodooI2CHID/blob/b52b5a1ec53ec58a372d82b8dbff9255c16e8d3d/VoodooI2CHID/VoodooI2CMultitouchHIDEventDriver.cpp#L166

I can confirm that if i left the "button" finger the drag is still "active" so the engine thinks my dragging finger was in fact the button finger because is the first finger i have on the trackpad

LeeBinder, 2019-12-27:
Hi @tmbt78  - this seems the same as https://github.com/alexandred/VoodooI2C/issues/208

Feel free to upvote that topic and subscribe there, too.

Seems as if we'll have to wait for the full implementation of VoodooInput + VoodooInputEngine into VoodooI2C. That'll take a while according to the main coder, though.

tmbt78, 2020-04-09:
Hello,
problem completely solved in the newer version (thanks to VoodooInput) !

closed 2020-04-09 by tmbt78