VoodooI2C/VoodooI2C #225: FTE 1200 only working after a sleep cycle

nganhquoc95, 2019-10-31:
Hi @alexandred, My trackpad not working with interrupt mode and GPIO mode, so I'm using polling mode. But I have an issue with my trackpad, it just only working after a sleep cycle. Any idea about this issue? Thank you!

ben9923, 2019-12-18:
Is this still an issue with latest release?
Can you attach a troubleshooting archive, otherwise? This is usually an ACPI config issue.

nganhquoc95, 2019-12-20:
Yes. For release 2.2 and latest release also.
This is the full report of my PC ([Archive.zip](https://github.com/alexandred/VoodooI2C/files/3988389/Archive.zip))
Thanks for help.

ben9923, 2019-12-20:
> Yes. For release 2.2 and latest release also.
> This is the full report of my PC ([Archive.zip](https://github.com/alexandred/VoodooI2C/files/3988389/Archive.zip))
> Thanks for help.

Why do you have VoodooI2CFTE.kext, VoodooI2CSynaptics.kext and VoodooI2CUPDDEngine.kext?
You cannot simply install every kext you see, only install what you need.
Start by removing those, you only need VoodooI2C + VoodooI2CHID.
If necessary (still have the issue), attach a fresh archive. Notice the one you attached has an empty kernel log.

nganhquoc95, 2019-12-20:
I was removed them, just only keep the VoodooI2C + VoodooI2CHID, but I get the same issues. Sorry about archive file. I have a fresh ([Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990722/Archive.zip))

ben9923, 2019-12-21:
> I was removed them, just only keep the VoodooI2C + VoodooI2CHID, but I get the same issues. Sorry about archive file. I have a fresh ([Archive.zip](https://github.com/alexandred/VoodooI2C/files/3990722/Archive.zip))

Can you also share an archive made with 2.2?
Why aren't you using GPIO, exactly? Which pins did you try?

If there's an available BIOS update, perform it. You'll need to re-extract and re-patch DSDT.

nganhquoc95, 2019-12-22:
They are the same, I face the same issue with two releases. I used with the GPIO mode before I used the Polling mode. 
Which pins I was used are `0x6D`, `0x55`, `0x41`, `0x17`, `0x1B`, `0x34`, but these are not working for me. The Trackpad layout was shown in the GPIO mode, but I can not use my Trackpad. I also checked the `Pointer Control` in the `Accessibility`, but my trackpad does not work. So I have used Polling mode from there, but the Polling mode is only working after the sleep cycle completed.

fredwilliamtjr, 2020-02-03:
Hello guys, good night, I am part of the Olalira community in Brazil, with the help of MaLd0n, we set up the EFI of my notebook. it is an Asus VivoBook X512FJ-EJ228T, it all works except the tackpad that is an ELAN1200 model, it only works if I put the computer to sleep, when he wakes up the trackpad is working normally. exactly like this issue.

kprinssu, 2020-05-15:
Please use the latest version of VoodooI2C. Closing as there haven't been any responses in a while. Please create a new issue if you experience any issues.

closed 2020-05-15 by kprinssu