VoodooI2C/VoodooI2C #226: comet lake  Intel(R) 400 Series Chipset Farmily   Request update support this device

fpx-one, 2019-11-21:
1.Intel(R) Serial 10 GPIO Host Controller - INT34BB

2.Intel(R) Serial 10 I2C Host Controller - 02E8

3.Intel(R) Serial I0 I2C Host Controller - 02E9

4.LPC Contoller .0284 for Intel(R) 400 Series Chipset Farmily 

5.comet lake  i5 10210U/ i7 10710U /i7 10510u 

Request update support this device 

The i5 device has successfully installed Mac OS 10.15.1

fpx-one, 2019-11-21:
@alexandred @coolstar @RehabMan @nuudles @

fpx-one, 2019-11-21:
https://github.com/daliansky/XiaoXinPro-13-2019-hackintosh

ben9923, 2019-12-16:
`pci8086,2e8` and `pci8086,2e8` were added to v2.3.
Keeping this issue open until someone actually manages to make the driver work...

neblogas, 2020-02-09:
i7 10510U
Lenovo S540 14-IML
pci8086,2e8
pci8086,2e9
pci8086,2ea - added into 2.3 info.plist
No trackpad at all.
I think it's related to GPI0 pinning but I can't figure it out.

neblogas, 2020-02-17:
@fpx-one there is a patch https://www.notion.so/Using-VoodooI2C-on-comet-lake-cpu-e-g-i5-10210u-142930887087445eaa533120455da5dc

kprinssu, 2020-03-25:
Added supported in https://github.com/alexandred/VoodooI2C/pull/246.

closed 2020-03-25 by kprinssu