VoodooI2C/VoodooI2C #230: This version of VoodooI2C can work with ELAN1200

lunjielee, 2019-12-15:
However I can't use my trackpad after upgrading to VoodooI2C 2.3
pci8086,a368 8750H

[Archive.zip](https://github.com/alexandred/VoodooI2C/files/3965184/Archive.zip)

ben9923, 2019-12-16:
You'll need to attach a full troubleshooting archive.

lunjielee, 2019-12-19:
I tried to load INT3450 under VoodooGPIOSunrisePointH and VoodooGPIOSunrisePointLP by modifying Info.plist inside VoodooGPIO.kext and now problem solved
[image]

ben9923, 2019-12-19:
> I tried to load INT3450 under VoodooGPIOSunrisePointH and VoodooGPIOSunrisePointLP by modifying Info.plist inside VoodooGPIO.kext and now problem solved
> [image]

This configuration is 100% invalid. INT3450 must only match CNL-H. Pin mapping is different for each platform.
Impossible to debug without appropriate troubleshooting files, and unmodified (correct) kexts...

lunjielee, 2019-12-20:
> This configuration is 100% invalid. INT3450 must only match CNL-H. Pin mapping is different for each platform.
> Impossible to debug without appropriate troubleshooting files, and unmodified (correct) [...]

Sorry about the wait, here is my Troubleshooting files, hope this would help with solve the problem
[Archive.zip](https://github.com/alexandred/VoodooI2C/files/3986611/Archive.zip)

lunjielee, 2019-12-20:
> This configuration is 100% invalid. INT3450 must only match CNL-H. Pin mapping is different for each platform.
> Impossible to debug without appropriate troubleshooting files, and unmodified (correct) [...]

btw I checked the log and found that polling mode works perfect but not GPIO
[image]

ben9923, 2019-12-20:
> Sorry about the wait, here is my Troubleshooting files, hope this would help with solve the problem
> [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3986611/Archive.zip)

Another unsupported configuration here - you have multiple, unnecessary satellite kexts (For example, VoodooI2CFTE).
Remove everything besides VoodooI2C and VoodooI2CHID.
Currently VoodooI2CELAN is attached to your device, while you should be using HID.

lunjielee, 2019-12-20:
> > > > I tried to load INT3450 under VoodooGPIOSunrisePointH and VoodooGPIOSunrisePointLP by modifying Info.plist inside VoodooGPIO.kext and now problem solved
> > > > [image]
> > > 
> > > 
> > > This configuration is 100% invalid. INT3450 must only match CNL-H. Pin mapping is different for each platform.
> > > Impossible to debug without appropriate troubleshooting files, and unmodified (correct) kexts...
> > 
> > 
> > Sorry about the wait, here is my Troubleshooting files, hope this would help with solve the problem
> > [Archive.zip](https://github.com/alexandred/VoodooI2C/files/3986611/Archive.zip)
> 
> Another unsupported configuration here - you have multiple, unnecessary satellite kexts (For example, VoodooI2CFTE).
> Remove everything besides VoodooI2C and VoodooI2CHID.
> Currently VoodooI2CELAN is attached to your device, while you should be using HID (You have ELAN1200).
> 
> If you still experience issues after fixing this, attach a more comprehensive archive using this:
> https://github.com/black-dragon74/OSX-Debug
> Make sure IOReg is included.

New debug log:
[debug_18158.zip](https://github.com/alexandred/VoodooI2C/files/3989190/debug_18158.zip)

ben9923, 2019-12-20:
> New debug log:
> [debug_18158.zip](https://github.com/alexandred/VoodooI2C/files/3989190/debug_18158.zip)
Your ACPI patching is wrong. I believe there is no bug, and GitHub issues is not the correct place for such troubleshooting.
You cannot have 2 DSDTs, values in _CRS are wrong and it's better to just patch `TPDM` to be equal to Zero than patching _CRS (To eliminate the need to patch `_INI` as well, and it's also quite simpler).
To continue troubleshooting please join the Gitter chat, there's a link in the README.
Keeping the issue open for now.

closed 2020-04-09 by ben9923