VoodooI2C/VoodooI2C #231: Version 2.3 make my laptop panic after wake

mrhuy92, 2019-12-18:
I'm using Dell Percision 5510 same XPS 9550 and using VoodooI2C 2.2, after update to version 2.3, when computer wakeup it got panic, panic report show the error came from VoodooI2C version 2.3.

LeeBinder, 2019-12-18:
Attach your kernel panic log from **/Library/Logs/DiagnosticReports** as zip and mention ben9923 with an @ right before his nick name.

mrhuy92, 2019-12-18:
[Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip](https://github.com/alexandred/VoodooI2C/files/3980081/Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip)
This is my kernel panic log after updated to version 2.3. @ben9923

ben9923, 2019-12-18:
> [Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip](https://github.com/alexandred/VoodooI2C/files/3980081/Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip)
> This is my kernel panic log after updated to version 2.3. @ben9923

Thank you for reporting :)
Can you please attach a new panic log made with `keepsyms=1` boot flag?

xxxzc, 2019-12-22:
> > [Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip](https://github.com/alexandred/VoodooI2C/files/3980081/Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip)
> > This is my kernel panic log after updated to version 2.3. @ben9923
> 
> Thank you for reporting :)
> Can you please attach a new panic log made with `keepsyms=1` boot flag?
> Also attach your DSDT, and a copy of IOReg made with 2.2.

I am using the same model and have the same issue. @ben9923 
[kernal_panic_xps15_9550_2_3.zip](https://github.com/alexandred/VoodooI2C/files/3992411/kernal_panic_xps15_9550_2_3.zip)

ben9923, 2020-01-04:
> > > [Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip](https://github.com/alexandred/VoodooI2C/files/3980081/Kernel_2019-12-19-000911_MacBook-Pro-cua-Nguyen-Huy.panic.zip)
> > > This is my kernel panic log after updated to version 2.3. @ben9923
> > 
> > 
> > Thank you for reporting :)
> > Can you please attach a new panic log made with `keepsyms=1` boot flag?
> > Also attach your DSDT, and a copy of IOReg made with 2.2.
> 
> I am using the same model and have the same issue. @ben9923
> [kernel_panic_xps15_9550_2_3.zip](https://github.com/alexandred/VoodooI2C/files/3992415/kernel_panic_xps15_9550_2_3.zip)

There's no actual copy of IORegistryExplorer attached...
Please also attach a full archive made with v2.2 (incl. Full ACPI setup, IOReg and kernel log).

LeeBinder, 2020-01-05:
@mrhuy92 : you can use [this improved script](https://www.tonymacx86.com/threads/tool-generate-proper-problem-reporting-files.235953/page-30#post-1881011) for problem reporting, attached at the bottom of that posting.

agrafuese, 2020-03-19:
Hello, myself and another user on insanelymac are having this issue on our Dell XPS 9550 laptops. Our Clover setups are using hotpatching, so I don't know how to provide DSDT files. I can generate the IOREG and crash log, but I am communicating with the other user to see if they are more knowledgeable about the DSDT part in our situation. Please let me know if that part isn't necessary and I will send the other things.

ben9923, 2020-04-09:
Hey,
Can you try v2.4 and see if it still panics?
If so, please attach a new KP log :)

golimpio, 2020-04-11:
> Hey,
> Can you try v2.4 and see if it still panics?
> If so, please attach a new KP log :)

It solved the KP after wake from sleep for me, but it seems that the double tap to drag is less reliable (a lot), it only works half of the time or less. But this is another issue though.

I'm running Catalina 10.15.4 (19E287) / Clover 5109

ben9923, 2020-04-11:
> It solved the KP after wake from sleep for me, but it seems that the double tap to drag is less reliable (a lot), it only works half of the time or less. But this is another issue though.
>
> I'm [...]

Not sure if it's in the VoodooI2C side or the VoodooInput side, but with my ELAN device I'm using 3 finger dragging (Or click to drag) for a while because it's not so reliable.

As the kernel panic issue is resolved, I'll close this issue. If any of other commenters still has such a panic, please re-open the issue.

closed 2020-04-11 by ben9923