VoodooI2C/VoodooI2C #233: Not working fix

Hunter9888x, 2019-12-23:
10.15 Catalina Hackintosh
Open /Applications/Utilities/Terminal

sudo mount -uw /
cd /System/Library/Extensions
sudo rm -rf AppleIntelLpssI2C.kext
sudo rm -rf AppleIntelLpssI2CController.kext
sudo kextcache -i /

install all voodooi2c kext files with a kext installer i use kext wizard https://www.hackintoshzone.com/files/file/17-kext-wizard/
In kext wizard install tab select kexts & install
In kext wizard maintenance tab click System/Library/Extensions & click execute

This fix works for me

LeeBinder, 2019-12-27:
Recommended title:
**[FIX] VoodooI2C not working in Catalina**

ben9923, 2019-12-28:
Those patches are not necessary with VoodooI2C v2.3+. It will load before Apple's I2C kexts.

closed 2019-12-28 by ben9923

Hunter9888x, 2019-12-28:
Acpibatterymanager.kext is necessary to fix trackpad system preferences what are u talking about & yes u must disable apple i2c controller

ben9923, 2019-12-28:
> Acpibatterymanager.kext is necessary to fix trackpad system preferences what are u talking about & yes u must disable apple i2c controller

I know a battery kext is necessary, it's a known issue for all hacks (not just I2C) for years. SMCBatteryManager is another, good alternative.

With version 2.3 you don't have to disable Apple's kexts, VoodooI2C can load before them.

GitHub issues is the place to document & discuss issues with the kext, I could not find any bug with the kext in this issue.

Hunter9888x, 2019-12-28:
I never got voodooii2c kext to work without disabling applei2c controller

ben9923, 2019-12-28:
> I never got voodooii2c kext to work without disabling applei2c controller

Have you tried latest release?
If you can reproduce this with latest release, attach a troubleshooting archive.

Hunter9888x, 2019-12-28:
Voodooi2c 2.3 is version im on

Hunter9888x, 2019-12-28:
For me anyway apple i2c controller gets loaded before voodooi2c

ben9923, 2019-12-28:
Please attach a troubleshooting archive then, we cannot investigate without actual debug info.

fewtarius, 2019-12-28:
@Hunter9888x, you absolutely do not need to disable Apple's kexts via patches or deletes nor do you need Acpibatterymanager.kext for version 2.3.  You do need working battery status but you can accomplish that with VirtualSMC + SMCBattery.  You may also need ForceKextsToLoad IOGraphicsFamily.kext. If you are still having trouble then you should enable the IRQ fixes - FixHPET, FixTMR, FixRTC, and FixIPIC.

You never ever want to touch /S/L/E, and you also don't need to install any kexts in /L/E or /S/L/E.

Hunter9888x, 2019-12-28:
Ben9923 how do i get debug info

fewtarius, 2019-12-28:
@Hunter9888x boot with debug=0x100 -v keepsyms=1 and then follow this guide:

https://github.com/alexandred/VoodooI2C/blob/master/Documentation/Troubleshooting.md

Hunter9888x, 2020-04-11:
@ben9923 version 2.4 fixed my issue with
Apple i2c controller