VoodooI2C/VoodooI2C #235: Panic on detachDevice

jief666, 2019-12-30:
I got this : 
```
Backtrace (CPU 2), Frame : Return Address
0xffffff81facd37c0 : 0xffffff8002a6bf96 mach_kernel : _handle_debugger_trap + 0x4c6
0xffffff81facd3810 : 0xffffff8002b95354 mach_kernel : _kdp_i386_trap + 0x114
0xffffff81facd3850 : 0xffffff8002b875e4 mach_kernel : _kernel_trap + 0x4e4
0xffffff81facd38c0 : 0xffffff8002a1dfb0 mach_kernel : _return_from_trap + 0xe0
0xffffff81facd38e0 : 0xffffff8002a6ba0c mach_kernel : _panic_trap_to_debugger + 0x21c
0xffffff81facd3a10 : 0xffffff8002a6b7cc mach_kernel : _panic + 0x5c
0xffffff81facd3a70 : 0xffffff8002b8780f mach_kernel : _kernel_trap + 0x70f
0xffffff81facd3be0 : 0xffffff8002a1dfb0 mach_kernel : _return_from_trap + 0xe0
0xffffff81facd3c00 : 0xffffff80030662b9 mach_kernel : __ZN15IORegistryEntry16detachFromParentEPS_PK15IORegistryPlane + 0x29
0xffffff81facd3d20 : 0xffffff7f88882d5c com.alexandred.VoodooI2CServices : __ZN17VoodooI2CServices12detachDeviceEPvS0_P9IOServiceP10IONotifier + 0x32
0xffffff81facd3d40 : 0xffffff800306f6a5 mach_kernel : __ZN9IOService14invokeNotifierEP18_IOServiceNotifier + 0xd5
0xffffff81facd3d90 : 0xffffff80030717db mach_kernel : __ZN9IOService15terminatePhase1Ej + 0x2fb
0xffffff81facd3e10 : 0xffffff7f8337380e com.apple.iokit.IOUSBFamily : __ZN14AppleUSBDevice9terminateEj + 0x11c
0xffffff81facd3e70 : 0xffffff7f83374012 com.apple.iokit.IOUSBFamily : __ZN14AppleUSBDevice19terminateThreadCallEP11thread_call + 0xd4
0xffffff81facd3ed0 : 0xffffff8002aa5b74 mach_kernel : _thread_call_delayed_timer + 0xeb4
0xffffff81facd3f40 : 0xffffff8002aa56d5 mach_kernel : _thread_call_delayed_timer + 0xa15
0xffffff81facd3fa0 : 0xffffff8002a1d557 mach_kernel : _call_continuation + 0x17
      Kernel Extensions in backtrace:
         com.alexandred.VoodooI2CServices(1.0)[8D02CFDB-E378-3061-8591-7E199076722D]@0xffffff7f88882000->0xffffff7f88886fff
         com.apple.iokit.IOUSBFamily(900.4.1)[2837DD62-F2B0-3357-A164-D44FEF6810E0]@0xffffff7f83345000->0xffffff7f833dffff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[9C640A56-0FCC-39B4-B469-31F6ED15228E]@0xffffff7f83251000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[93B5E6D7-8F74-39E2-A9C6-6B25F334733D]@0xffffff7f832c4000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[FB0D8863-27B6-39BE-889D-E4A32ED357F8]@0xffffff7f832bc000
```

I put the computer to sleep and left it, while it was in the process to sleep, for the night. Don't know exactly when that happened. It only happened once if few days of usage of my new hackintosh.

ben9923, 2020-04-05:
Hey :)
Please try latest VoodooI2C release and see if this problem still reproduces.

jief666, 2020-05-03:
I don't remember if I have updated, but didn't get the problem. Thanks.

closed 2020-05-03 by jief666