VoodooI2C/VoodooI2C #236: Comet Lake Unknown Synopsys component type: 0xffffffff

thefiredragon, 2020-01-02:
Is there a way to fix Unknown Synopsys component type: 0xffffffff on Comet Lake  i7 1051u?
I could provide logs. 
Think it makes no sense to try patch my DSDT before this is solved or I am wrong here at this point?

best Regards

ben9923, 2020-01-04:
Hey there,
Unfortunately I'm not aware of anyone with 10th Gen hardware who could make VoodooI2C properly load :(
This could be some sort of an ACPI issue that patching (no idea what exactly) might solve.

thefiredragon, 2020-01-09:
Could we debug this?

fewtarius, 2020-01-09:
Several of us have 10th gen hardware and would love to help debug this.  The problem appears to begin within VoodooGPIO.  In the data I provided, you can see it is failing on an interrupt request attempting to instantiate VoodooGPIOCoffeeLakeLP on a comet lake machine.

https://github.com/alexandred/VoodooI2C/issues/234

thefiredragon, 2020-01-10:
Okay thanks, how I can create a debug report?
I don't have a kernel panic when I use voodooi2c perhaps this could be helpful,
best Regards

fewtarius, 2020-01-10:
I'd expect that there will be multiple problems around supporting comet lake, I'm chiming in with a me-to.  You can build a debug package by following this:

https://github.com/alexandred/VoodooI2C/blob/master/Documentation/Troubleshooting.md

3ig, 2020-01-22:
So ... still no solutions for this problem. Does anyone have any progress ?

neblogas, 2020-02-04:
I have the same CPU i7 10510U
Lenovo S540 14
Installer does not boot with VoodooI2C.kext
Without the kext I can boot into installer but the trackpad does not work.
In README I can see that these ID's are fully supported:
Hardware ID:                                                                    PCI\VEN_8086&DEV_**02E8**&SUBSYS_380417AA&REV_00
Hardware ID:                                                                    PCI\VEN_8086&DEV_**02E9**&SUBSYS_380317AA&REV_00

However I have also this one:
Hardware ID:                                                                    PCI\VEN_8086&DEV_**02EA**&SUBSYS_380617AA&REV_00

which is not in the supported list.
Maybe thats the issue? Is there a way to support it?

williambj1, 2020-02-04:
@neblogas 

You can manually add your id in VoodooI2C's info.plist and have a try. The problem appears to be something wrong with the controller (related to code or ACPI) but no the IDs.

neblogas, 2020-02-04:
[image]
I get this when booting with VoodooI2C

thefiredragon, 2020-02-04:
I Don't have Boot panics with voodooi2c it Just dont patch correctly with this error

neblogas, 2020-02-14:
```
2020-02-14 12:49:31.269809+0100 0xc1       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::Loading GPIO Data for CannonLake-LP
2020-02-14 12:49:31.270408+0100 0xc1       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Init!
2020-02-14 12:49:31.270670+0100 0xc1       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-02-14 12:49:31.270900+0100 0xc1       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-02-14 12:49:31.271241+0100 0xc1       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-02-14 12:49:31.271664+0100 0x76       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
2020-02-14 12:49:31.351034+0100 0xf4       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e8 Starting I2C controller
2020-02-14 12:49:31.353167+0100 0xf5       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e9 Starting I2C controller
2020-02-14 12:49:31.356766+0100 0xf4       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e8 Set PCI power state D0
2020-02-14 12:49:31.357060+0100 0xf4       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e8 Publishing nub
2020-02-14 12:49:31.357418+0100 0xf9       Default     0x0                  0      0    kernel: (kernel) VoodooI2CControllerDriver::pci8086,2e8 Probing controller
2020-02-14 12:49:31.357565+0100 0xf5       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e9 Set PCI power state D0
2020-02-14 12:49:31.357810+0100 0xf5       Default     0x0                  0      0    kernel: (kernel) VoodooI2CPCIController::pci8086,2e9 Publishing nub
2020-02-14 12:49:31.357948+0100 0xf9       Default     0x0                  0      0    kernel: (kernel) VoodooI2CControllerDriver::pci8086,2e8 Unknown Synopsys component type: 0xffffffff
2020-02-14 12:49:31.414039+0100 0xfe       Default     0x0                  0      0    kernel: (kernel) VoodooI2CControllerDriver::pci8086,2e9 Probing controller
2020-02-14 12:49:31.419961+0100 0xfe       Default     0x0                  0      0    kernel: (kernel) VoodooI2CControllerDriver::pci8086,2e9 Unknown Synopsys component type: 0xffffffff
```
I just can't figure it out, where the issue is

neblogas, 2020-02-17:
@phu54321 made a patch to work. can somebody test it:
https://www.notion.so/Using-VoodooI2C-on-comet-lake-cpu-e-g-i5-10210u-142930887087445eaa533120455da5dc
@thefiredragon 
@3ig 
@ben9923 
@fewtarius 
@williambj1

3ig, 2020-02-17:
I can confirm your patch works. Marvellous job. My trackpad starter working in pooling mode without any dstd patching.
[image]

thefiredragon, 2020-02-17:
@3ig could you provide your patched voodooi2c?
Polling mode means you only patched windows 10 patch to dsdt and don't change anything ?

Great job! so much thanks.

williambj1, 2020-02-17:
Awesome job!

Confirmed on Lenovo XiaoXin Pro 13'.

[VoodooI2C-10th-patched.zip](https://github.com/alexandred/VoodooI2C/files/4213075/VoodooI2C-10th-patched.zip)

If this built kext violated your @phu54321 rights, please let me know and I'll delete it as soon as possible.

startpenghubingzhou, 2020-02-17:
> @phu54321 made a patch to work. can somebody test it:
> https://www.notion.so/Using-VoodooI2C-on-comet-lake-cpu-e-g-i5-10210u-142930887087445eaa533120455da5dc
> @thefiredragon
> @3ig
> @ben9923
> @fewtarius
> @williambj1

Seems that this patch has sucessfully worked. So  now we need to figure out another question: making this hack only work in 10th because it will make another platforms not work. Tested in 8th KBR and now I ll try to make it.

3ig, 2020-02-17:
> @3ig could you provide your patched voodooi2c?
> Polling mode means you only patched windows 10 patch to dsdt and don't change anything ?
> 
> Great job! so much thanks.

Yes ... my reply was kind of misleading i did hotpatch for Windows 10 (XOSI)

thefiredragon, 2020-02-17:
thanks it's working for me, too ;)

phu54321, 2020-02-17:
> Awesome job!
> 
> Confirmed on Lenovo XiaoXin Pro 13'.
> 
> [VoodooI2C-10th-patched.zip](https://github.com/alexandred/VoodooI2C/files/4213075/VoodooI2C-10th-patched.zip)
> 
> If this built kext violated your @phu54321 rights, please let me know and I'll delete it as soon as possible.

Im okay :) Just quote me

startpenghubingzhou, 2020-02-17:
I have build a new kext based on @phu54321 's work, now I have added a detection about 10th CPU. Can anyone help me to test it?
[VoodooI2C.kext test3.zip](https://github.com/alexandred/VoodooI2C/files/4214256/VoodooI2C.kext.test3.zip)

neblogas, 2020-02-17:
@startpenghubingzhou gonna test it later.

neblogas, 2020-02-17:
@startpenghubingzhou have tested it and it works with windows 10 DSDT patch. Thanks!

I'm attaching some debug files because saw some error messages in system.log
[Send me Tomass-MacBook-Pro.zip](https://github.com/alexandred/VoodooI2C/files/4215322/Send.me.Tomass-MacBook-Pro.zip)

fewtarius, 2020-02-17:
> I have built a new kext based on @phu54321 's work. For now I have added a detection about 10th CPU. Can anyone help me to test it?
> [VoodooI2C.kext test3.zip](https://github.com/alexandred/VoodooI2C/files/4214256/VoodooI2C.kext.test3.zip)

I2CHID doesn't seem to load with just this kext.  Can you folks push this code into a repo somewhere for analysis and testing?

startpenghubingzhou, 2020-02-18:
> have tested it and it works with windows 10 DSDT patch. Thanks!

Yeah good，and for now it's time to call @kprinssu  to merge this code. I ' ll contact with him as soon as posssibles.

startpenghubingzhou, 2020-02-18:
For some laptop which run 14 or later, you can test with this one:
[VoodooI2C test4.zip](https://github.com/alexandred/VoodooI2C/files/4216785/VoodooI2C.test4.zip)

kprinssu, 2020-02-18:
Please create a PR. I don't have merge permissions but I'd love to see what the changes are.

startpenghubingzhou, 2020-02-18:
> Please create a PR. I don't have merge permissions but I'd love to see what the changes are.

Ok I ll make it after breakfast.

startpenghubingzhou, 2020-02-18:
@kprinssu  PR was sent, see https://github.com/alexandred/VoodooI2C/pull/245. Can' t figure out why CircleCI building failed.

startpenghubingzhou, 2020-02-18:
> > I have built a new kext based on @phu54321 's work. For now I have added a detection about 10th CPU. Can anyone help me to test it?
> > [VoodooI2C.kext test3.zip](https://github.com/alexandred/VoodooI2C/files/4214256/VoodooI2C.kext.test3.zip)
> 
> I2CHID doesn't seem to load with just this kext, my device still kernel panics as before. Can you folks push this code into a repo somewhere for analysis and testing?
> 
> If I boot without the I2CHID kext at all I can see that it sets property 0D on startup, finds all of the devices, indicates a GPIO conflict and informs that the satellite kext should work in polling mode.

This may caused by i2chid not i2c so I need KP log to check.

fewtarius, 2020-02-18:
> This may caused by i2chid not i2c so I need KP log to check.

It looks like a VoodooGPIO panic, it's the same one from the issue I reported previously.

[image]

I'll get a fresh pic when I can.

startpenghubingzhou, 2020-02-18:
> It looks like a VoodooGPIO panic, it's the same one from the issue I reported previously.
>
> [image]
>
> I'll get a fresh pic when I can.

Did you use vanilla version of voodooi2c to have a try?

fewtarius, 2020-02-18:
> Did you use vanilla version of voodooi2c to have a try?

Yes.  My VoodooGPIO panic is probably unrelated to this patch.

startpenghubingzhou, 2020-02-19:
> Yes. My VoodooGPIO panic is probably unrelated to this patch.
if you can sure that this panic is unrelated to this patch, maybe opening a new issue is better.

fewtarius, 2020-02-19:
> if you can sure that this panic is unrelated to this patch, maybe opening a new issue is better.

I was hoping they were related, but it's probably not worth opening another issue as I'm using macOS less and less lately.

kprinssu, 2020-02-27:
@fewtarius I haven't worked on the GPIO kext, but panic that looks like an invalid GPIO pin number.

ben9923 maybe more help here.

thefiredragon, 2020-03-25:
on 10.15.3 Trackpad was working with the patch above. 

After upgrade to 10.15.4 trackpad works after sleep.

kprinssu, 2020-04-07:
Closing as Comet Lake support has been merged in with PR https://github.com/alexandred/VoodooI2C/pull/246. Please create new Github issue for any new issues.

closed 2020-04-07 by kprinssu