VoodooI2C/VoodooI2C #237: Clicking and Dragging issue Elan 1300

astone1, 2020-01-09:
I've noticed that after clicking and dragging various UI elements that sometimes there is still a click registered for around a second whenever I am not clicking the trackpad button.  This results in difficulty in highlighting text, dragging various UI elements, etc due to their being a delay in registering whenever a click is done.

Goshin, 2020-02-10:
It's not a bug, it's a feature.

https://support.apple.com/guide/mac-help/change-mouse-trackpad-preferences-unac899/mac

> without drag lock: Double-tap an item, then drag it without lifting your finger after the second tap; when you lift your finger, the item stops moving.
The item can still be dragged for a fraction of a second (so you can reposition your finger if it’s at the edge of the trackpad). To immediately prevent further dragging, tap the trackpad once.

joeyave, 2020-04-05:
> It's not a bug, it's a feature.
>
> https://support.apple.com/guide/mac-help/change-mouse-trackpad-preferences-unac899/mac

is there any way to disable this feature?

closed 2020-04-05 by kprinssu

ben9923, 2020-04-11:
> is there any way to disable this feature?

Use Click (Not tap) to drag or 3 finger drag. It's a feature of double tap dragging as @Goshin said...