VoodooI2C/VoodooI2C #238: Kernel Panic with VoodooPS2Controller in backtrace, MacOS 10.15.2

kasti0, 2020-01-09:
I´m using VoodooI2C + I2CHID alongside with VoodooPS2 + VoodooInput.
Every time I boot my laptop gets a kernel panic for about three times in a row before it can boot successfully. When I remove/ disable the I2C kexts or boot in verbose boot the laptop can always boot on the first time.
See Kernel Panic log below.

`panic(cpu 0 caller 0xffffff801466520a): Kernel trap at 0xffffff7f98923980, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000002ba0c000, CR4: 0x00000000003626e0
RAX: 0xffffff803ab18c90, RBX: 0xffffff8039f78f90, RCX: 0xffffff803adf6380, RDX: 0xffffff803bbe8e00
RSP: 0xffffff812a6c3e90, RBP: 0xffffff812a6c3e90, RSI: 0xffffff7f98923a2e, RDI: 0x0000000000000000
R8:  0xffffff8014ea4490, R9:  0x0000000000000001, R10: 0x00000000000000a7, R11: 0xffffff803aa6d7b0
R12: 0xffffff803adf63d0, R13: 0xffffff803bbe8e00, R14: 0xffffff803bbe8e00, R15: 0xffffff803adf6380
RFL: 0x0000000000010246, RIP: 0xffffff7f98923980, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff812a6c38f0 : 0xffffff801453bb1b mach_kernel : _handle_debugger_trap + 0x47b
0xffffff812a6c3940 : 0xffffff80146733e5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff812a6c3980 : 0xffffff8014664e5e mach_kernel : _kernel_trap + 0x4ee
0xffffff812a6c39d0 : 0xffffff80144e2a40 mach_kernel : _return_from_trap + 0xe0
0xffffff812a6c39f0 : 0xffffff801453b207 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff812a6c3af0 : 0xffffff801453b5eb mach_kernel : _panic_trap_to_debugger + 0x21b
0xffffff812a6c3b40 : 0xffffff8014cd24f9 mach_kernel : _panic + 0x61
0xffffff812a6c3bb0 : 0xffffff801466520a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff812a6c3d30 : 0xffffff8014664f08 mach_kernel : _kernel_trap + 0x598
0xffffff812a6c3d80 : 0xffffff80144e2a40 mach_kernel : _return_from_trap + 0xe0
0xffffff812a6c3da0 : 0xffffff7f98923980 as.acidanthera.voodoo.driver.PS2Controller : __ZN18ApplePS2Controller19notificationHandlerEPvP9IOServiceP10IONotifier + 0x2a
0xffffff812a6c3e90 : 0xffffff8014c16f15 mach_kernel : __ZN9IOService14invokeNotifierEP18_IOServiceNotifier + 0xf5
0xffffff812a6c3ef0 : 0xffffff8014c19c7b mach_kernel : __ZN9IOService14doServiceMatchEj + 0x20b
0xffffff812a6c3f50 : 0xffffff8014c1c4dd mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x15d
0xffffff812a6c3fa0 : 0xffffff80144e213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         as.acidanthera.voodoo.driver.PS2Controller(2.1)[7FDF2328-873A-3EA1-8F5C-48765CBA873A]@0xffffff7f98921000->0xffffff7f98930fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3467C105-9354-3BD2-85F3-E5AEF94BD275]@0xffffff7f95293000

BSD process name corresponding to current thread: kernel_task
Boot args: brcmfx-country=DE keepsyms=1 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64
Kernel UUID: **************************
Kernel slide:     0x0000000014200000
Kernel text base: 0xffffff8014400000
__HIB  text base: 0xffffff8014300000
System model name: MacBookPro15,2 (Mac-827FB448E656EC26)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization

System uptime in nanoseconds: 1215747515
last loaded kext at 1211865844: >!UMergeNub	900.4.2 (addr 0xffffff7f97088000, size 12288)
loaded kexts:
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.3
org.coolstar.VoodooGPIO	1.1
as.acidanthera.voodoo.driver.PS2!C	2.1.0
as.vit9696.SMCProcessor	1.0.9
as.acidanthera.BrcmFirmwareStore	2.5.0
xyz.racermaster.NoTouchID	1.0.3
as.lvs1974.AirportBrcmFixup	2.0.4
as.vit9696.!AALC	1.4.4
ru.usrsse2.SMCBatteryManager	1
as.vit9696.VirtualSMC	1.0.9
as.vit9696.WhateverGreen	1.3.5
org.vanilla.driver.CPUFriend	1.1.9
as.vit9696.Lilu	1.4.0
com.alexandred.VoodooI2CServices	1
>!AFileSystemDriver	3.0.1
>!AVirtIO	1.0
@filesystems.hfs.kext	522.0.9
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
>!ATopCaseHIDEventDriver	3420.1
@filesystems.apfs	1412.61.1
>AirPort.Brcm4360	1400.1.1
>AirPort.BrcmNIC	1400.1.1
>!ASmartBatteryManager	161.0.0
>!AAHCIPort	341.0.2
>!ARTC	2.0
>!AHPET	1.8
>!AACPIButtons	6.1
>!AACPIEC	6.1
>!ASMBIOS	2.1
>!AAPIC	1.7
$!AImage4	1
@nke.applicationfirewall	303
$TMSafetyNet	8
@!ASystemPolicy	2.0.0
|EndpointSecurity	1
>!UMergeNub	900.4.2
|IOAudio!F	300.2
@vecLib.kext	1.2.0
|IOSerial!F	11
|IOSurface	269.6
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AHS!BDriver	3420.1
>IO!BHIDDriver	7.0.2f4
|IO!B!F	7.0.2f4
|IO!BPacketLogger	7.0.2f4
>!AActuatorDriver	3420.2
>!AMultitouchDriver	3420.2
>!AInputDeviceSupport	3420.4
>!AXsanScheme	3
|IOAHCIBlock!S	316.40.3
|IO80211!F	1200.12.2b1
>mDNSOffloadUserClient	1.0.1b8
>corecapture	1.0.4
|IOSkywalk!F	1
|IONVMe!F	2.1.0
>!AEFINVRAM	2.1
>!A!ILpssI2C	3.0.60
>!A!ILpssGspi	3.0.60
|IOAHCI!F	290.0.1
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFIRuntime	2.1
|IOGraphics!F	569.3
|IOHID!F	2.0.0
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|IOSMBus!F	1.1
>DiskImages	493.0.0
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!AKeyStore	2
>!UTDM	489.60.3
|IOSCSIBlockCommandsDevice	422.0.2
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.40.7
|IOSCSIArchitectureModel!F	422.0.2
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	800.14
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

`

@fewtarius

fewtarius, 2020-01-09:
Thanks for tagging me.  @ben9923, @kasti0 and I were talking about this issue today and it appears to be similar to or the same as https://github.com/acidanthera/bugtracker/issues/475.  @kasti0 is using the latest VoodooPS2Controller, the latest VoodooInput, and the latest VoodooI2C kexts and has this panic.  @alexandred mentioned that it has been fixed, but there may still be a defect somewhere.

ErrorErrorError, 2020-02-19:
Any news on this issue? The only way to avoid this kernel panic is by pressing `esc` and `spacebar` repeatedly on boot or load VoodooI2C and the other necessary kexts after the system is booted.

kasti0, 2020-02-20:
No there aren’t any news on this unfortunately. I get around the issue by just pressing and holding any button during boot.

kasti0, 2020-02-24:
@fewtarius @ben9923 the newest VoodooPS2 update (2.1.2) seems to fix the issue. I’m not sure if I should just close this issues or if there still may be something wrong with VoodooI2C. Hope you guys can give me more information on what to do.

xMuu, 2020-02-25:
> @fewtarius @ben9923 the newest VoodooPS2 update (2.1.2) seems to fix the issue. I’m not sure if I should just close this issues or if there still may be something wrong with VoodooI2C. Hope you guys can give me more information on what to do.

I have the same problem and I got it fixed after using the 2.1.2.

ben9923, 2020-04-05:
Closing as the issue wasn't VoodooI2C's and resolved on VoodooPS2 side :)

closed 2020-04-05 by ben9923