VoodooI2C/VoodooI2C #240: Need help with this error when patching DSDT

trankimlonghai, 2020-01-17:
Hi, I was on my way in patching DSDT to make my trackpad HID work. But I got this error, how to solve this ?
My specs: i7 9750H, GTX 1660Ti, UHD630, 16GBRAM.... (Rog Strix G - G531GU)

https://imgur.com/a/d5iE2XB

I'm using both kext VoodooI2C and VoodooI2CHID, but I also noticed that VoodooI2CHID also not be installed ?

williambj1, 2020-01-17:
@trankimlonghai 

https://github.com/williambj1/Hackintosh-EFI-Asus-Zephyrus-S-GX531/blob/master/Shared/ACPI/SSDT-I2CBus.dsl

This may help you, seems that Asus is using a very similar motherboard on all the gaming product line, this EFI has confirmed to work in GX701 as well.

Since Coffee Lake, the hardcoded values from Broadwell in Linux for I2C bus speeds are no longer suitable. Asus offers a variable called `USTP` that could enable the native bus speed values given by the BIOS, however `USTP` is disabled by default and requires a patch for DSDT to enable it. I used a rename in the bootloader with an SSDT.

trankimlonghai, 2020-01-17:
> @trankimlonghai
>
> https://github.com/williambj1/Hackintosh-EFI-Asus-Zephyrus-S-GX531/blob/master/Shared/ACPI/SSDT-I2CBus.dsl
>
> This may help you, seems that Asus is using a very similar motherboard on [...]

Whoa, I don't know about this, but.. what a surprise anyway. I will have a try on this. Thanks a lot on your quick reply and support. I will report later ^^

trankimlonghai, 2020-01-17:
> @trankimlonghai
>
> https://github.com/williambj1/Hackintosh-EFI-Asus-Zephyrus-S-GX531/blob/master/Shared/ACPI/SSDT-I2CBus.dsl
>
> This may help you, seems that Asus is using a very similar motherboard on [...]

Report ! Happy to say that it worked ! Thank you very much ! <3 Have a nice day ^^

trankimlonghai, 2020-01-17:
@williambj1  hi, can I ask you one more thing ?
In your CLOVER made for rog strix, how to control AURA mode ? I tried using your CLOVER and modified some, and everything seems work perfectly but about HDMI I'm not tested it yet, but abotu AURA control mode, I tried Fn+ Left and Right arrow (Also apply patch Fn - Media arrow key in AsusSMC patch of hiepvip). How can I fix this ? :''P

williambj1, 2020-01-17:
@trankimlonghai 

Please don't post off-topic contents here.

## I strongly suggest you delete your comment above. And I will delete this comment once you deleted yours.

If you have more questions ask me in an issue in my repo or in Gitter, **thank you for your cooperation**

closed 2020-03-25 by kprinssu