VoodooI2C/VoodooI2C #243: Clicking and Dragging (I2CHID)

aspring14, 2020-02-09:
When clicking down on the trackpad and letting go the OS still thinks that it's held down for about 2-3 seconds which makes highlighting text and dragging difficult.

Goshin, 2020-02-10:
It's not a bug, it's a feature.

https://support.apple.com/guide/mac-help/change-mouse-trackpad-preferences-unac899/mac

> without drag lock: Double-tap an item, then drag it without lifting your finger after the second tap; when you lift your finger, the item stops moving.
The item can still be dragged for a fraction of a second (so you can reposition your finger if it’s at the edge of the trackpad). To immediately prevent further dragging, tap the trackpad once.

aspring14, 2020-02-11:
This is a different thing. On this, my finger will still be on the trackpad and it will still be dragging for about **2-3** seconds when I stop pressing down on it, but with my finger still on the trackpad.

Goshin, 2020-02-11:
I believe it is the same thing. I can reproduce what you described. Any kind of dragging with the touchpad will be locked for a while.

Why not take off your finger when finishing dragging? You can check out other solutions in the documentation.

closed 2020-02-20 by aspring14