VoodooI2C/VoodooI2C #247: "Print Screen" Button

Ab2774, 2020-02-22:
Hi
There's a button in my keyboard called "Print Screen" or "PrtSc" as shown in the photo
when I click that button my trackpad get disabled completely until I press that button again 
Is there a way to disable that button or at least make it work with "Fn" keys just like the brightness keys 
Thanks.
[image]

Goshin, 2020-02-22:
Are you using the VoodooPS2Controller for your keyboard? It is a "feature" of VoodooPS2Controller.

Try to remap the PS/2 scan code for the `Print Screen` key. Check out the tutorial here.
[How to Use Custom Keyboard Mapping](https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/wiki/How-to-Use-Custom-Keyboard-Mapping)

Ab2774, 2020-02-23:
Thank you so much, I've done some searching until I found this:
https://github.com/acidanthera/VoodooPS2/blob/master/Docs/ACPI/SSDT-PrtSc-F13.dsl
I used this SSDT and it's much easier
Thanks anyway

closed 2020-02-23 by Ab2774