VoodooI2C/VoodooI2C #250: Strange lines (input?) at boot, during uptime (in console) and at shutdown in 10.15.3 on a Lenove Ideapad s540-14IWL

Hasodikis, 2020-03-03:
On a lenovo ideapad s540-14iwl with an I5-8265U. 
At startup, during operation and at shut down we have a constant (ghost?) input from VoodooI2CMT2SimulatorDevice. 
We managed to get the Touchpad to  work normaly, but because of this never ending imput CPU consumes 3-4 times more power at idle and cpu temps are u 6-7 degrees.
It goes like this :
default 10:09:18.246438+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 681, current y is: 671, and finger number is 0
default 10:09:18.252377+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 680, current y is: 683, and finger number is 0
default 10:09:18.258327+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 680, current y is: 687, and finger number is 0
default 10:09:18.264343+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 679, current y is: 690, and finger number is 0
default 10:09:18.270267+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 679, current y is: 690, and finger number is 0
default 10:09:18.276323+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 678, current y is: 693, and finger number is 0
default 10:09:18.282270+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 677, current y is: 696, and finger number is 0
default 10:09:18.288121+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 676, current y is: 699, and finger number is 0
default 10:09:18.294213+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 676, current y is: 701, and finger number is 0
default 10:09:18.300014+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 676, current y is: 701, and finger number is 0
default 10:09:18.306049+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 676, current y is: 703, and finger number is 0
default 10:09:18.311899+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 675, current y is: 704, and finger number is 0
default 10:09:18.317924+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 675, current y is: 705, and finger number is 0
default 10:09:18.323813+0200 kernel VoodooI2CMT2SimulatorDevice current x is: 675, current y is: 706, and finger number is 0
When you move a finger on touchad.... output position changes
I am attaching relevant files (DSDTs,, kexts, GenI2c screenshot, ioreg, system report, and rtf with info).
This behaviour is confirmed by marianopela an his ideapad s540-14IML (with a 10 gen CPU)
I am attaching relevant files
[VoodooI2C report.zip](https://github.com/alexandred/VoodooI2C/files/4280679/VoodooI2C.report.zip)

kprinssu, 2020-03-24:
It seems like you are using a debug version of VoodooI2C. Can you please try downloading from the release page and see if those lines show up in the console?

Hasodikis, 2020-03-24:
The release version doesn’t work at all.....unfortunately.... in GPIO

Hasodikis, 2020-03-24:
We have to invite marianopela to this. We have the same problem and he gave me the kexts that worked with the problem. 

Now I am using the normal kexts in poling mode.

marianopela, 2020-03-25:
> It seems like you are using a debug version of VoodooI2C. Can you please try downloading from the release page and see if those lines show up in the console?

It was the first patched release kindly provided by @startpenghubingzhou 
Sadly it's the only one on which @Hasodikis and I were able to run in interrupts mode

kprinssu, 2020-03-25:
@startpenghubingzhou what exactly were the changes in that kext?

marianopela, 2020-04-30:
This can be close, as It's not relevant now, and probably never was

Hasodikis, 2020-04-30:
Mariano... no matter what I tried.. the problem is with the kexts .....
The touchpad works in Interrupt mode (GPIO) . That we now.
 Its works with modified kexts only .... and not with master.
We have to identify the differences and then correct the error. 
It s not that it affects only one person .... it affects a class devices...

marianopela, 2020-04-30:
> The touchpad works in Interrupt mode (GPIO) . That we now.

Sadly I'm quite sure I was mistaken that one time, as I retried with those same kexts and verified it was really running in polling mode. Either our devices have a buggy GPIO, or there is something else we still need to consider, but it's not related to those old kexts

Hasodikis, 2020-04-30:
If you try it with the newest official kexts (2.4.2) and the TPAD and GIO SSDTs that the touchpad is recognised being in GPIO mode. 
The furthest I have gone is a tiny movement at the login for less than a second before it stops working.  
It' s either the ACPI pin or the Kexts...

Hasodikis, 2020-04-30:
I am trying one more time with this https://translate.google.com/translate?sl=zh-CN&tl=en&u=https%3A%2F%2Fwww.penghubingzhou.cn%2F2019%2F01%2F06%2FVoodooI2C%2520DSDT%2520Edit%2F

kprinssu, 2020-05-21:
@Hasodikis Please create a new issue with the troubleshooting archive.

closed 2020-05-21 by kprinssu