VoodooI2C/VoodooI2C #251: In the case of GPIO interruption, ELAN1200 could not continue to detect the trackpad after detecting 5 fingers

Jie2GG, 2020-03-08:
Hi developer, sorry to interrupt. 

My trackpad device is  ELAN1200, using version 2.3 of VoodooI2C And work in GPIO interrupt mode. But I using app “FingerMgmt” find trackpad device abnormal work. The specific performance is as follows:

* Sliding the trackpad with one to four fingers didn't always respond to my actions
* When I use my 5 fingers, the trackpad will always recognize my fingers and stop responding. Sleep doesn't return until Notebook computer wake up

This is the log and ACPI patch information:

* [ASUS S5300FN.ioreg](https://github.com/Jie2GG/Hackintosh-ASUS-S5300FN/blob/Other/ASUS%20S5300FN.ioreg?raw=true)
* [DSDT(orgin).aml](https://github.com/Jie2GG/Hackintosh-ASUS-S5300FN/blob/Other/DSDT(orgin).aml?raw=true)
* [SSDT-ETPD.aml](https://github.com/Jie2GG/Hackintosh-ASUS-S5300FN/blob/Other/SSDT-ETPD.aml?raw=true)
* [VoodooI2C.log](https://raw.githubusercontent.com/Jie2GG/Hackintosh-ASUS-S5300FN/Other/VoodooI2C.log)

best-sda, 2020-03-21:
Hi, I have a similar problem, after I clicked with 5 fingers, every time I press the trackpad, and one click is recognized as a large number of individual clicks. The device and dsdt and ssdt are the same.

Jie2GG, 2020-03-22:
> Hi, I have a similar problem, after I clicked with 5 fingers, every time I press the trackpad, and one click is recognized as a large number of individual clicks. The device and dsdt and ssdt are the same.

I've had this problem before, and it comes up in polling mode.

soffyo, 2020-03-22:
> > Hi, I have a similar problem, after I clicked with 5 fingers, every time I press the trackpad, and one click is recognized as a large number of individual clicks. The device and dsdt and ssdt are the same.
> 
> I've had this problem before, and it comes up in polling mode.

The problem of single click behaving as multiple clicks has happened to me even in interrupt mode. elan1200.

Jie2GG, 2020-03-22:
> > > Hi, I have a similar problem, after I clicked with 5 fingers, every time I press the trackpad, and one click is recognized as a large number of individual clicks. The device and dsdt and ssdt are the same.
> > 
> > 
> > I've had this problem before, and it comes up in polling mode.
> 
> The problem of single click behaving as multiple clicks has happened to me even in interrupt mode on elan1200. The only fix I've found is #244

Sorry, this solution doesn't work on my computer.

ben9923, 2020-04-29:
Do you still have this issue with latest release?

Jie2GG, 2020-04-30:
> Do you still have this issue with latest release?

Thank you very much, respect developer! Latest VoodooI2C solve my touchpad this isuess. Thank you very much!

Jie2GG, 2020-08-18:
> Do you still have this issue with latest release?

I found a new problem, There are two GPIO pins 0x6d and 0x51 on my trackpad. 
With 0x6d, everything works fine.
At 0x51, however, the above problem occurs.

Jie2GG, 2020-08-18:
I am using the latest VoodooI2C 2.4.4

closed 2020-08-18 by Jie2GG