VoodooI2C/VoodooI2C #252: VoodooI2C doesn't detect more than 3 fingers

jazzysoggy, 2020-03-12:
I have a Synaptics HID-Device, but when I use the HID Satellite kext, only 2 fingers can be detected at a time. The I2C Chipset series is Broadwell, and I'm using the latest VoodooPS2 kext. Everything else is working fine. The laptop model is Acer E5-573G-52G3.

ben9923, 2020-03-14:
It's likely that your Synaptics trackpad only supports limited, 3-finger input in HID mode.
Try VoodooI2CSynaptics instead of VoodooI2CHID, change device name in Info.plist to match yours.

jazzysoggy, 2020-03-15:
The trackpad should be fully HID. According to Windows, it's a HID Synaptics device, and it has Precision drivers which support up to 4 fingers. Also, do you find the device name under Hardware Ids in Windows? The device ID seems to be SYN1B81.

HID\VEN_SYN&DEV_1B81&Col04 is the first line of the hardware ids section

jazzysoggy, 2020-03-15:
I tried the Synaptics trackpad kext, still only can detect 2 fingers.

jazzysoggy, 2020-03-15:
Also, adding another thought is that the trackpad also has a PS2 mode, but I've set it so that it's in I2C mode.

jazzysoggy, 2020-03-20:
Managed to get the I2C devices detected using OpenCore. VoodooI2C is much better than VoodooPS2.