VoodooI2C/VoodooI2C #253: panic if you put 5 fingers on the touchpad

deissh, 2020-03-13:
os: Catalina 10.15.3
model: ux310eq

Step-by-step
1. put 5 fingers on the touchpad
2. panic!

```
panic(cpu 0 caller 0xffffff800fc652fa): Kernel trap at 0xffffff7f93187d20, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000260, CR3: 0x0000000025c03000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000005, RCX: 0x0000000000000001, RDX: 0x0000000000000004
RSP: 0xffffff9134c3bdd0, RBP: 0xffffff9134c3be00, RSI: 0xffffff7f9312cd40, RDI: 0x0000000000000000
R8:  0x0000000000000004, R9:  0x0000000000000000, R10: 0xffffff7f910bce28, R11: 0xffffff7f93176ffa
R12: 0xffffff80361aa8c0, R13: 0x0000000000000006, R14: 0xffffff8035f38200, R15: 0x0000000000000000
RFL: 0x0000000000010246, RIP: 0xffffff7f93187d20, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000260, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff9134c3b830 : 0xffffff800fb3bb2b 
0xffffff9134c3b880 : 0xffffff800fc734d5 
0xffffff9134c3b8c0 : 0xffffff800fc64f4e 
0xffffff9134c3b910 : 0xffffff800fae2a40 
0xffffff9134c3b930 : 0xffffff800fb3b217 
0xffffff9134c3ba30 : 0xffffff800fb3b5fb 
0xffffff9134c3ba80 : 0xffffff80102d2aa9 
0xffffff9134c3baf0 : 0xffffff800fc652fa 
0xffffff9134c3bc70 : 0xffffff800fc64ff8 
0xffffff9134c3bcc0 : 0xffffff800fae2a40 
0xffffff9134c3bce0 : 0xffffff7f93187d20 
0xffffff9134c3be00 : 0xffffff7f9311d9f7 
0xffffff9134c3be50 : 0xffffff7f9317914f 
0xffffff9134c3bea0 : 0xffffff7f9106f944 
0xffffff9134c3bf30 : 0xffffff7f9106e751 
0xffffff9134c3bf70 : 0xffffff7f9317c1f9 
0xffffff9134c3bfa0 : 0xffffff800fae213e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[F14EC4BD-41DF-30E4-9C76-03E9A4F17C24]@0xffffff7f9105e000->0xffffff7f910fcfff
            dependency: com.apple.iokit.IOReportFamily(47)[988360A2-2E10-3014-A119-BE81BC045A10]@0xffffff7f91057000
         com.alexandred.VoodooI2C(2.2)[C4D440B9-4E95-3D3C-8A6F-ACCF068F771E]@0xffffff7f93117000->0xffffff7f93144fff
            dependency: com.alexandred.VoodooI2CServices(1)[8D02CFDB-E378-3061-8591-7E199076722D]@0xffffff7f9305e000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3D78401B-5D2D-33BC-9E41-DD2164EA874D]@0xffffff7f90596000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[F14EC4BD-41DF-30E4-9C76-03E9A4F17C24]@0xffffff7f9105e000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[ADD485B5-3EF8-37C4-B3C5-F86326E497A4]@0xffffff7f9052f000
            dependency: org.coolstar.VoodooGPIO(1.1)[F56452A5-3644-3927-832D-74D30E0617B6]@0xffffff7f93101000
         com.alexandred.VoodooI2CHID(1.0)[FAC2BB90-0D1D-3180-AD9D-98BC6F64344A]@0xffffff7f93176000->0xffffff7f93186fff
            dependency: com.alexandred.VoodooI2C(2.2)[C4D440B9-4E95-3D3C-8A6F-ACCF068F771E]@0xffffff7f93117000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3D78401B-5D2D-33BC-9E41-DD2164EA874D]@0xffffff7f90596000
            dependency: com.apple.iokit.IOGraphicsFamily(569.4)[1F9B5D88-52DB-3A16-8373-4F608A3CB2D8]@0xffffff7f9144c000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[F14EC4BD-41DF-30E4-9C76-03E9A4F17C24]@0xffffff7f9105e000
         com.blankmac.VoodooI2CUPDDEngine(1.0)[B475F4D9-2398-3DBA-90E0-84F56B4A70FB]@0xffffff7f93187000->0xffffff7f93188fff
            dependency: com.alexandred.VoodooI2C(2.2)[C4D440B9-4E95-3D3C-8A6F-ACCF068F771E]@0xffffff7f93117000

BSD process name corresponding to current thread: kernel_task
Boot args: -v dart=0 darkwake=0 -disablegfxfirmware "Kernel=/System/Library/Kernels/kernel"  -

Mac OS version:
19D76

Kernel version:
Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64
Kernel UUID: A8DDE75C-CD97-3C37-B35D-1070CC50D2CE
Kernel slide:     0x000000000f800000
Kernel text base: 0xffffff800fa00000
__HIB  text base: 0xffffff800f900000
System model name: iMac14,2 (Mac-27ADBB7B4CEE8E61)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 351429748315
last loaded kext at 135753363167: @filesystems.smbfs	3.4.1 (addr 0xffffff7f9319c000, size 446464)
last unloaded kext at 78066353174: ru.usrsse2.SMCLightSensor	1 (addr 0xffffff7f93063000, size 16384)
loaded kexts:
org.tw.CodecCommander	2.7.1
com.realtek.driver.RtWlanU1827	1827.4.b36
com.blankmac.VoodooI2CUPDDEngine	1
org.rehabman.voodoo.driver.PS2Keyboard	2.0.1
com.alexandred.VoodooI2CHID	1
ru.joedm.SMCSuperIO	1.1.1
com.alexandred.VoodooI2C	2.2
org.coolstar.VoodooGPIO	1.1
org.rehabman.driver.FakePCIID	1.3.14
org.rehabman.driver.!ASmartBatteryManager	1.90.1
org.rehabman.voodoo.driver.PS2!C	2.0.1
org.rehabman.driver.VoodooTSCSync	1.5.0
as.vit9696.SMCProcessor	1.1.1
xyz.racermaster.NoTouchID	1.0.2
as.vit9696.WhateverGreen	1.3.3
com.rehabman.driver.USBInjectAll	0.7.1
ru.usrsse2.SMCLightSensor	1
com.alexandred.VoodooI2CServices	1
as.lvs1974.HibernationFixup	1.3.2
ru.usrsse2.SMCBatteryManager	1
as.vit9696.VirtualSMC	1.1.1
as.vit9696.!AALC	1.4.7
as.vit9696.Lilu	1.3.8
org.tgwbd.driver.NullCPUPowerManagement	1.0.0d2
@filesystems.smbfs	3.4.1
>AudioAUUC	1.70
>!AHDAHardwareConfigDriver	283.15
@fileutil	20.036.15
@filesystems.msdosfs	1.10
@filesystems.autofs	3.0
>!AHDA	283.15
@AGDCPluginDisplayMetrics	4.7.2
>!AUpstreamUserClient	3.6.8
>!AMCCSControl	1.13
>!AHV	1
|IOUserEthernet	1.0.1
|IO!BSerialManager	7.0.3f5
@private.KextAudit	1.0
>ACPI_SMC_PlatformPlugin	1.0.0
>!A!ISKLGraphics	14.0.4
>eficheck	1
>pmtelemetry	1
@Dont_Steal_Mac_OS_X	7.0.0
>!A!ISlowAdaptiveClocking	4.0.0
>!A!IPCHPMC	2.0.1
>!A!ISKLGraphicsFramebuffer	14.0.4
>!ATopCaseHIDEventDriver	3430.1
>!AVirtIO	1.0
@filesystems.hfs.kext	522.0.9
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@filesystems.apfs	1412.81.1
>!AAHCIPort	341.0.2
>!AACPIButtons	6.1
>!ARTC	2.0
>!AHPET	1.8
>!ASMBIOS	2.1
>!AACPIEC	6.1
>!AAPIC	1.7
$!AImage4	1
@nke.applicationfirewall	303
$TMSafetyNet	8
@!ASystemPolicy	2.0.0
|EndpointSecurity	1
@kext.triggers	1.0
>DspFuncLib	283.15
@kext.OSvKernDSPLib	529
>!ASMBus!C	1.0.18d1
|IOAVB!F	800.17
>!ASSE	1.0
|IO!BHost!CUSBTransport	7.0.3f5
|IO!BHost!CTransport	7.0.3f5
>IOPlatformPluginLegacy	1.0.0
@!AGPUWrangler	4.7.2
|IOSlowAdaptiveClocking!F	1.0.0
|IONDRVSupport	569.4
>!AHDA!C	283.15
|IOHDA!F	283.15
>IOPlatformPlugin!F	6.0.0d8
@!AGraphicsDeviceControl	4.7.2
|IOAccelerator!F2	438.3.1
@plugin.IOgPTPPlugin	810.1
|IOEthernetAVB!C	1.1.0
|IOSkywalk!F	1
>usb.networking	5.0.0
>usb.!UHostCompositeDevice	1.2
>!AHS!BDriver	3430.1
>IO!BHIDDriver	7.0.3f5
|IO!B!F	7.0.3f5
|IO!BPacketLogger	7.0.3f5
>!AActuatorDriver	3430.1
>!AMultitouchDriver	3430.1
>!AInputDeviceSupport	3430.1
|IOAudio!F	300.2
@vecLib.kext	1.2.0
|IOSerial!F	11
|IOSurface	269.6
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOGraphics!F	569.4
|IOAHCIBlock!S	316.80.1
|IOAHCI!F	290.0.1
|IOUSB!F	900.4.2
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|IOSMBus!F	1.1
>DiskImages	493.0.0
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!AKeyStore	2
>!UTDM	489.80.2
|IOSCSIBlockCommandsDevice	422.0.2
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.40.7
|IOSCSIArchitectureModel!F	422.0.2
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOTimeSync!F	810.1
|IONetworking!F	3.4
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1

```

ben9923, 2020-03-13:
Are you using VoodooI2CUPDDEngine?
If not, remove it.

kprinssu, 2020-05-15:
Closing as @deissh has not replied in two months. Please first update to 2.4.2 and only reopen if the issue persists.

closed 2020-05-15 by kprinssu