VoodooI2C/VoodooI2C #260: Building from source error:Command PhaseScriptExecution failed with a nonzero exit code

tiger511, 2020-03-24:
I've tried to build it from source.I've installed cpplint and custom cldoc.During xcodebuild it fails with following error: 

cd /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C\ Satellites/VoodooI2CFTE
    /bin/sh -c /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C\\\ Satellites/VoodooI2CFTE/build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/Script-121060FE21577A8F00CA19A9.sh
build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c:0:  No copyright message found.  You should have a line: "Copyright [year] <Copyright Owner>"  [legal/copyright] [5]
build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c:2:  Line ends in whitespace.  Consider deleting these extra spaces.  [whitespace/end_of_line] [4]
build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c:5:  Weird number of spaces at line-start.  Are you using a 2-space indent?  [whitespace/indent] [3]
build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c:5:  Line ends in whitespace.  Consider deleting these extra spaces.  [whitespace/end_of_line] [4]
build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c:9:  Extra space before last semicolon. If this should be an empty statement, use {} instead.  [whitespace/semicolon] [5]
Done processing VoodooI2CFTE/VoodooI2CFTETouchpadDriver.hpp
Done processing VoodooI2CFTE/VoodooI2CFTETouchpadDriver.cpp
Done processing VoodooI2CFTE/VoodooI2CFTEConstants.h
Done processing build/VoodooI2CFTE.build/Release/VoodooI2CFTE.build/DerivedSources/VoodooI2CFTE_info.c
Total errors found: 5
Command PhaseScriptExecution failed with a nonzero exit code

** BUILD FAILED **

tiger511, 2020-03-24:
I've added filter "-whitespace,-legal/copyright" to cpplint and got past this erro.But i'm having another error at the end:

WriteAuxiliaryFile /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/VoodooI2C.build/Release/VoodooI2C.build/Script-AC0955921F4EE9B00052E343.sh (in target 'VoodooI2C' from project 'VoodooI2C')
    cd /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C
    write-file /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/VoodooI2C.build/Release/VoodooI2C.build/Script-AC0955921F4EE9B00052E343.sh

PhaseScriptExecution Add\ GPIO\ and\ Services\ Plugin /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/VoodooI2C.build/Release/VoodooI2C.build/Script-AC0955921F4EE9B00052E343.sh (in target 'VoodooI2C' from project 'VoodooI2C')
    cd /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C
    /bin/sh -c /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/VoodooI2C.build/Release/VoodooI2C.build/Script-AC0955921F4EE9B00052E343.sh
mv: rename /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/Release/VoodooGPIO.kext to /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/Release/VoodooI2C.kext/Contents/PlugIns/VoodooGPIO.kext: No such file or directory
mv: rename /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/Release/VoodooI2CServices.kext to /Users/tiger/Downloads/git/VoodooI2C/VoodooI2C/build/Release/VoodooI2C.kext/Contents/PlugIns/VoodooI2CServices.kext: No such file or directory
Command PhaseScriptExecution failed with a nonzero exit code

** BUILD FAILED **

kprinssu, 2020-03-24:
Simple way to bypass is to remove the build phase (remove the bash commands/script from Xcode) and this will allow you to build. If you cannot do this, *please* wait for the next build of VoodooI2C.

I am going to close this as it's not exactly an issue with VoodooI2C, rather an issue with your environment.

closed 2020-03-24 by kprinssu