VoodooI2C/VoodooI2C #263: slow touchpad click state updates

ToonvanStrijp, 2020-03-25:
Alright this issue is hard to explain but I'll do my best.

When dragging windows I click down on the touchpad (clicky sound) and drag the window to the desired location. Then I let go the click by lifting my finger slightly, but the tip of my finger keeps touching the touchpad. It appears that macos still thinks I'm holding it down since for a short amount of time the windows that I was dragging follows along, then after 500ms or so the click state is updated and the windows is released.

This causes a very unpleasant touchpad experience. (Still way better than normally).

The implementation that I tested was [https://github.com/jaromeyer/XPS9570-Catalina](https://github.com/jaromeyer/XPS9570-Catalina)

Goshin, 2020-03-26:
Well, Drag lock is not a bug, it's a feature.

https://support.apple.com/guide/mac-help/change-mouse-trackpad-preferences-unac899/mac

> without drag lock: Double-tap an item, then drag it without lifting your finger after the second tap; when you lift your finger, the item stops moving.
The item can still be dragged for a fraction of a second (so you can reposition your finger if it’s at the edge of the trackpad). To immediately prevent further dragging, tap the trackpad once.

https://voodooi2c.github.io/#Supported%20Gestures/Supported%20Gestures

> Despite everything working as intended, Apple's implementation of tap to drag is (in my humble opinion) poorly implemented and the behaviour is very unusual (especially for those used to Windows or VoodooPS2). If you enable tap to drag without drag lock as per the instructions below, you will notice that there is still a short amount of drag lock that lasts for about 1-1.5 seconds. This is unfortunately unavoidable and occurs even on real macbooks. The best option, in my opinion, is three finger drag. You can reconfigure the usual three finger gestures (such as swiping between spaces) to four finger gestures so that the drag doesn't clash with these.

ToonvanStrijp, 2020-03-27:
Wow haha It sure feels like a bug sometimes, to bad you can't disable this `1-1.5` seconds delay on this.

closed 2020-03-27 by ToonvanStrijp