VoodooI2C/VoodooI2C #265: surface laptop 1 i2c keyboard not working

osxme, 2020-03-29:
I noticed that the keyboard of surface laptop 1 is hid, it does not work under mojave 10.14.6, I have installed voodooi2c, and currently the touchpad is OK. Can you help?

HID \ VEN_MSHW & DEV_0096 & Col01
HID \ MSHW0096 & Col01
HID \ * MSHW0096 & Col01
HID \ VID_045E & UP: 0001_U: 0006
HID_DEVICE_SYSTEM_KEYBOARD
HID_DEVICE_UP: 0001_U: 0006
HID_DEVICE

This is the hardware ID information of Windows

ben9923, 2020-03-30:
If keyboard is indeed connected via I2C, there's a VoodooI2CHID fork that adds keyboard support.
It was not merged to upstream and needs to be rebased on upstream/master, you can give it a try...
https://github.com/alexandred/VoodooI2CHID/pull/3

Very few keyboards are connected via I2C, sure your keyboard is one of them?

kprinssu, 2020-05-15:
@osxme there is no support for I2C based keyboards or cameras. We also do not plan to support I2C keyboards

However, PRs are welcome.

closed 2020-05-15 by kprinssu