VoodooI2C/VoodooI2C #267: Trackpad Strange behavior

llbasso, 2020-04-02:
Hello people,

I'm using macOS Hackintosh on a Lenovo S145-15IWL device with an ELAN trackpad. I am currently on version 10.15.4 Catalina and VoodooIC 2.3.

After installing macOS I noticed a strange behavior with the cursor, for example: When I am typing text the cursor moves to the middle or the beginning of the text and this causes me to have to go back to the current position of the text again .

I don't know if it makes sense, but I'm using a USB mouse as well and I noticed that this behavior happens when I'm using the trackpad and mouse. When I disconnect the USB mouse and just use the trackpad everything works fine. If I want to use the USB mouse I need to disable the trackpad in System Preferences -> Accessibility -> Point Control, "Ignore built-in trackpad when mouse or wireless trackpad is present" option

Can you tell me if someone has the same scenario and faces the same problem?

kprinssu, 2020-04-03:
Please attach troubleshooting files. Please see https://github.com/alexandred/VoodooI2C/blob/master/Documentation/Troubleshooting.md

kprinssu, 2020-05-15:
@llbasso please reopen if you attach the troubleshooting archive. closing as you have not replied in over a month.

closed 2020-05-15 by kprinssu