VoodooI2C/VoodooI2C #273: Voodooinput broke dragging on ELAN1200 polling

brunogermain, 2020-04-06:
Hi, I'm on Mojave with an Asus UX310UA (KabyLake) using Elan1200 trackpad.
Since Voodooinput was introduced in v2.3 with commit 690b16e, dragging and window resizing became very buggy and unusable. Using v2.3 previous to that commit works very well, only freezing every now and then

marianopela, 2020-04-06:
Try these. [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4436919/Archive.zip)

They are build with latest commits on master, and should work just fine

brunogermain, 2020-04-06:
> Try these. [Archive.zip](https://github.com/alexandred/VoodooI2C/files/4436919/Archive.zip)
> 
> They are built with latest commits on master, and should work just fine

Something's broken in the archive. Kexts won't even install correctly. Here's the output:

Invalid signature -67050 for kext <OSKext 0x7f82417405f0 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/VoodooInput.kext/", ID = "me.kishorprins.VoodooInput" }
Invalid signature -67050 for kext <OSKext 0x7f82417376e0 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/VoodooI2CHID.kext/", ID = "com.alexandred.VoodooI2CHID" }
Invalid signature -67050 for kext <OSKext 0x7f824173d5a0 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/VoodooI2CServices.kext/", ID = "com.alexandred.VoodooI2CServices" }
Invalid signature -67050 for kext <OSKext 0x7f8241731140 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/VoodooI2C.kext/Contents/PlugIns/VoodooGPIO.kext/", ID = "org.coolstar.VoodooGPIO" }
Invalid signature -67050 for kext <OSKext 0x7f8241729230 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/VoodooI2C.kext/", ID = "com.alexandred.VoodooI2C" }
Invalid signature -67050 for kext <OSKext 0x7f824166b870 [0x7fff9f4278e0]> { URL = "file:///Library/Extensions/AsusSMC.kext/", ID = "com.hieplpvip.AsusSMC" }
kxld[com.alexandred.VoodooI2CServices]: The super class vtable 'vtable for IOService' for vtable 'vtable for VoodooI2CLogger' is out of date. Make sure your kext has been built against the correct headers.
kxld[com.alexandred.VoodooI2CServices]: The super class vtable 'vtable for IOService' for vtable 'vtable for VoodooI2CServices' is out of date. Make sure your kext has been built against the correct headers.
Link failed (error code 5).
Prelink failed for com.alexandred.VoodooI2CServices; omitting from prelinked kernel.
kxld[org.coolstar.VoodooGPIO]: The super class vtable 'vtable for IOService' for vtable 'vtable for VoodooGPIO' is out of date. Make sure your kext has been built against the correct headers.
kxld[org.coolstar.VoodooGPIO]: The vtable for VoodooGPIOCannonLakeH was not patched because its parent, the vtable for VoodooGPIO, was not patchable.
kxld[org.coolstar.VoodooGPIO]: The vtable for VoodooGPIOCannonLakeLP was not patched because its parent, the vtable for VoodooGPIO, was not patchable.
kxld[org.coolstar.VoodooGPIO]: The vtable for VoodooGPIOSunrisePointH was not patched because its parent, the vtable for VoodooGPIO, was not patchable.
kxld[org.coolstar.VoodooGPIO]: The vtable for VoodooGPIOSunrisePointLP was not patched because its parent, the vtable for VoodooGPIO, was not patchable.
Link failed (error code 5).
Prelink failed for org.coolstar.VoodooGPIO; omitting from prelinked kernel.
Can't use /Library/Extensions/VoodooI2C.kext/Contents/PlugIns/VoodooI2CServices.kext - not linked.
Prelink failed for com.alexandred.VoodooI2C; omitting from prelinked kernel.
Can't use /Library/Extensions/VoodooI2C.kext - not linked.
Prelink failed for com.alexandred.VoodooI2CHID; aborting prelink.
Failed to generate prelinked kernel.
Child process /usr/sbin/kextcache[704] exited with status 71.
Error 107 rebuilding /System/Library/PrelinkedKernels/prelinkedkernel

marianopela, 2020-04-06:
Sorry, I used 10.15 compiler. I'll send you another archive

brunogermain, 2020-04-06:
> My bad, I used 10.15 compiler. I'll send you another archive
> 
> [I2C Mojave.zip](https://github.com/alexandred/VoodooI2C/files/4437064/I2C.Mojave.zip)
> Here you are, please git it a try

They seem to be working .
Thanks!

ben9923, 2020-04-09:
So I guess actual 2.4 release is not affected by this issue? Can it be closed?

brunogermain, 2020-04-10:
> So I guess actual 2.4 release is not affected by this issue? Can it be closed?

Dragging working fine here

closed 2020-04-10 by ben9923