VoodooI2C/VoodooI2C #274: Lenovo V130 driver failing with exception on loading

marcellinus77, 2020-04-06:
Model: Lenovo V130-15IKB / 81HN
Driver is loading but failing with timeout exception on loading

In Polling mode, very basic but strange behavior, kind of working (very slow, cursor randomly start moving without input)

Kernel / Driver Exception:

2020-04-05 05:35:12.328839+0200 0x13e1     Default     0x0                  0      0    kernel: IOCommandGate::disable() called when not gated
2020-04-05 05:35:12.328844+0200 0x13e1     Default     0x0                  0      0    kernel:       Kernel Extensions in backtrace:
2020-04-05 05:35:12.328853+0200 0x13e1     Default     0x0                  0      0    kernel:          com.alexandred.VoodooI2C(2.3)[20247460-5715-3135-A598-F7D673F657B7]@0xffffff7f8347c000->0xffffff7f834adfff
2020-04-05 05:35:12.328875+0200 0x13e1     Default     0x0                  0      0    kernel:             dependency: com.alexandred.VoodooI2CServices(1)[66104E8E-A22B-3AC4-922E-CA959EFFABB5]@0xffffff7f83447000
2020-04-05 05:35:12.328882+0200 0x13e1     Default     0x0                  0      0    kernel:             dependency: com.apple.iokit.IOACPIFamily(1.4)[3E8E9870-2E41-33F6-ACBF-35DDC08F22B5]@0xffffff7f8138e000
2020-04-05 05:35:12.328893+0200 0x13e1     Default     0x0                  0      0    kernel:             dependency: com.apple.iokit.IOHIDFamily(2.0.0)[4D861DA4-FE3A-3D9D-8123-038E9FBFE182]@0xffffff7f80fbb000
2020-04-05 05:35:12.328899+0200 0x13e1     Default     0x0                  0      0    kernel:             dependency: com.apple.iokit.IOPCIFamily(2.9)[BFDEB8D4-50FE-3DDB-87B7-F6A504393830]@0xffffff7f80c95000
2020-04-05 05:35:12.328906+0200 0x13e1     Default     0x0                  0      0    kernel:             dependency: org.coolstar.VoodooGPIO(1.1)[565F1D03-F815-3D35-9B0F-78E2FF4FBFD4]@0xffffff7f83457000
2020-04-05 05:35:12.515472+0200 0x13e1     Default     0x0                  0      0    kernel: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Timed out waiting for bus to change state
[Send me Marcellinuss-iMac.zip](https://github.com/alexandred/VoodooI2C/files/4437695/Send.me.Marcellinuss-iMac.zip)
[ioregs-marcellinus.zip](https://github.com/alexandred/VoodooI2C/files/4437697/ioregs-marcellinus.zip)

ben9923, 2020-04-09:
Can you give v2.4 a try?

marcellinus77, 2020-04-09:
Tried both, HID and Synaptics, bot load but do not work. Tried in polling mode (HID) kind of working but not really useable, also touchpad never shows up in control panel, not even in polling mode

marianopela, 2020-04-10:
> Tried both, HID and Synaptics, bot load but do not work. Tried in polling mode (HID) kind of working but not really useable, also touchpad never shows up in control panel, not even in polling mode

I also had a problem with trackpad not showing in prefpane. I solved by removing all of VoodooPS2 Plugins (VoodooInput included, as it is also part of VoodooI2C atm), apart from VoodooPS2Keyboard

ben9923, 2020-07-14:
Glad it doesn't panic anymore... Any chance latest release works better?
If not, please attach a _full_ troubleshooting archive when it's installed (As you tried both HID and Synaptics - probably two archives).

ben9923, 2020-12-22:
Closing as issue is inactive. Feel free to reopen as necessary.

closed 2020-12-22 by ben9923