VoodooI2C/VoodooI2C #275: VoodooI2C version 2.4 Two finger scroll and Zoom no longer working

JayMonkey, 2020-04-09:
Hi Guys,

Just reporting my findings with the recently released VoodooIC2 (Version 2.4)

I have a HP spectre X360 laptop which has a 4K touch screen and active pen/stylus and have been using VoodooIC2 for a few years now and it always worked great.

I just updated to the recently released version 2.4 of VoodooI2C and the touch screen (single finger) and pen work as before but scrolling (Horizontal and Vertical) requires three fingers where as in all previous versions it only required two fingers (as it should be) ... this also means other two finger gestures (such as zoom) no longer work.

Rolling back to version 2.3 and two finger gestures work again so the issue seems to be with version 2.4.

I seem to remember this being an issue sometime ago where VoodooIC2 treated the stylus as a finger (index 0) so its possible that this is the same bug ?

Also reported this on TMx86 VoodooI2C support thread ...

Cheers
Jay

nguyentdat27, 2020-04-09:
How about double tap and hold to drag, does it work on your HP? In my XPS 15, i can only drag with three fingers

laelsirus, 2020-04-09:
I have the same issue on my Latitude 5290 2-in-1 touchscreen.
Starting with version 2.4, gestures of two or more points are recognized only by mouse movement.

ben9923, 2020-04-09:
Hey @JayMonkey , thanks for reporting :)
Can you try building VoodooI2C using latest master code?
Make sure you’re running the VoodooInput bootstrap command first.

I’ll make a new release if it’s resolved with latest VoodooInput.

JayMonkey, 2020-04-09:
> Hey @JayMonkey , thanks for reporting :)
> Can you try building VoodooI2C using latest master code?
> Make sure you’re running the VoodooInput bootstrap command first.
> 
> I’ll make a new release if it’s resolved with latest VoodooInput.

@ben9923,

Will do and report back ...

JayMonkey, 2020-04-09:
@ben9923,

Built VoodooI2C and VoodooHID from the master repo using the method detailed here :-

[https://voodooi2c.github.io/#Build%20Environment/Build%20Environment](url)

All dependencies (VoodooInput ..etc) installed

Both builds successful ... Installed built kexts .... but issues remans

Not quote sure what you meant by 

> Make sure you’re running the VoodooInput bootstrap command first.

I have the latest  VoodooInput from **acidanthera** installed along with VoodooPS2Controller for the synaptics track pad. I also tried installing the version of VoodooInput from the VoodooI2C dependencies folder but the issue remans.

Cheers
Jay

ben9923, 2020-04-09:
> @ben9923,
> 
> 
> 
> Built VoodooI2C and VoodooHID from the master repo using the method detailed here : -
> 
> 
> 
> [https://voodooi2c.github.io/#Build%20Environment/Build%20Environment](url)
> 
> 
> 
> All dependencies (VoodooInput ..etc) installed
> 
> 
> 
> Both builds successful ... Installed built kexts .... but issue remains
> 
> 
> 
> Not quote sure what you meant by : -
> 
> 
> 
> > Make sure you’re running the VoodooInput bootstrap command first.
> 
> 
> 
> I have the latest  VoodooInput from **acidanthera** installed along with VoodooPS2Controller for the synaptics track pad. I also tried installing the version of VoodooInput from the VoodooI2C dependencies folder but the issue remans.
> 
> 
> 
> Cheers
> 
> Jay
> 
> 
> 
> 

You’re going to need VoodooInput built from source :)

JayMonkey, 2020-04-10:
> You’re going to need VoodooInput built from source :)
> 
> Make sure you don’t have VoodooInput installed twice (“regular” + PlugIns).

@ben9923,

Ok .. built VoodooInput from master repo  : -

[https://github.com/acidanthera/VoodooInput](url)

Copied built VoodooInput.kext and VoodooInput.dSYM to VoodooI2C dependencies

Built VoodooI2 and VoodooHID 

Installed VoodooInput.kext, VoodooI2C.kext and VoodooHID.kext ... rebooted

Issue still remains .... ?  (Two finger gestures don't work)

All builds where successful ....

Cheers
Jay

kprinssu, 2020-04-10:
This could be related to recently added VoodooInput support and VoodooI2CHID may need an extra patch for it work properly. I can reproduce this on my Surface style tablet/laptop.

blankmac, 2020-04-16:
This will be reproducible on any device that presents a stylus on transducer 0.  Prior to VoodooInput, this was corrected in the native multitouch engine with a stylus check.

kprinssu, 2020-04-16:
Thank @blankmac, the VoodooInput integration doesn't have this check and needs to be added in,

blankmac, 2020-04-27:
Fixed.  https://github.com/alexandred/VoodooI2C/pull/295

closed 2020-04-27 by blankmac

kprinssu, 2020-04-28:
@JayMonkey please use 2.4.2 as it contains @blankmac's fixes.