VoodooI2C/VoodooI2C #277: Device patched and detected, but not working - not implemented in ACPI table

elechocruz, 2020-04-09:
Hello, I have a Chinese Avita Clarus 14 laptop and I can't get the trackpad to work, I already have the battery indicator working.
First of all, say that sometimes if the trackpad works (like a Magic Mouse, some gestures and only one click working), usually if I delete all the kext from the clover Other folder and restart, it works, but when you turn off the laptop it stops working again, and sometimes even deleting the kext does not work. This was done before patching or having the battery indicator working and after that too. Check in the LOG and the only thing I found is an error that says:
 "kernel: (kernel) VoodooI2CControllerNub :: pci8086,9d62 SSCN not implemented in ACPI tables
2020-04-09 11: 47: 13.788485-0500 0x114 Default 0x0 0 0 kernel: (kernel) VoodooI2CControllerNub :: pci8086,9d62 FMCN not implemented in ACPI tables ".
I don't know if that's where the problem is, I hope someone can help, thank you very much.

Specs: 
Avita Clarus 14 
i5-7y54, 8Gb, Intel HD 615
My device name is: TPDX
[MacBook de Gerardo.zip](https://github.com/alexandred/VoodooI2C/files/4457722/MacBook.de.Gerardo.zip)
[Log.zip](https://github.com/alexandred/VoodooI2C/files/4457723/Log.zip)
[DSDT.aml.zip](https://github.com/alexandred/VoodooI2C/files/4457728/DSDT.aml.zip)

sorry if i am not very clear but i am new to this it is my first laptop hackintosh.

ben9923, 2020-04-09:
Did you also install VoodooI2CHID?

elechocruz, 2020-04-09:
> Did you also install VoodooI2CHID?

Yes, I have already installed that kext

elechocruz, 2020-04-10:
> Did you also install VoodooI2CHID?

After several reboots my computer detected the trackpad again, now I am sure it is well patched because now I do detect it perfectly.
Take another IOReg to see if someone with more experience finds the difference from when it works to when it doesn't, regarding the bootlog and the kernel log I did not find any difference, even the error described above was still present.

[MacBook de Gerardo working.zip](https://github.com/alexandred/VoodooI2C/files/4461993/MacBook.de.Gerardo.working.zip)

ben9923, 2020-04-10:
> After several reboots my computer detected the trackpad again, now I am sure it is well patched because now I do detect it perfectly.
> Take another IOReg to see if someone with more experience finds [...]

Please attach the output archive of this program:
https://www.olarila.com/files/Utils/RunMe.app.zip

elechocruz, 2020-04-10:
> Please attach the output archive of this program:
> https://www.olarila.com/files/Utils/RunMe.app.zip

Sure, here is the file when is working 
[Send me MacBook-de-Gerardo WORKING.zip](https://github.com/alexandred/VoodooI2C/files/4463128/Send.me.MacBook-de-Gerardo.WORKING.zip)

And here is when is NOT working
[Send me MacBook-de-Gerardo.zip](https://github.com/alexandred/VoodooI2C/files/4463140/Send.me.MacBook-de-Gerardo.zip)

To make it work I did what I mentioned before, I deleted my kext from the Kexts / Other folder, and put them back there, rebuild the kextcache with sudo kextcache -system-prelinked-kernel reboot and it worked, then I rebooted and it no longer worked .

ben9923, 2020-04-10:
> Sure, here is the file when is working
> [Send me MacBook-de-Gerardo WORKING.zip](https://github.com/alexandred/VoodooI2C/files/4463128/Send.me.MacBook-de-Gerardo.WORKING.zip)
>
> And here is when is NOT [...]

You don't have latest v2.4 kexts. Update VoodooI2C + VoodooI2CHID and give it a try.

elechocruz, 2020-04-10:
> You don't have latest v2.4 kexts. Update VoodooI2C + VoodooI2CHID and give it a try.

I tried with 2.4 too and I still have the same problem :(

ben9923, 2020-04-10:
> > You don't have latest v2.4 kexts. Update VoodooI2C + VoodooI2CHID and give it a try.
> 
> I tried with 2.4 too and I still have the same problem :(

Please attach updated archives.

elechocruz, 2020-04-10:
I couldn't get it to work, but here is the file from when it doesn't work in version 2.4
[Send me MacBook-de-Gerardo.zip](https://github.com/alexandred/VoodooI2C/files/4463702/Send.me.MacBook-de-Gerardo.zip)

ben9923, 2020-04-14:
> I couldn't get it to work, but here is the file from when it doesn't work in version 2.4
> [Send me MacBook-de-Gerardo.zip](https://github.com/alexandred/VoodooI2C/files/4463702/Send.me.MacBook-de-Gerardo.zip)

You messed up your _CRS method... Keep it native.

Those "not implemented" warnings you see should be just fine on 6th/7th generation devices.

SSCN & FMCN methods you added to your trackpad device have no effect, they should be under the I2C controller itself. Remove those.

Mateo1234454545, 2020-04-14:
@ben9923 , what about 8th gen Intel i5-8250u?
Should I implement sscn and fmcn ?

ben9923, 2020-09-10:
Closing the issue as it was not active for a few months.
Feel free to re-open as necessary, with an updated troubleshooting archive using latest release.

closed 2020-09-10 by ben9923

manishrech786, 2020-12-27:
i have same issue in avita liber

manishrech786, 2020-12-27:
> Closing the issue as it was not active for a few months.
> Feel free to re-open as necessary, with an updated troubleshooting archive using latest release.

hlp me. brother

kprinssu, 2020-12-27:
@manishrech786 please do not spam closed GitHub issues. Please use Gitter for support.