VoodooI2C/VoodooI2C #278: USB Touchscreen not working if VoodooI2C is injected via bootloader

najdanovicivan, 2020-04-10:
The only way I could make my USB TouchScreen on X270 working is to install VoodooI2C.kext and VoodooI2CHID.kext in /Library/Extensions 

When I try to inject those kexts using either OpenCore or Clover VoodooI2CTouchscreenHIDEventDriver does not load.

kprinssu, 2020-04-25:
Can you double check you are using version 2.3 or newer of VoodooI2C? This should have been fixed since 2018 with a patch that al3xt or coolstar contributed to.

If not, can you upload a copy of your IORegistry and EFI/OC or EFI/Clover folder?

najdanovicivan, 2020-04-27:
Strange thing. I was using version 2.4 installed it couple of weeks ago and noticed the issue,

I've updated to 2.4.1 and injected via OpenCore and it's working now. This is on my Lenovo X270

My friend have Lenovo X240 with Clover install and had exactly the same issue. I'll ask him to try to move kexts to Clover to see if issue can still be replicated and get IORegistry export if it does

kprinssu, 2020-04-27:
Glad to hear 2.4.1 works. I am going to close the issue, please feel free to reopen if it's not working.

closed 2020-04-27 by kprinssu