VoodooI2C/VoodooI2C #280: [BUG] VoodooI2C Trackpad not working in macOS Recovery mode

dreamwhite, 2020-04-11:
As per title I can manage to get Trackpad working while in Recovery Mode on macOS. I'm using latest release of VoodooI2C and trackpad was activated using GPIO Pinning guide.

ben9923, 2020-04-11:
Which satellite are you using?
Have you updated to v2.4?

Please attach your kernel log from Recovery mode ([See command to use here](https://voodooi2c.github.io/#Common%20Errors/Common%20Errors)).

dreamwhite, 2020-04-11:
> Which satellite are you using?

I'm using `VoodooI2CHID.kext` as my trackpad device, `TPD0` has `_CID` `PNP0C50`

> Have you updated to v2.4?

Yeah, just updated the kexts and they're working as charm.

Just a question about Coolstar I2C Patches. Are they still needed or I can disable them? :uhm:

ben9923, 2020-04-11:
> > Which satellite are you using?
> 
> I'm using `VoodooI2CHID.kext` as my trackpad device, `TPD0` has `_CID` `PNP0C50`
> 
> > Have you updated to v2.4?
> 
> Yeah, just updated the kexts and they're working as charm.
> 
> Just a question about Coolstar I2C Patches. Are they still needed or I can disable them? :uhm:

Those patches are not needed since v2.3 :)

So when you're saying they're working as charm, it includes Recovery too? :)

dreamwhite, 2020-04-11:
> So when you're saying they're working as charm, it includes Recovery too?

Yes, they're working. Sorry for stupid question.

> Those patches are not needed since v2.3 :)

Okay, I'll try disabling them and seeing if there is any change

Many thanks :)

closed 2020-04-11 by ben9923